openwrt/openwrt #16994: Please add CONFIG_NF_CT_NETLINK_TIMEOUT for all current versions

brada4, 2024-11-17:
### Describe the bug

It happens that I need to adapt the timeout for qosmate to avoid reverting connection direction (or in more popular case to stop youtube freezing while maintaining DNS and NTP state number under control)

### OpenWrt version

r24128-cb8b632f1a

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link RE200 v4

### Image kind

Official downloaded image

### Steps to reproduce

Taken from `man 8 nft` example
```
cat c.nft 
           table ip filter {
                   ct timeout customtimeout {
                           protocol tcp;
                           l3proto ip
                           policy = { established: 120, close: 20 }
                   }

                   chain output {
                           type filter hook output priority filter; policy accept;
                           ct timeout set "customtimeout"
                   }
           }
# Ctrl-D
```

### Actual behaviour

Bails with error

### Expected behaviour

Works

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vincejv, 2024-12-24:
does this PR address this?

- https://github.com/openwrt/openwrt/pull/17267

brada4, 2024-12-24:
Yes, but http3 like jetty has 5min keepalive time, which needs default up, then ntp dns snmp need to be adjusted for smaller one to save ct state count....