openwrt/openwrt #16996: apk: imagebuilder package_depends broken

jszinger, 2024-11-17:
### Describe the bug

The SNAPSHOT image builder command `make package_depends PACKAGE=usbutils` dies with the error `bash: line 1: /image-builder/openwrt-imagebuilder-x86-64.Linux-x86_64/staging_dir/host/bin/opkg: No such file or directory`

### OpenWrt version

SNAPSHOT r28114-84665993d6

### OpenWrt release

SNAPSHOT r28114-84665993d6

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Steps to reproduce

Download and extract https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-imagebuilder-x86-64.Linux-x86_64.tar.zst

`make package_depends PACKAGE=usbutils`

### Actual behaviour

OK: 16 MiB in 130 packages
bash: line 1: /image-builder/openwrt-imagebuilder-x86-64.Linux-x86_64/staging_dir/host/bin/opkg: No such file or directory
make: *** [Makefile:367: package_depends] Error 127

### Expected behaviour

Command completes without error and prints the list of dependencies.

### Additional info

```
$ make info
Current Target: "x86/64"
Current Architecture: "x86_64"
Current Revision: "r28114-84665993d6"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd uci uclient-fetch urandom-seed urngd partx-utils mkf2fs e2fsprogs kmod-button-hotplug grub2-bios-setup procd-ujail dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe apk-mbedtls
Available Profiles:

generic:
    Generic x86/64
    Packages: kmod-amazon-ena kmod-amd-xgbe kmod-bnx2 kmod-dwmac-intel kmod-e1000e kmod-e1000 kmod-forcedeth kmod-fs-vfat kmod-igb kmod-igc kmod-ixgbe kmod-r8169 kmod-tg3
    hasImageMetadata: 0
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-11-18:
> bash: line 1: /image-builder/openwrt-imagebuilder-x86-64.Linux-x86_64/staging_dir/host/bin/opkg: No such file or directory

This one is actually bug, its needs to be fixed for apk (its opkg copy&pasta) https://github.com/openwrt/openwrt/blob/a5d23e3aee9bd616044c3eadfe2dbcad5d18b163/target/imagebuilder/files/Makefile#L366, but then

robimarko, 2024-11-19:
Well, I am having a weird issue reproducing this:
```
[robimarko@fedora openwrt-imagebuilder-qualcommax-ipq807x.Linux-x86_64]$ make package_depends PACKAGE=kitty-terminfo
make: TMPDIR value /home/robimarko/Building/AX3600/apk-work/openwrt-imagebuilder-qualcommax-ipq807x.Linux-x86_64/tmp: No such file or directory
make: using default temporary directory '/tmp'
ERROR: Unable to setup the cache: No such file or directory
ERROR: Failed to open apk database: No such file or directory
make[1]: *** [Makefile:203: package_reload] Error 99
make: *** [Makefile:362: package_depends] Error 2
```

After some poking around it seems to fail on `package_reload` now before we even get to calling OPKG instead of APK

robimarko, 2024-11-19:
Fun fact, OPKG support for `package_whatdepends` is broken in IB ever since APK was initially added.

commit 03eeeffb (2024-11-20): imagebuilder: fix APK package_depends support

Currently trying to use the 'package_depends' feature of ImageBuilder will
result in it trying to call OPKG:
make package_depends PACKAGE="usbutils"
OK: 26 MiB in 142 packages
bash: line 1: openwrt-imagebuilder-qualcommax-ipq807x.Linux-x86_64/staging_dir/host/bin/opkg: No such file or directory
make: *** [Makefile:363: package_depends] Error 127

It looks like a copy/paste error when APK support was being added, so lets
fix it by calling APK and its "info --depends" command.

Fixes: #16996
Fixes: d788ab376f85 ("build: add APK package build capabil [... cut]