openwrt/openwrt #17000: jq: compiled without ONIGURUMA regex libary

Wokkocher, 2024-11-17:
### Describe the bug

Hello all. This is a feature request after I asked on openwrt forum already. I was told to ask here again and tag @ratkaj in/at/on this issue.

[Feature request on forum](https://forum.openwrt.org/t/package-jq-has-no-regex-support/)

I am trying to write a script to parse a JSON file from mediathekviewweb API to script some movie downloading. However, this is working on my PC but not on my device. After some debugging I stumbled upon an unexpected error:

`jq: error (at <stdin>:0): jq was compiled without ONIGURUMA regex libary. match/test/sub and related functions are not available.`

As I stated on forum posts, one can work around this, so there is no high pressure to solve this. I would like to ask for a second package version with onigurma compiled in (e.g. `jq-full`), so jq can perform regular expressions. I am not experienced enough to do this, I never compiled anything, I am used to scripting.

May I ask if someone is so kind to do this for me at some time?
Thank you all!

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ath79/nand

### Device

GL.iNet GL-AR300M (NOR)

### Image kind

Official downloaded image

### Steps to reproduce

```
root@device:~$ opkg update
root@device:~$ opkg install jq
root@device:~$ curl -s -H "Content-Type: text/plain" -d '{ "queries" : [ {"fields" : ["topic"], "query" : "Sendung Maus"} ] , "size" : 9999 }' "https://mediathekviewweb.de/api/query" > resp.txt
root@device:~$ cat resp.txt | jq '.result.results | .[0].title | scan("foo")'
```

### Actual behaviour

No regex engine is compiled in and therefore such functions like `scan` / `match` / `sub` cannot be used. A call will yield this error:
`jq: error (at <stdin>:0): jq was compiled without ONIGURUMA regex libary. match/test/sub and related functions are not available.`

### Expected behaviour

No error if one is using functions like `scan` / `match` / `sub` .

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-17:
Helps to report in proper repo.
https://github.com/openwrt/packages/blob/fc335523b781becb2216f5889d86da04161406c6/utils/jq/Makefile#L30

Wokkocher, 2024-11-17:
Oops... This wasn't on purpose, it was just my first issue on github at all. I hope someone can move it if possible.

brada4, 2024-11-17:
You can copy / paste,, thats faster.

Wokkocher, 2024-11-18:
Done. Thank you anyway and sorry again.

closed 2024-11-18 by ynezz