openwrt/openwrt #17001: apk: WARNING: updating and opening xxxx/packages.adb: UNTRUSTED signature

peterwillcn, 2024-11-18:
### Describe the bug

apk: WARNING: updating and opening xxxx/packages.adb: UNTRUSTED signature

### OpenWrt version

r28124-efc0c4666b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-11-18:
If you don't follow the guidelines when creating the ticket, like steps to reproduce etc. do not expect anyone looking into this issue.

peterwillcn, 2024-11-18:
> If you don't follow the guidelines when creating the ticket, like providing steps to reproduce etc. do not expect anyone looking into this issue.

The reproduction has been added :-)

ynezz, 2024-11-18:
#### Diffconfig

In case of a self-built image, please attach diffconfig.
```shell
./scripts/diffconfig.sh
```

peterwillcn, 2024-11-18:
> #### Diffconfig
> In case of a self-built image, please attach diffconfig.
> 
> ```shell
> ./scripts/diffconfig.sh
> ```

Done

peterwillcn, 2024-11-18:
Warning should not be given, or the verification signature should not be ignored in the configuration( apk config has the same configuration parameters as opkg.conf: Comment out the line # option check_signature).

peterwillcn, 2024-11-18:
> Do you've your key `public-key.pem` installed/available on your device in `/etc/apk/keys` ?

Fixed The import of public-key.pem has been fixed. Thank you very much, @ynezz

ynezz, 2024-11-18:
It should work automatically, so I'm wondering what went wrong, that it didn't worked out of the box, care to elaborate?

https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/base-files/Makefile#L125-L128

peterwillcn, 2024-11-18:
> It should work automatically, so I'm wondering what went wrong, that it didn't worked out of the box, care to elaborate?
> 
> https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/base-files/Makefile#L125-L128

[image]

peterwillcn, 2024-11-18:
@ynezz  The public key for self build image should be public-key.pem, not openwrt-snapshotsprom. Please add a switch to use snapshotsprom or self public key-pem compilation options, please refer to the above photos for details.

ynezz, 2024-11-18:
> The public key for self build image should be public-key.pem, not openwrt-snapshots.pem

You can have multiple keys in there, as you can use multiple feeds, from different sources, using different keys. This should work and if it doesn't work, then there is some problem.  That `openwrt-snapshots.pem` key comes from `openwrt-keyring` package

https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/system/openwrt-keyring/Makefile#L34-L37

So you're saying, that having both keys in the `/etc/apk/keys/` directory on your board makes `apk update` fail with your custom feed? If you add some official feed, does it work? Is it a problem of `apk-openssl` ? I've tried multiple public keys with `apk-mbedtls` and they were working.

peterwillcn, 2024-11-18:
> > 1, I have a question: Is the correct variable value for $BUILD_KEY_APK_PUB during compilation openwrt-snapshots.pem or public-key.pem?
> 
> 1. `public-key.pem` comes from `BUILD_KEY_APK_PUB` definition https://github.com/openwrt/openwrt/blob/53eab6198bf87d12055409bc4a38d58b68291b2f/rules.mk#L288-L290
> 2. `public-key.pem` is provided by `base-files` package https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/base-files/Makefile#L125-L128
> 3. `openwrt-snapshots.pem` is provided by `openwrt-keyring` package https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/system/openwrt-keyring/Makefile#L34-L37
> 
> > 2, There is no problem with multiple public keys, but the problem lies in the missing public-key.pem only openwrt-snapshots.pem during the compilation process. the problem may be that during the installation (apk add openwrt-keyring) process of openwrt-keyring, `$(INSTALLDATA) $(PKG_BUILD-DIR)/apk/*. pem $(1)/etc/apk/keys/did not copy the key to/etc/apk/keys/`
> 
> I see the problem now, you've `CONFIG_BUILDBOT=y` so in that case, the key is not being installed:
> 
> https://github.com/openwrt/openwrt/blob/efc0c4666b5aa0cf0067d794964f12d9a45190c6/package/base-files/Makefile#L124-L129

@ynezz  Oh, it turned out to be a problem caused by my incorrect configuration. `CONFIG_BUILDBOT=y`. Thank you very much for learning a lot from you.