openwrt/openwrt #17010: ZyXEL XGS1250-12 initramfs network broken

achterin, 2024-11-18:
### Describe the bug

Trying to install the current snapshot build on this device results in no traffic being routed, thus preventing the installation of the sysupgrade imsage.

### OpenWrt version

r28124-efc0c4666b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl930x

### Device

Zyxel XGS1250-12 Switch

### Image kind

Official downloaded image

### Steps to reproduce

Follow the installation steps in the wiki.
Install the initramfs via web-ui, reboot, set the uboot-env as described, reboot, observe broken network.

### Actual behaviour

Network is broken, ping does not work.

### Expected behaviour

Working network.

### Additional info

Complete bootlog:

`U-Boot Version V1.0.0.4 (Dec 01 2022 - 15:46:54)

Board: RTL9300 CPU:800MHz LX:175MHz DDR:600MHz
DRAM:  128 MB
SPI-F: MXIC/C22018/MMIO16-1/ModeC 1x16 MB (plr_flash_info @ 83f74a30)
Loading 65536B env. variables from offset 0xe0000
Net:   Net Initialization Skipped
No ethernet found.
 0 
Enable network
RTCORE Driver Module Initialize
  IOAL init
  Hardware-profile probe  GPIO probe (unit 0): (found)
  GPIO Init

 (XGS1250_12)
  Hardware-profile init
  GPIO probe (unit 0): (found)
  GPIO Init rtl9300_gpio_init had already been initialized!

  SPI init (unit 0) 
  I2C probe (unit 0)
  I2C init (unit 0)
  NIC probe (unit 0)
  Loader RTNIC Driver Module Initialize
  IOAL init
RTK Driver Module Initialize
  MAC probe (unit 0)
    Chip 9302 (found)
[... 409 lines cut ...]
[   46.119561] switch: port 12(lan12) entered disabled state
[   46.125670] rtl83xx-switch switch@1b000000 lan12: entered allmulticast mode
[   46.134433] rtl83xx-switch switch@1b000000 lan12: entered promiscuous mode
[   46.149116] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   49.237861] rtl83xx-switch switch@1b000000 lan9: Link is Up - 1Gbps/Full - flow control rx/tx
[   49.247480] switch: port 9(lan9) entered blocking state
[   49.253316] switch: port 9(lan9) entered forwarding state
[   50.734458] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   50.741805] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported

BusyBox v1.36.1 (2024-11-17 19:54:54 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r28124-efc0c4666b
 -----------------------------------------------------
`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

achterin, 2024-11-18:
Bootlog:

> U-Boot Version V1.0.0.4 (Dec 01 2022 - 15:46:54)
>
> Board: RTL9300 CPU:800MHz LX:175MHz DDR:600MHz
> DRAM:  128 MB
> SPI-F: MXIC/C22018/MMIO16-1/ModeC 1x16 MB (plr_flash_info @ 83f74a30)
> Loading 65536B [...]

Neustradamus, 2024-11-26:
@oliv3r, @svanheule, @TobleMiner, @achterin, @plaurien, @chapichapo, @Biergartenparadoxon, @plappermaul, @SimonFischer04, @0xharshal, @mr-wewlad, @Borromini, @jonasjelonek, @lucize, @tristan-k, @pierrepinon: Can you look?

SimonFischer04, 2024-11-29:
@achterin @Neustradamus 

1. did it work before?

are you using the Aquantia (10G) ports?.  The Realtek ones do not work yet. See: https://github.com/openwrt/openwrt/pull/16457#issuecomment-2423118360.

and you have configured / temporarily disabled the firewall (/etc/init.d/firewall stop)?

Both of these have also tripped me up a few times. :)

achterin, 2024-11-29:
For the time being I've downgraded to the last stable (23.05.5) which works fine. I'm pretty sure I've tested all RJ-45 ports and while all indicated an established link, no packets were shown via ifconfig on the switch side.

No, I did not disable the firewall.

plaurien, 2024-11-29:
Thank you all.
Can I highlight that it fails when using the non primary vlan1? So if I trunk one port with 2 vlans and use another port as access port on something else than vlan1 it does not see any traffic.

Pierre

SimonFischer04, 2024-12-17:
> This doesn't work on snapshot for me right now:

Then the snapshot may not include firewall package. If ping works on Aquantia then it should not be firewall issue anyway. (and switching traffic also, just a problem if you try to ping/accesss the switch interface  ip)

> I can confirm Aquantia ports work, Realtek ports do not

yes, see the liked comment. the realtek ones did not work before either, so it's kind of "expected broke" currently. There are some things missing to get it working,

achterin, 2024-12-17:
> yes, see the liked comment. the realtek ones did not work before either, so it's kind of "expected broke" currently. There are some things missing to get it working,

sorry, but thats not true. when i tried the latest 23.05 stable the realtek ports worked fine.

SimonFischer04, 2024-12-18:
> > yes, see the liked comment. the realtek ones did not work before either, so it's kind of "expected broke" currently. There are some things missing to get it working,
> 
> sorry, but thats not true. when i tried the latest 23.05 stable the realtek ports worked fine.
> 

Sorry, then i don't know when it broke.
@plappermaul asked me to test https://github.com/openwrt/openwrt/pull/16457#issuecomment-2423118360
and there (after some initial confusion) nothing broke. And as i have interpreted his comment, this was expected.
Maybe he can give some insight.

achterin, 2024-12-18:
My guess is that it broke during the kernel update for this target. would be nice if @plappermaul can give use some insights why it was expected to not work and what is missing that was there before. i tried to look into this but found very little detail.

plappermaul, 2024-12-23:
With the above information I'm confused about the state of that device. To narrow that down please create three initramfs images based on:

- Latest commit of 23.05 branch
- 24.10 branch up to commit "[realtek: rework hardware-assisted indirect mdio access](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=0f7131baf3c8fe08b963802d467d9a2755facfbd)
- Latest commit of 24.10 branch

For all three versions plase answer the following questions:

- Do the Realtek ports **send/receive** traffic?
- Do the Realtek ports **forward** traffic?
- Do the Aquantia ports **send/receive** traffic?
- Do the Aquantia ports **forward** traffic?

Best regards.

Markus

achterin, 2025-01-20:
> With the above information I'm confused about the state of that device. To narrow that down please create three initramfs images based on:
>
> * Latest commit of 23.05 branch
> * 24.10 branch up to [...]

Sorry for the delay!
My test setup is the following:
PC1 connected to Port 1 (Realtek Port)
PC2 connected to Port 9 (Aquantia Port)

Instead of the last commit of each branch i just used the latest official image available on download.openwrt.org

23.05.5:
Realtek Port -> Switch works
Aquantia Port -> Switch works
Realtek Port <-> Aquantia Port does **not** work

24.10 branch up to commit 0f7131baf3c8fe08b963802d467d9a2755facfbd:
Realtek Port -> Switch works
Aquantia Port -> Switch works
Realtek Port <-> Aquantia Port works

24.10.0-rc5:
Realtek Port -> Switch does **not** work
Aquantia Port -> Switch works
Realtek Port <-> Aquantia Port does **not** work

Hope this helps. If I should test anything else, let me know.

Borromini, 2025-01-22:
@achterin Could you run guy bisect between both so we can narrow it down to a specific commit? I have an XGS1250-12 as well but unfortunately do not have the time at this point.

@TobleMiner @0xharshal Could you guys maybe take a peek as well?

achterin, 2025-01-25:
@Borromini i just checked the git log and to me it seems that the following commits introduce the kernel switch of the realtek target and during development @plappermaul said that the realtek ports not working with kernel 6.6 is expected.

plappermaul, 2025-01-26:
A quick idea without any chance of testing. I assume that you see some boot messages like "...port ... mode ... **speed -1** ...". This is one driver bit I could not refactor for RTL93XX. To get an idea what I did for RTL83XX see https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=cd958d945be06a1b810d55912f32d6e922d952ea

That means:
- Cleanup rtl93xx_phylink_mac_config()
- Enhance rtl93xx_phylink_mac_link_up()

achterin, 2025-01-26:
> A quick idea without any chance of testing. I assume that you see some boot messages like "...port ... mode ... **speed -1** ...". This is one driver bit I could not refactor for RTL93XX. To get an [...]

I guess you mean these ones:

```
[   34.778155] Using MAC 0000f44d5c77d4e9
[   34.782571] RESETTING 9300, CPU_PORT 28
[   34.987212] rtl838x-eth 1b00a300.ethernet eth0: configuring for fixed/internal link mode
[   34.996215] In rtl838x_mac_config, mode 1
[   35.015025] rtl838x-eth 1b00a300.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   35.026492] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   35.033864] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   35.118375] rtl83xx-switch switch@1b000000 lan1: configuring for phy/xgmii link mode
[   35.127075] rtl93xx_phylink_mac_config port 0, mode 0, phy-mode: xgmii, speed -1, link 0
[   35.136135] rtl93xx_phylink_mac_config SDS is 2
[   35.142792] 8021q: adding VLAN 0 to HW filter on device lan1
[   35.188839] switch: port 1(lan1) entered blocking state
[   35.194737] switch: port 1(lan1) entered disabled state
[   35.200598] rtl83xx-switch switch@1b000000 lan1: entered allmulticast mode
[   35.208335] rtl838x-eth 1b00a300.ethernet eth0: entered allmulticast mode
[   35.216805] rtl83xx-switch switch@1b000000 lan1: entered promiscuous mode
[   35.291157] rtl930x_set_l3_egress_intf writing to index 0: 00000201 10101492
[   35.302270] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   35.328515] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   35.501855] rtl83xx-switch switch@1b000000 lan2: configuring for phy/xgmii link mode
[   35.510558] rtl93xx_phylink_mac_config port 1, mode 0, phy-mode: xgmii, speed -1, link 0
[   35.519617] rtl93xx_phylink_mac_config SDS is -1
[   35.525732] 8021q: adding VLAN 0 to HW filter on device lan2
[   35.604884] switch: port 2(lan2) entered blocking state
[   35.610724] switch: port 2(lan2) entered disabled state
[   35.616652] rtl83xx-switch switch@1b000000 lan2: entered allmulticast mode
[   35.647193] rtl83xx-switch switch@1b000000 lan2: entered promiscuous mode
[   35.741850] rtl83xx-switch switch@1b000000 lan3: configuring for phy/xgmii link mode
[   35.750563] rtl93xx_phylink_mac_config port 2, mode 0, phy-mode: xgmii, speed -1, link 0
[   35.759608] rtl93xx_phylink_mac_config SDS is -1
[   35.765535] 8021q: adding VLAN 0 to HW filter on device lan3
[   35.859115] switch: port 3(lan3) entered blocking state
[   35.865012] switch: port 3(lan3) entered disabled state
[   35.870873] rtl83xx-switch switch@1b000000 lan3: entered allmulticast mode
[   35.927608] rtl83xx-switch switch@1b000000 lan3: entered promiscuous mode
[   36.053491] rtl83xx-switch switch@1b000000 lan4: configuring for phy/xgmii link mode
[   36.062193] rtl93xx_phylink_mac_config port 3, mode 0, phy-mode: xgmii, speed -1, link 0
[   36.071252] rtl93xx_phylink_mac_config SDS is -1
[   36.076744] 8021q: adding VLAN 0 to HW filter on device lan4
[   36.178180] switch: port 4(lan4) entered blocking state
[   36.184020] switch: port 4(lan4) entered disabled state
[   36.189947] rtl83xx-switch switch@1b000000 lan4: entered allmulticast mode
[   36.250520] rtl83xx-switch switch@1b000000 lan4: entered promiscuous mode
[   36.414352] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   36.421706] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   36.446582] rtl83xx-switch switch@1b000000 lan5: configuring for phy/xgmii link mode
[   36.455297] rtl93xx_phylink_mac_config port 4, mode 0, phy-mode: xgmii, speed -1, link 0
[   36.464369] rtl93xx_phylink_mac_config SDS is -1
[   36.470058] 8021q: adding VLAN 0 to HW filter on device lan5
[   36.565921] switch: port 5(lan5) entered blocking state
[   36.571768] switch: port 5(lan5) entered disabled state
[   36.577708] rtl83xx-switch switch@1b000000 lan5: entered allmulticast mode
[   36.654915] rtl83xx-switch switch@1b000000 lan5: entered promiscuous mode
[   36.752480] rtl83xx-switch switch@1b000000 lan6: configuring for phy/xgmii link mode
[   36.761192] rtl93xx_phylink_mac_config port 5, mode 0, phy-mode: xgmii, speed -1, link 0
[   36.770257] rtl93xx_phylink_mac_config SDS is -1
[   36.775733] 8021q: adding VLAN 0 to HW filter on device lan6
[   36.868988] switch: port 6(lan6) entered blocking state
[   36.874892] switch: port 6(lan6) entered disabled state
[   36.880753] rtl83xx-switch switch@1b000000 lan6: entered allmulticast mode
[   36.889373] rtl83xx-switch switch@1b000000 lan6: entered promiscuous mode
[   37.053666] rtl83xx-switch switch@1b000000 lan7: configuring for phy/xgmii link mode
[   37.062376] rtl93xx_phylink_mac_config port 6, mode 0, phy-mode: xgmii, speed -1, link 0
[   37.071434] rtl93xx_phylink_mac_config SDS is -1
[   37.077068] 8021q: adding VLAN 0 to HW filter on device lan7
[... 16 lines cut ...]
[   37.655204] switch: port 9(lan9) entered blocking state
[   37.661047] switch: port 9(lan9) entered disabled state
[   37.666975] rtl83xx-switch switch@1b000000 lan9: entered allmulticast mode
[   37.675628] rtl83xx-switch switch@1b000000 lan9: entered promiscuous mode
[   37.770799] rtl83xx-switch switch@1b000000 lan10: configuring for phy/usxgmii link mode
[   37.779793] rtl93xx_phylink_mac_config port 25, mode 0, phy-mode: usxgmii, speed -1, link 0
[   37.789142] rtl93xx_phylink_mac_config SDS is 7
[   37.839749] 8021q: adding VLAN 0 to HW filter on device lan10
[... cut ...]
[   38.071070] switch: port 11(lan11) entered disabled state
[   38.077200] rtl83xx-switch switch@1b000000 lan11: entered allmulticast mode
[   38.085980] rtl83xx-switch switch@1b000000 lan11: entered promiscuous mode
[   38.145127] rtl83xx-switch switch@1b000000 lan12: configuring for inband/1000base-x link mode
[   38.154716] rtl93xx_phylink_mac_config port 27, mode 2, phy-mode: 1000base-x, speed -1, link 0
```

What seems odd to me is that the "speed -1" log messages also appear for the non realtek-ports.

janh, 2025-02-27:
> * Cleanup rtl93xx_phylink_mac_config()
> * Enhance rtl93xx_phylink_mac_link_up()

@plappermaul: I quickly adapted these changes for RTL930x (https://github.com/janh/openwrt/commit/fa36a3ff1a29ff489f63c1f0562a7397334e165c), and it does actually work on my XGS1010-12!

The RTL8218D ports work fine, and SFP also still works with this change (tested only with WTD RTXM191-400, as I don't have another module).

However, the two RTL8226 ports only work at 2.5G (link works at other speeds, but no data is transferred).

achterin, 2025-02-27:
> > * Cleanup rtl93xx_phylink_mac_config()
> > * Enhance rtl93xx_phylink_mac_link_up()
> 
> [@plappermaul](https://github.com/plappermaul): I quickly adapted these changes for RTL930x ([janh@fa36a3f](https://github.com/janh/openwrt/commit/fa36a3ff1a29ff489f63c1f0562a7397334e165c)), and it does actually work on my XGS1010-12!
> 
> The RTL8218D ports work fine, and SFP also still works with this change (tested only with WTD RTXM191-400, as I don't have another module).
> 
> However, the two RTL8226 ports only work at 2.5G (link works at other speeds, but no data is transferred).

interesting, i also tried to rewrite it but could not get the ports to work. i'll test your code and see if it works with my xgs1250. thanks!

janh, 2025-03-04:
The RTL8226 issue I mentioned earlier is unrelated (the PHY switches between HSGMII and SGMII depending in link speed, which is currently not handled).

I'll probably send a pull request or patch in the next few days (but of course you can already try the changes in my xgs1010-12 branch now).

langerma, 2025-03-05:
id'd be happy to test anything for the xgs1250-12 as my switch is not yet in production

antanasbruzas, 2025-03-10:
I can also help with testing if needed

achterin, 2025-03-10:
> The RTL8226 issue I mentioned earlier is unrelated (the PHY switches between HSGMII and SGMII depending in link speed, which is currently not handled).
> 
> I'll probably send a pull request or patch in the next few days (but of course you can already try the changes in my xgs1010-12 branch now).

just tested your changes on my xgs1250-12 and can confirm that the realtek ports do work now. cant wait for your pr. please also open up a backport pr for the latest openwrt release as the ports are also broken in this release. thanks!

phqzgunsfjror, 2025-03-16:
Sounds great. When do we expect to have that in snapshot version for testing?