openwrt/openwrt #17014: Wifi network quality drops after ~24 hours with conflicting DHCP lease

bitcores, 2024-11-19:
### Describe the bug

If a static lease is assigned to a device using one interface (MAC address), perhaps wired Ethernet, and the device connects later using a different interface (different MAC address), for example wireless, and stays connected persistently the Wifi network quality degrades and eventually requires a restart.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR1043ND v2

### Image kind

Official downloaded image

### Steps to reproduce

Configure a static lease for a device connected by wired ethernet.
Disconnect the device and allow the lease to expire.
Connect the device to the wireless network and leave it connected.
Increased wifi usage may speed up the degradation (unconfirmed).

### Actual behaviour

After a period of time (up to one or two days) wifi connectivity degrades and devices lose connectivity to the wifi network.

### Expected behaviour

Wifi connectivity should not be impacted.

### Additional info

A repeated warning log message similar to 
```
Mon Nov 18 19:38:17 2024 daemon.warn dnsmasq-dhcp[1]: not giving name sdropi.lan to the DHCP lease of 172.16.31.45 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 172.16.17.3
```
is indication that the device is in the state what will lead to the degradation of the wifi connectivity.

This issue similar in outcome to that described in the ticket [https://dev.archive.openwrt.org/ticket/12372](https://dev.archive.openwrt.org/ticket/12372), but I am unsure if it is related (it seems there have been a few issues similar but no one has shown a log with this warning).

I have experienced this issue for more than a year. For a time I was using a cronjob to restart the wifi to mitigate it, but it wasn't always effective.
After clearing the offending static lease the wifi connectivity is stable for weeks. I have replicated the issue twice since discovering the cause which is why I am now opening this issue.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-11-19:
First upgrade to 23.05.5.

bitcores, 2024-11-22:
Updated to 23.05.05 and have confirmed that the Wifi network still degrades.
Will need more time to confirm stable Wifi in the absence of the offending static lease.

bitcores, 2025-01-03:
I've got some more information now to add to this report.

First, I attempted to replicate the issue on another device, a Buffalo WMR-300 which has a Ralink/MediaTek MT7620 chipset (23.05.5) but was unsuccessful. This may indicate that the issue affects a small range of chipsets (Qualcomm Atheros only?)

Second, my TP-Link TL-WR1043ND v2 seems to be absolutely stable on 23.05.5 without the offending static lease, but will suffer degraded Wifi performance whenever I recreate the static lease.

Third, the dnsmasq warning message is due to enabling forward/reverse DNS when configuring the static lease but this is NOT related to the issue.  Even with forward/reverse DNS disabled the problem still occurs.