openwrt/openwrt #17025: Kernel module rtw89_8852be causes reboot

Melkior81, 2024-11-20:
### Describe the bug

After loading module and firmware OpenWrt reboots. No error messages are visible, or saved to file (I mounted HDD for this)

### OpenWrt version

r28026-faebcc61eb

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Steps to reproduce

Opkg install kmod-rtw89-8852be

### Actual behaviour

Infant restart adter loading module and firmware

### Expected behaviour

Wifi card working, no restart

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-20:
? Can you reproduce the problem on bare metal
? can you capture serial console output (virsh console vmname)

Melkior81, 2024-11-21:
Unfortunately I only used openwrt on proxmox and passed device. So there is
no serial co sole. Last message I see before reboot is about loaded
firmware. No error.
I may be able to test this running openwrt from usb key, but it will take
time. Also I know of another person having identical problem.  So it's not
just my error.

Melkior81, 2024-11-21:
> You can test VM with same card with like debian VM, it tells whether
> problem is with virtualisation or openwrt

I tested no live debian and passed card works fine in VM.

brada4, 2024-11-21:
Get serial output from crash. Ask proxmox to help you if needed.

Melkior81, 2024-11-22:
This is serial output after: "opkg install kmod-rtw89-8852be":
"
[  176.087051] kmodloader: loading kernel modules from /etc/modules.d/*
[  176.091892] rtw89_8852be 0000:01:00.0: loaded firmware
rtw89/rtw8852b_fw-1.bin
[  176.094112] BUG: unable to handle page fault for address:
0000000000001090
[  176.094898] #PF: supervisor write access in kernel mode
[  176.095500] #PF: error_code(0x0002) - not-present page
[  176.096087] PGD 800000000c4d4067 P4D 800000000c4d4067 PUD d6f5067 PMD 0
[  176.096775] Oops: 0002 [#1] SMP PTI
[  176.097251] CPU: 0 PID: 5631 Comm: kmodloader Tainted: G           O
  6.6.60 #0
[  176.097996] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS
rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[  176.098962] RIP: 0010:0xffffffffa0d80b09 ***@***.***
+0xa000]
[  176.099707] Code: 90 89 f6 48 03 b7 a8 3d 07 00 88 16 c3 cc cc cc cc 90
90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 89 f6 48 03 b7 a8 3d 07 00
<66> 89 16 c3 cc cc cc cc 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40
[  176.101440] RSP: 0018:ffffc900004cfa10 EFLAGS: 00010206
[  176.102052] RAX: ffffffffa0d80b00 RBX: ffff88800b682020 RCX:
0000000000000000
[  176.102810] RDX: 0000000000000718 RSI: 0000000000001090 RDI:
ffff88800b682020
[  176.103566] RBP: ffffc900004cfa40 R08: ffffc900004cf9ec R09:
ffffc900004cf9ec
[  176.104329] R10: 0000000000000001 R11: ffffffff81c60d90 R12:
0000000000000015
[  176.105085] R13: 0000000000000719 R14: ffffc900004cfa88 R15:
0000000000000719
[  176.105849] FS:  00007fd9352b4b28(0000) GS:ffff88801f400000(0000)
knlGS:0000000000000000
[  176.106677] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  176.107355] CR2: 0000000000001090 CR3: 000000000b510000 CR4:
00000000000006f0
[  176.108127] Call Trace:
[  176.108597]  <TASK>
[  176.109034]  ? 0xffffffff8103a230
[  176.109554]  ? 0xffffffff8103a30f
[  176.110064]  ? 0xffffffff81071efc
[  176.110581]  ? 0xffffffffa0d80b09 ***@***.***+0xa000]
[  176.111305]  ? 0xffffffff81197d6d
[  176.111825]  ? 0xffffffffa0d80b09 ***@***.***+0xa000]
[  176.112547]  ? 0xffffffff810f9b7b
[  176.113053]  ? 0xffffffff81073152
[  176.113561]  ? 0xffffffff8107224a
[  176.114066]  ? 0xffffffff810723c1

"

Melkior81, 2024-11-22:
This is longer. First I installed all modules and then after starting VM this ends up in console:
"
[    8.311170] rtw89_8852be 0000:01:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    8.313175] BUG: unable to handle page fault for address: 0000000000001090
[    8.313894] #PF: supervisor write access in kernel mode
[    8.314552] #PF: error_code(0x0002) - not-present page
[    8.315145] PGD 8000000004d5f067 P4D 8000000004d5f067 PUD 4cf2067 PMD 0 
[    8.315827] Oops: 0002 [#1] SMP PTI
[    8.316311] CPU: 1 PID: 1922 Comm: kmodloader Tainted: G           O       6.6.60 #0
[    8.317052] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[    8.318015] RIP: 0010:0xffffffffa0bd6b09 [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.318730] Code: 90 89 f6 48 03 b7 a8 3d 07 00 88 16 c3 cc cc cc cc 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 89 f6 48 03 b7 a8 3d 07 00 <66> 89 16 c3 cc cc cc cc 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40
[    8.320416] RSP: 0018:ffffc9000218fa10 EFLAGS: 00010206
[    8.321033] RAX: ffffffffa0bd6b00 RBX: ffff88800d102020 RCX: 0000000000000000
[    8.321771] RDX: 0000000000000718 RSI: 0000000000001090 RDI: ffff88800d102020
[    8.322503] RBP: ffffc9000218fa40 R08: ffffc9000218f9ec R09: ffffc9000218f9ec
[    8.323234] R10: 0000000000000001 R11: ffffffff81c60d90 R12: 0000000000000015
[    8.323963] R13: 0000000000000719 R14: ffffc9000218fa88 R15: 0000000000000719
[    8.324698] FS:  00007fca5633eb28(0000) GS:ffff88801f500000(0000) knlGS:0000000000000000
[    8.325481] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.326129] CR2: 0000000000001090 CR3: 0000000004b92000 CR4: 00000000000006e0
[    8.326859] Call Trace:
[    8.327300]  <TASK>
[    8.327735]  ? 0xffffffff8103a230
[    8.328220]  ? 0xffffffff8103a30f
[    8.328706]  ? 0xffffffff81071efc
[    8.329186]  ? 0xffffffffa0bd6b09 [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.329857]  ? 0xffffffff81197d6d
[    8.330334]  ? 0xffffffffa0bd6b09 [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.331006]  ? 0xffffffff810f9b7b
[    8.331483]  ? 0xffffffff81073152
[    8.331962]  ? 0xffffffff8107224a
[    8.332436]  ? 0xffffffff810723c1
[    8.332911]  ? 0xffffffff8125d210
[    8.333380]  ? 0xffffffff8107250e
[    8.333854]  ? 0xffffffff81c92631
[    8.334321]  ? 0xffffffff81e00c27
[    8.334793]  ? 0xffffffff81c60d90
[    8.335257]  ? 0xffffffffa0bd6b00 [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.335914]  ? 0xffffffffa0bd6b09 [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.336566]  ? 0xffffffffa0bd721c [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.337224]  0xffffffffa0bdb63d [rtw89_pci@00000000a4d01d0d+0xa000]
[    8.337869]  0xffffffff814d29e3
[    8.338322]  0xffffffff8183020b
[    8.338777]  0xffffffff81830423
[    8.339217]  0xffffffff818304ff
[    8.339659]  0xffffffff818306d6
[    8.340092]  ? 0xffffffff81830650
[    8.340532]  0xffffffff8182dea5
[    8.340962]  0xffffffff8182fa29
[    8.341384]  0xffffffff8182f1d5
[    8.341814]  ? 0xffffffffa0dd1010 [rtw89_8852be@00000000942ac253+0x1000]
[    8.342445]  0xffffffff81831699
[    8.342864]  ? 0xffffffffa0dd1010 [rtw89_8852be@00000000942ac253+0x1000]
[    8.343493]  0xffffffff814d102d
[    8.343912]  0xffffffffa0dd1033 [rtw89_8852be@00000000942ac253+0x1000]
[    8.344525]  0xffffffff81000815
[    8.344941]  ? 0xffffffff81289fbe
[    8.345356]  0xffffffff81156673
[    8.345761]  0xffffffff811584cf
[    8.346150]  0xffffffff8115878e
[    8.346526]  ? 0xffffffff8115878e
[    8.346910]  0xffffffff811588c5
[    8.347276]  0xffffffff8100335a
[    8.347647]  0xffffffff81c8bc4d
[    8.348005]  0xffffffff81e00130
[    8.348359] RIP: 0033:0x7fca562f6968
[    8.348743] Code: 4c 89 4c 24 48 4c 8b 4c 24 60 48 89 54 24 10 48 8b 7e 08 48 89 74 24 18 48 8b 51 18 c7 44 24 08 28 00 00 00 48 8b 76 10 0f 05 <48> 89 c7 e8 ab 5d fe ff 48 83 c4 58 c3 41 57 41 56 41 55 41 54 55
[    8.350189] RSP: 002b:00007ffc40d744f0 EFLAGS: 00000206 ORIG_RAX: 00000000000000af
[    8.350835] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007fca562f6968
[... 1 lines cut ...]
[    8.352070] RBP: 0000000000407300 R08: 0000000000000000 R09: 0000000000000802
[    8.352694] R10: 0000000000000000 R11: 0000000000000206 R12: 00007fca5627d350
[    8.353316] R13: 0000000000000003 R14: 0000000000405485 R15: 0000000000001818
[    8.353944]  </TASK>
[    8.354264] Modules linked in: rtw89_8852be(O+) rtw89_8852b(O) ath9k_common(O) rtw89_8852b_common(O) qcserial pppoe ppp_async option nft_fib_inet nf_flow_table_inet mt7921e(O) mt7921_common(O) mt76x0e(O) mt76x0_common(O) cdc_mbim ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) wireguard vhost_net vhost usb_wwan sierra rtw89_pci(O) rtw89_core(O) rndis_host qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt792x_lib(O) mt7915e(O) mt76x2e(O) mt76x2_common(O) mt76x02_lib(O)
[... cut ...]
[    8.377104] R10: 0000000000000001 R11: ffffffff81c60d90 R12: 0000000000000015
[    8.377939] R13: 0000000000000719 R14: ffffc9000218fa88 R15: 0000000000000719
[    8.378761] FS:  00007fca5633eb28(0000) GS:ffff88801f500000(0000) knlGS:0000000000000000
[    8.379630] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.380379] CR2: 0000000000001090 CR3: 0000000004b92000 CR4: 00000000000006e0
[    8.381217] Kernel panic - not syncing: Fatal exception
[    8.382081] Kernel Offset: disabled
[    8.382726] ---[ end Kernel panic - not syncing: Fatal exception ]---

"

brada4, 2024-11-22:
Also paste log unchanged, this "privacy enhancement" makes rest of posts useless.
```
**@***.***+0xa000]
``?

Melkior81, 2024-11-22:
> Also paste log unchanged, this "privacy enhancement" makes rest of posts useless.
> 
> ```
> **@***.***+0xa000]
> ``?
> ```

All information was unchanged. Only github has hidden it - I added "<code>" above also I added below again to be sure.

brada4, 2024-11-22:
```
modinfo rtw89_pci 
...
parm:           disable_clkreq:Set Y to disable PCI clkreq support (bool)
parm:           disable_aspm_l1:Set Y to disable PCI ASPM L1 support (bool)
parm:           disable_aspm_l1ss:Set Y to disable PCI L1SS support (bool)
```

Set all three parameters to Y using new line in /etc/modules.conf
```
options rtw89_pci disable_clkreq=Y disable_aspm_l1=Y disable_aspm_l1ss=Y
```
That will disable PCIE power save options just in case they confuse host.
Any better?

Melkior81, 2024-11-22:
No change:
<code>
root@OpenWrt:~# modinfo rtw89_pci
filename:       /lib/modules/6.6.60/rtw89_pci.ko
license:        Dual BSD/GPL
depends:        rtw89_core,mac80211
retpoline:      Y
name:           rtw89_pci
vermagic:       6.6.60 SMP mod_unload 
parm:           disable_clkreq (bool)
parm:           disable_aspm_l1 (bool)
parm:           disable_aspm_l1ss (bool)

[    8.127672] rtw89_8852be 0000:01:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    8.129927] BUG: unable to handle page fault for address: 0000000000001090
[    8.130642] #PF: supervisor write access in kernel mode
[    8.131226] #PF: error_code(0x0002) - not-present page
[    8.131804] PGD 800000000396d067 P4D 800000000396d067 PUD 4ad5067 PMD 0 
[    8.132510] Oops: 0002 [#1] SMP PTI
[    8.132994] CPU: 0 PID: 1919 Comm: kmodloader Tainted: G           O       6.6.60 #0
[    8.133758] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[    8.134712] RIP: 0010:0xffffffffa0ba5b09 [rtw89_pci@000000006c935d05+0xa000]
[    8.135441] Code: 90 89 f6 48 03 b7 a8 3d 07 00 88 16 c3 cc cc cc cc 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 89 f6 48 03 b7 a8 3d 07 00 <66> 89 16 c3 cc cc cc cc 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40
[    8.137112] RSP: 0018:ffffc9000220fa10 EFLAGS: 00010206
[    8.137755] RAX: ffffffffa0ba5b00 RBX: ffff88800d102020 RCX: 0000000000000000
[    8.138483] RDX: 0000000000000718 RSI: 0000000000001090 RDI: ffff88800d102020
[    8.139229] RBP: ffffc9000220fa40 R08: ffffc9000220f9ec R09: ffffc9000220f9ec
[    8.139956] R10: 0000000000000001 R11: ffffffff81c60d90 R12: 0000000000000015
[    8.140686] R13: 0000000000000719 R14: ffffc9000220fa88 R15: 0000000000000719
[    8.141419] FS:  00007f012736ab28(0000) GS:ffff88801f400000(0000) knlGS:0000000000000000
[    8.142202] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.142857] CR2: 0000000000001090 CR3: 00000000050b6000 CR4: 00000000000006f0
[    8.143591] Call Trace:
[    8.144032]  <TASK>
[    8.144453]  ? 0xffffffff8103a230
[    8.144940]  ? 0xffffffff8103a30f
[    8.145426]  ? 0xffffffff81071efc
[    8.145905]  ? 0xffffffffa0ba5b09 [rtw89_pci@000000006c935d05+0xa000]
[    8.146574]  ? 0xffffffff81197d6d
[    8.147052]  ? 0xffffffffa0ba5b09 [rtw89_pci@000000006c935d05+0xa000]
[    8.147748]  ? 0xffffffff810f9b7b
[    8.148227]  ? 0xffffffff81073152
[    8.148708]  ? 0xffffffff8107224a
[    8.149179]  ? 0xffffffff810723c1
[    8.149649]  ? 0xffffffff8125d210
[    8.150117]  ? 0xffffffff8107250e
[    8.150589]  ? 0xffffffff81c92631
[    8.151055]  ? 0xffffffff81e00c27
[    8.151593]  ? 0xffffffff81c60d90
[    8.152059]  ? 0xffffffffa0ba5b00 [rtw89_pci@000000006c935d05+0xa000]
[    8.152722]  ? 0xffffffffa0ba5b09 [rtw89_pci@000000006c935d05+0xa000]
[    8.153375]  ? 0xffffffffa0ba621c [rtw89_pci@000000006c935d05+0xa000]
[    8.154028]  0xffffffffa0baa63d [rtw89_pci@000000006c935d05+0xa000]
[    8.154695]  0xffffffff814d29e3
[    8.155146]  0xffffffff8183020b
[    8.155615]  0xffffffff81830423
[    8.156055]  0xffffffff818304ff
[    8.156495]  0xffffffff818306d6
[    8.156926]  ? 0xffffffff81830650
[    8.157390]  0xffffffff8182dea5
[    8.157818]  0xffffffff8182fa29
[    8.158247]  0xffffffff8182f1d5
[    8.158678]  ? 0xffffffffa0aac010 [rtw89_8852be@0000000027d4d75a+0x1000]
[    8.159317]  0xffffffff81831699
[    8.159736]  ? 0xffffffffa0aac010 [rtw89_8852be@0000000027d4d75a+0x1000]
[    8.160390]  0xffffffff814d102d
[    8.160820]  0xffffffffa0aac033 [rtw89_8852be@0000000027d4d75a+0x1000]
[    8.161470]  0xffffffff81000815
[    8.161900]  ? 0xffffffff81289fbe
[    8.162341]  0xffffffff81156673
[... 11 lines cut ...]
[    8.168733] RDX: 0000000000405485 RSI: 0000000000001818 RDI: 00007f01272a94b0
[    8.169375] RBP: 0000000000407300 R08: 0000000000000000 R09: 0000000000000802
[    8.170017] R10: 0000000000000000 R11: 0000000000000202 R12: 00007f01272a94b0
[    8.170669] R13: 0000000000000003 R14: 0000000000405485 R15: 0000000000001818
[    8.171320]  </TASK>
[    8.171653] Modules linked in: rtw89_8852be(O+) rtw89_8852b(O) ath9k_common(O) rtw89_8852b_common(O) qcserial pppoe ppp_async option nft_fib_inet nf_flow_table_inet mt7921e(O) mt7921_common(O) mt76x0e(O) mt76x0_common(O) cdc_mbim ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) wireguard vhost_net vhost usb_wwan sierra rtw89_pci(O) rtw89_core(O) rndis_host qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt792x_lib(O) mt7915e(O) mt76x2e(O) mt76x2_common(O) mt76x02_lib(O)
[    8.171699]  mt7615e(O) mt7615_common(O) mt7603e(O) mt76_connac_lib(O) mt76(O) mac80211(O) lzo libchacha20poly1305 huawei_cdc_ncm curve25519_x86_64 chacha_x86_64 cfg80211(O) cdc_ncm cdc_ether ax88179_178a asix vhost_iotlb usbserial usbnet usblp ts_kmp ts_fsm ts_bm slhc sky2 r8169 r8152 poly1305_x86_64 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcurv [... cut ...]
[    8.195940] R10: 0000000000000001 R11: ffffffff81c60d90 R12: 0000000000000015
[    8.196807] R13: 0000000000000719 R14: ffffc9000220fa88 R15: 0000000000000719
[    8.197695] FS:  00007f012736ab28(0000) GS:ffff88801f400000(0000) knlGS:0000000000000000
[    8.198638] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.199417] CR2: 0000000000001090 CR3: 00000000050b6000 CR4: 00000000000006f0
[    8.200316] Kernel panic - not syncing: Fatal exception
[    8.201117] Kernel Offset: disabled
[    8.201778] ---[ end Kernel panic - not syncing: Fatal exception ]---
</code>

brada4, 2024-11-22:
Check a bit newer/older firmware file?
https://gitlab.com/kernel-firmware/linux-firmware/-/commits/main/rtw89/rtw8852b_fw-1.bin?ref_type=heads
Compare boot messages around pci slot to debian?

Melkior81, 2024-11-23:
Tried latest, previous firmware, oldest one and the one from debian and some older previous format (according to this https://bugzilla.kernel.org/show_bug.cgi?id=217207). All cause the same error. 
I think it's the kernel module itself. 
Additional information: When no card present - modules load without issue.

brada4, 2024-11-23:
Does the card work on bare metal?

Melkior81, 2024-11-23:
No. It's completely broken. 
I tried to install 8852be module, and it complained about missing dependencies.
After apk install pciutils - restart
After trying to add missing cfg80211 - restart

Hepi34, 2024-11-30:
I have the same issue. It also just crashes and reboots after the firmare load line.

chenms-x, 2025-04-14:
Any update for this, I got the same issue.

Melkior81, 2025-04-19:
I gave up. Someone from openwrt broke the driver and noone wants to fix it. Above they want logs with driver gebugging enabled. I'm not a kernel developer, but maybe if this above CONFIG_DEBUG_INFO is set then the logs will contain function names they ask for and maybe then someone can fix it.

dubhater, 2025-04-19:
@Melkior81 Do you know where the modified source code is? I might recognise the problem if I see it.

chenms-x, 2025-04-20:
[ 1006.135108] rtw89_8852be 0000:05:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[ 1006.142646] rtw89_8852be 0000:05:00.0: enabling device (0000 -> 0003)
[ 1006.149419] BUG: unable to handle page fault for address: 0000000000001090
[ 1006.156347] #PF: supervisor write access in kernel mode
[ 1006.161606] #PF: error_code(0x0002) - not-present page
[ 1006.166775] PGD 1064ef067 P4D 1064ef067 PUD 107f49067 PMD 0
[ 1006.172469] Oops: 0002 [#1] SMP NOPTI
[ 1006.176171] CPU: 2 PID: 9275 Comm: insmod Tainted: G S      W  O       6.6.73 #0
[ 1006.183612] Hardware name: Default string Default string/Default string, BIOS GF1264NP124LV11R007 11/20/2023
[ 1006.193483] RIP: 0010:rtw89_pci_disable_intr_v2+0x169/0x1a0 [rtw89_pci]
[ 1006.200142] Code: 90 89 f6 48 03 b7 90 3e 07 00 88 16 c3 cc cc cc cc 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 89 f6 48 03 b7 90 3e 07 00 <66> 89 16 c3 cc cc cc cc 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40
[ 1006.218974] RSP: 0018:ffffc90000c27a00 EFLAGS: 00010206
[ 1006.224233] RAX: ffffffffa09639c0 RBX: ffff888108382020 RCX: 0000000000000000
[ 1006.231405] RDX: 0000000000000718 RSI: 0000000000001090 RDI: ffff888108382020
[ 1006.238575] RBP: ffffc90000c27a30 R08: ffffc90000c279dc R09: ffffc90000c279dc
[ 1006.245753] R10: 0000000000000005 R11: ffffffff81a51620 R12: 0000000000000015
[ 1006.252923] R13: 0000000000000719 R14: ffffc90000c27a78 R15: 0000000000000719
[ 1006.260101] FS:  00007f21f200eb28(0000) GS:ffff888277b00000(0000) knlGS:0000000000000000
[ 1006.268227] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1006.274010] CR2: 0000000000001090 CR3: 000000010804c000 CR4: 0000000000350ee0
[ 1006.281188] Call Trace:
[ 1006.283672]  <TASK>
[ 1006.285802]  ? show_regs+0x60/0x70
[ 1006.289247]  ? __die+0x1f/0x70
[ 1006.292338]  ? page_fault_oops+0x14c/0x430
[ 1006.296466]  ? mp_map_gsi_to_irq+0x99/0xc0
[ 1006.300601]  ? exc_page_fault+0x3e3/0x620
[ 1006.304653]  ? asm_exc_page_fault+0x27/0x30
[ 1006.308876]  ? __pfx_pci_conf1_write+0x10/0x10
[ 1006.313355]  ? rtw89_pci_disable_intr_v2+0x160/0x1a0 [rtw89_pci]
[ 1006.319404]  ? rtw89_pci_disable_intr_v2+0x169/0x1a0 [rtw89_pci]
[ 1006.325446]  ? rtw89_pci_fill_txaddr_info_v1+0x419/0x780 [rtw89_pci]
[ 1006.331841]  rtw89_pci_probe+0xb3a/0xce0 [rtw89_pci]
[ 1006.336842]  pci_device_probe+0xa0/0x160
[ 1006.340798]  really_probe+0x168/0x300
[ 1006.344499]  __driver_probe_device+0x73/0x120
[ 1006.348894]  driver_probe_device+0x1f/0x90
[ 1006.353027]  __driver_attach+0x86/0x140
[ 1006.356894]  ? __pfx___driver_attach+0x10/0x10
[ 1006.361372]  bus_for_each_dev+0x62/0xa0
[ 1006.365245]  driver_attach+0x19/0x20
[ 1006.368861]  bus_add_driver+0xe5/0x210
[ 1006.372652]  ? __pfx_init_module+0x10/0x10 [rtw89_8852be]
[ 1006.378093]  driver_register+0x59/0x110
[ 1006.381964]  ? __pfx_init_module+0x10/0x10 [rtw89_8852be]
[ 1006.387404]  __pci_register_driver+0x5d/0x70
[ 1006.391709]  init_module+0x23/0xff0 [rtw89_8852be]
[ 1006.396535]  do_one_initcall+0x6f/0x2e0
[ 1006.400406]  ? __kmem_cache_alloc_node+0x16e/0x250
[ 1006.405233]  do_init_module+0x63/0x210
[ 1006.409015]  load_module+0x18e9/0x1aa0
[ 1006.412798]  __do_sys_init_module+0xf1/0x130
[ 1006.417105]  ? __do_sys_init_module+0xf1/0x130
[ 1006.421583]  __x64_sys_init_module+0x15/0x20
[ 1006.425889]  x64_sys_call+0x11aa/0x1c90
[ 1006.429754]  do_syscall_64+0x3d/0x90
[ 1006.433363]  entry_SYSCALL_64_after_hwframe+0x78/0xe2
[ 1006.438456] RIP: 0033:0x7f21f1fc6968
[ 1006.442072] Code: 4c 89 4c 24 48 4c 8b 4c 24 60 48 89 54 24 10 48 8b 7e 08 48 89 74 24 18 48 8b 51 18 c7 44 24 08 28 00 00 00 48 8b 76 10 0f 05 <48> 89 c7 e8 ab 5d fe ff 48 83 c4 58 c3 41 57 41 56 41 55 41 54 55
[ 1006.460898] RSP: 002b:00007ffd71b75cd0 EFLAGS: 00000202 ORIG_RAX: 00000000000000af
[ 1006.468505] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007f21f1fc6968
[ 1006.475683] RDX: 00007f21f1f5f410 RSI: 0000000000001878 RDI: 00007f21f1f5d070
[ 1006.482854] RBP: 00007ffd71b76ebc R08: 0000000000000000 R09: 0000000000010302
[ 1006.490033] R10: 0000000000000000 R11: 0000000000000202 R12: 00007f21f1f5d070
[ 1006.497210] R13: 0000000000000003 R14: 00007f21f1f5f410 R15: 0000000000001878
[ 1006.504390]  </TASK>
[ 1006.506603] Modules linked in: rtw89_8852be(O+) rtw89_8852b(O) rtw89_8852b_common(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet i915 video rtw89_pci(O) rtw89_core(O) pppox ppp_generic nft_tproxy nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mac80211(O) cfg80211(O) wmi slhc r8169 nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c forcedeth e1000e drm_display_helper drm_buddy crc_ccitt compat(O) bnx2 i2c_dev dwmac_intel dwmac_generic stmmac_platform stmmac ixgbe e1000 amd_xgbe mdio_devres mdio tun raid10 raid1 raid0 md_mod nls_utf8 pcs_xpcs ena sha512_ssse3 sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac nls_iso8859_1 nls_cp437 igb igc vfat fat button_hotplug(O) tg3 realtek phylink mii libphy
[ 1006.596792] CR2: 0000000000001090
[ 1006.600146] ---[ end trace 0000000000000000 ]---
[ 1006.604797] RIP: 0010:rtw89_pci_disable_intr_v2+0x169/0x1a0 [rtw89_pci]
[... cut ...]
[ 1006.657057] R10: 0000000000000005 R11: ffffffff81a51620 R12: 0000000000000015
[ 1006.664236] R13: 0000000000000719 R14: ffffc90000c27a78 R15: 0000000000000719
[ 1006.671413] FS:  00007f21f200eb28(0000) GS:ffff888277b00000(0000) knlGS:0000000000000000
[ 1006.679546] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1006.685330] CR2: 0000000000001090 CR3: 000000010804c000 CR4: 0000000000350ee0
[ 1006.692507] Kernel panic - not syncing: Fatal exception
[ 1006.697805] Kernel Offset: disabled
[ 1006.701333] Rebooting in 3 seconds..

chenms-x, 2025-04-20:
Got this log for crash, hope can help you fix it.

dubhater, 2025-04-20:
@chenms-x That is helpful. What does this command say?
```
addr2line -e rtw89_pci.ko -i rtw89_pci_probe+0xb3a
```

chenms-x, 2025-04-20:
[image]

[image]

[image]
I added some logs for it, it crashed in rtw89_dbi_read8
backports-6.12.6/drivers/net/wireless/realtek/rtw89/pci.c
kernel also have drivers, which one should be used?

chenms-x, 2025-04-20:
[image]

dubhater, 2025-04-20:
Aha, I see the problem. rtw89_pci_cfg_dac() ends up calling rtw89_pci_ops_write16(), so it can't be called before rtwpci->mmap is initialised.

It looks like the problem was introduced in this commit in 6.11: https://github.com/torvalds/linux/commit/1fd4b3fe52efd5ad1647966f619c10988e7a4457

It's still there in the latest upstream code.

dubhater, 2025-04-20:
By the way, you can paste something like ``pr_err("%s %d\n", __func__, __LINE__);`` in any function to get this sort of debugging messages more easily.

chenms-x, 2025-04-20:
OK， thanks. So will it be fixed?

chenms-x, 2025-04-20:
[image]
I add this code. Now it will not crash, but I didn't see the device name with ifconfig -a

dubhater, 2025-04-20:
It will be fixed if someone reports the problem to the maintainer at the linux-wireless mailing list.

Melkior81, 2025-04-20:
Hmm, but the driver worked in literally any other linux that I tried.

dubhater, 2025-04-21:
@Melkior81 The code that crashes is called conditionally:

```C
static int rtw89_pci_read_config_byte(struct rtw89_dev *rtwdev, u16 addr,
				      u8 *value)
{
	struct rtw89_pci *rtwpci = (struct rtw89_pci *)rtwdev->priv;
	enum rtw89_core_chip_id chip_id = rtwdev->chip->chip_id;
	struct pci_dev *pdev = rtwpci->pdev;
	int ret;

	ret = pci_read_config_byte(pdev, addr, value);
	if (!ret)
		return 0;

	if (chip_id == RTL8852A || rtw89_is_rtl885xb(rtwdev))
		ret = rtw89_dbi_read8(rtwdev, addr, value);           // crash

	return ret;
}
```

First it tries pci_read_config_byte(). If that doesn't work, then it tries an alternative method to achieve the same thing, with rtw89_dbi_read8(). Maybe something in the Openwrt kernel configuration makes pci_read_config_byte() fail.

pkshih, 2025-05-06:
Hi

I made a patch [1]. Please try if this helps your case.

[1] https://lore.kernel.org/linux-wireless/20250506015356.7995-1-pkshih@realtek.com/T/#u

closed 2025-05-21 by nbd168