openwrt/openwrt #17028: MT7615d is made error for openwrt-23.05.5

azx123123, 2024-11-21:
### Describe the bug

[image]

### OpenWrt version

23.05.5

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

phicomm k2p

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-21:
Are you compiling OpenWRT? Looks like you stuffed mediatek proprietary driver on top of OpenWRT?

castiel652, 2024-11-21:
> Are you compiling OpenWRT? Looks like you stuffed mediatek proprietary driver on top of OpenWRT?

Yes he did.
This can be closed now.

azx123123, 2024-11-21:
yes.I use the mt7615d from the lede's repository.I copy the mt folder to the source.But it's ok to be compled with lede's source.

LGA1150, 2024-11-22:
This package is not maintained by OpenWrt. Please complain to the package maintainer, not here.

closed 2024-11-24 by azx123123