openwrt/openwrt #17029: make openwrt-X86/64 local,and then install openvpn in webpage，but run err call trace-No experience, need help

kite896, 2024-11-21:
### Describe the bug

i make openwrt successful,with checkout tag 25.05.4.also  try 25.05.5 and same err.
but when i download Official image 25.05.4 and run in vmware,no err.
so i still try my local img both in vaware,and i find it always hapen in local,make in ubuntu 24.04 LTS
install easyrsa and openvpn,and then run easyrsa and openvpn,but call trace:

[  102.760626] BUG: kernel NULL pointer dereference, address: 0000000000000028
[  102.762610] #PF: supervisor read access in kernel mode
[  102.764078] #PF: error_code(0x0000) - not-present page
[  102.765585] PGD 8000000005dba067 P4D 8000000005dba067 PUD 6184067 PMD 0 
[  102.767511] Oops: 0000 [#1] SMP PTI
[  102.768535] CPU: 1 PID: 3378 Comm: openvpn Not tainted 5.15.167 #0
[  102.770329] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
[  102.773376] RIP: 0010:_10+0x52d/0x69c0 [tun]
[  102.774673] Code: c0 48 c7 c1 40 71 0d a0 ba c0 0c 00 00 48 89 e5 41 55 65 4c 8b 2c 25 c0 bb 01 00 41 54 49 89 f4 31 f6 53 49 8b 85 f8 06 00 00 <48> 8b 78 28 e8 aa ef 7c e1 48 85 c0 0f 84 17 01 00 00 ba ff ff ff
[  102.780099] RSP: 0018:ffffc900027bbc28 EFLAGS: 00010246
[  102.781689] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffffffffa00d7140
[  102.783835] RDX: 0000000000000cc0 RSI: 0000000000000000 RDI: ffff8880053fea98
[  102.785922] RBP: ffffc900027bbc40 R08: 0000000000000000 R09: ffff8880031a9000
[  102.787987] R10: 90ffffffff918a8b R11: 0000000000000000 R12: ffff88800626ee00
[  102.790065] R13: ffff8880055b4740 R14: ffff8880053fea98 R15: ffffffffa00d55c0
[  102.792118] FS:  00007fcde1db1b48(0000) GS:ffff88802d240000(0000) knlGS:0000000000000000
[  102.794484] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  102.796216] CR2: 0000000000000028 CR3: 00000000063c4003 CR4: 00000000003706e0
[  102.798528] Call Trace:
[  102.799261]  <TASK>
[  102.799937]  ? show_regs.part.0+0x1e/0x24
[  102.801122]  ? __die+0x55/0x99
[  102.802040]  ? page_fault_oops+0x9b/0x230
[  102.803250]  ? try_to_wake_up+0x171/0x250
[  102.804454]  ? exc_page_fault+0x403/0x570
[  102.805622]  ? ep_autoremove_wake_function+0xd/0x40
[  102.807023]  ? asm_exc_page_fault+0x27/0x30
[  102.808226]  ? _10+0x52d/0x69c0 [tun]
[  102.809293]  ? kobj_lookup+0xf1/0x160
[  102.810359]  misc_open+0x11c/0x160
[  102.811383]  chrdev_open+0xdd/0x200
[  102.812395]  ? cdev_dynamic_release+0x80/0x80
[  102.813695]  do_dentry_open+0x1d9/0x3d0
[  102.814905]  vfs_open+0x28/0x30
[  102.815821]  path_openat+0xcf1/0x1000
[  102.816899]  do_filp_open+0x9c/0x130
[  102.817938]  ? alloc_fd+0x54/0x180
[  102.818926]  do_sys_openat2+0x80/0x140
[  102.820013]  __x64_sys_open+0x5c/0x70
[  102.821075]  x64_sys_call+0xf08/0x1ae0
[  102.822150]  do_syscall_64+0x3c/0x90
[  102.823188]  entry_SYSCALL_64_after_hwframe+0x6c/0xd6
[  102.824625] RIP: 0033:0x7fcde1d90399
[  102.825669] Code: c3 8b 07 85 c0 75 24 49 89 fb 48 89 f0 48 89 d7 48 89 ce 4c 89 c2 4d 89 ca 4c 8b 44 24 08 4c 8b 4c 24 10 4c 89 5c 24 08 0f 05 <c3> e9 c8 cf ff ff 41 54 b8 02 00 00 00 55 48 89 f5 be 00 88 08 00
[  102.831277] RSP: 002b:00007ffc629671a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000002
[  102.833421] RAX: ffffffffffffffda RBX: 0000000000000002 RCX: 00007fcde1d90399
[  102.835419] RDX: 0000000000000000 RSI: 0000000000008002 RDI: 000000000046351a
[  102.837414] RBP: 00007fcde1db1b48 R08: 0000000000000000 R09: 0000000000000000
[  102.839371] R10: 0000000000000000 R11: 0000000000000246 R12: 00007fcde180c158
[  102.841359] R13: 00007fcde17ec928 R14: 00007fcde1803400 R15: 0000000000000000
[  102.843501]  </TASK>
[  102.844190] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c igc forcedeth e1000e crc_ccitt bnx2 i2c_dev ixgbe e1000 amd_xgbe mdio_devres mdio tun nls_utf8 ena nls_iso8859_1 nls_cp437 igb vfat fat button_hotplug tg3 realtek libphy
[  102.861074] CR2: 0000000000000028
[  102.862020] ---[ end trace 90becc92b950a37f ]---
[  102.863328] RIP: 0010:_10+0x52d/0x69c0 [tun]
[  102.864725] Code: c0 48 c7 c1 40 71 0d a0 ba c0 0c 00 00 48 89 e5 41 55 65 4c 8b 2c 25 c0 bb 01 00 41 54 49 89 f4 31 f6 53 49 8b 85 f8 06 00 00 <48> 8b 78 28 e8 aa ef 7c e1 48 85 c0 0f 84 17 01 00 00 ba ff ff ff
[  102.869894] RSP: 0018:ffffc900027bbc28 EFLAGS: 00010246
[  102.871377] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffffffffa00d7140
[  102.873509] RDX: 0000000000000cc0 RSI: 0000000000000000 RDI: ffff8880053fea98
[  102.875584] RBP: ffffc900027bbc40 R08: 0000000000000000 R09: ffff8880031a9000
[  102.877669] R10: 90ffffffff918a8b R11: 0000000000000000 R12: ffff88800626ee00
[  102.879825] R13: ffff8880055b4740 R14: ffff8880053fea98 R15: ffffffffa00d55c0
[  102.882322] FS:  00007fcde1db1b48(0000) GS:ffff88802d240000(0000) knlGS:0000000000000000
[... 183 lines cut ...]
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_KERNEL_PARTSIZE=120
CONFIG_TARGET_ROOTFS_PARTSIZE=900
CONFIG_USE_FS_ACL_ATTR=y
CONFIG_VMDK_IMAGES=y
CONFIG_PACKAGE_libmbedtls=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kite896, 2024-11-21:
> Invalid Target/Subtarget reported. `X86/64`
> Is this from a supported device?
 ofcurse

ynezz, 2024-11-21:
> so i want to know what difference between official and self-build
You can compare the diff configs, official one  https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/config.buildinfo

kite896, 2024-11-23:
> > so i want to know what difference between official and self-build
> 
> You can compare the diff configs, official one https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/config.buildinfo

thank you very much! I diff configs,and find so many different,i try config one of them seems will solve this problem ：“CONFIG_ALL_KMODS=y”
But unfortunately the error was not solved.Then I gave up and directly copied the official config and build again.
......ps:build llvmtakes a lot of time
And then,The problem was solved and no more errors were reported or reboot！openvpn started.
So it should be a problem with the configuration file. should I close this issue？

kite896, 2024-12-03:
renew:I checkout to 24.10-rc1 tag,then make and repeat the same steps.
But I found that when opkg installing openvpn, it prompted that kmod-tun could not be found.
So I rebuild kmod-tun into the package, and openvpn can start normally.

closed 2024-12-03 by kite896