openwrt/openwrt #17030: EAPOL-Key timeout causing packet loss when moving clients between bands

jmencak, 2024-11-21:
### Describe the bug

During roaming between 2.4GHz and 5GHz on the same SSID (no ieee80211r/ieee80211k/ieee80211v enabled), I'm seeing cca 2 second lasting wifi connectivity dropouts with logs on my OpenWRT 24.10 AP from hostapd `WPA: EAPOL-Key timeout` causing packet loss.  This is 100% reproducible, but only in one direction.  When it works well in one direction (e.g. 2.4GHz -> 5GHz) with no timeout, there are timeouts in the other direction (e.g. 5GHz -> 2.4GHz).

### OpenWrt version

r28023-fe048b5e87

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

I'm not using luci at all.  All configuration from the command line.  Here is my `/etc/config/wireless`.  I tried to keep it as simple as possible, not even using `ax`.

1) Use the following config

```

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel '10'
	option band '2g'
	option country 'CZ'
	option txpower '20'
	option htmode 'HT40'
	option log_level '1'
	option disabled '0'

config wifi-iface 'ap_g'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'openwrt'
	option encryption 'psk2+ccmp'
	option key 'my-secret-pass'
	option ieee80211w '1'
	option disabled '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option channel '36'
	option band '5g'
	option country 'CZ'
	option txpower '23'
	option htmode 'VHT80'
	option log_level '1'
	option disabled '0'

config wifi-iface
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'openwrt'
	option encryption 'psk2+ccmp'
	option key 'my-secret-pass'
	option ieee80211w '1'
	option disabled '0'
```

2) Connect your wireless client to `openwrt` SSID.

3) Move closer to the AP to get associated to 5GHz.

4) Move further away from the AP, to get connected to 2.4GHz.

5) If you now do not see any hostapd `WPA: EAPOL-Key timeout` messages in the hostapd logs, move back to the AP to get connected to the 5GHz band again.  After that, check the logs and you'll see hostapd `WPA: EAPOL-Key timeout` messages.

### Actual behaviour

```
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: authentication OK (open system)
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-AUTHENTICATE.indication(aa:aa:aa:aa:aa:aa, OPEN_SYSTEM)
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DELETEKEYS.request(aa:aa:aa:aa:aa:aa)
Thu Nov 21 11:28:27 2024 daemon.info hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: authenticated
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: association OK (aid 1)
Thu Nov 21 11:28:27 2024 daemon.info hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: associated (aid 1)
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-REASSOCIATE.indication(aa:aa:aa:aa:aa:aa)
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DELETEKEYS.request(aa:aa:aa:aa:aa:aa)
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: binding station to interface 'phy1-ap0'
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: event 1 notification
Thu Nov 21 11:28:27 2024 daemon.notice hostapd: phy0-ap0: Prune association for aa:aa:aa:aa:aa:aa
Thu Nov 21 11:28:27 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED aa:aa:aa:aa:aa:aa
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: start authentication
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.1X: unauthorizing port
Thu Nov 21 11:28:27 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: sending 1/4 msg of 4-Way Handshake
Thu Nov 21 11:28:28 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: EAPOL-Key timeout
Thu Nov 21 11:28:28 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: sending 1/4 msg of 4-Way Handshake
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy0-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DISASSOCIATE.indication(aa:aa:aa:aa:aa:aa, 1)
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy0-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DELETEKEYS.request(aa:aa:aa:aa:aa:aa)
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: EAPOL-Key timeout
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: sending 1/4 msg of 4-Way Handshake
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: received EAPOL-Key frame (2/4 Pairwise)
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: sending 3/4 msg of 4-Way Handshake
Thu Nov 21 11:28:29 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu Nov 21 11:28:30 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED aa:aa:aa:aa:aa:aa auth_alg=open
Thu Nov 21 11:28:30 2024 daemon.debug hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.1X: authorizing port
Thu Nov 21 11:28:30 2024 daemon.info hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa RADIUS: starting accounting session 575CA03914BA578B
Thu Nov 21 11:28:30 2024 daemon.info hostapd: phy1-ap0: STA aa:aa:aa:aa:aa:aa WPA: pairwise key handshake completed (RSN)
Thu Nov 21 11:28:30 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED aa:aa:aa:aa:aa:aa

```

### Expected behaviour

No major packet loss (2s disconnnects) and `WPA: EAPOL-Key timeout` messages.

### Additional info

- I have an old Archer C5 router and the same configuration works just fine with no timeouts on exactly the same OpenWRT source snapshot.
- I've tried many different configurations, hostapd versions and types (the default one, openssl based) and nothing seems to help.  I suspect a kernel driver issue, but I haven't tried debugging that and probably don't have the skills to even try that.

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-29:
2s is very reasonable and typical authentication time.

jmencak, 2024-11-29:
> 2s is very reasonable and typical authentication time.

Thank you for your response.  However, 2s is far too much and unusable for audio/video calls.  With Archer C5 I have never seen latency higher than 100ms when going from 2GHz to 5GHz and back and more importantly, no `WPA: EAPOL-Key timeout` messages in the logs.

brada4, 2024-11-29:
STA does not respond to (or receive) first two challenge attempts. It kind of goes fast once 3rd is responded.

jmencak, 2024-11-29:
I've seen the same issue even without using any authentication (no WPA/WEP/...). I don't have the logs for that scenario, but I can provide them.  Also, this happens for any STA.  In the logs, you see a Google Pixel 5 Android phone, but I've reproduced this also with my Fedora 40 Linux laptop.  So I'm pretty sure this is an issue with the AP.

brada4, 2024-11-29:
You can try to see what is sent/received with debug level 0 (enormous volume with hexdump text of every packet) , or set monitor device (wifi card in monitor mode) almost between 2 devices and again check if all retransmits are +/- same.
Or if you have similar AP at hand try with same setup, it should be ok kicking (disconnecting) client from one AP and waiting for it to get to another.

jmencak, 2024-11-29:
I should probably change the title of this issue, because I do not see the `EAPOL-Key timeouts` with `encryption=none`, but the huge 2-3s packet loss is nevertheless still present.  I did two experiments now and disabled authentication completely.  I used two routers.  The "misbehaving" BPI-R3 and Archer C5 router which doesn't have any packet loss and I never see higher latency than 100ms during 5GHz -> 2.4GHz (and back) transitions.  In both cases, I used my Fedora 40 laptop as STA.  I started connected to the 5GHz AP and then moved further away from it to switch to 2.4GHz AP.

### Archer C5 AC1200 v1.2

#### Configuration

```
# cat /etc/config/wireless
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:00.0'
	option band '5g'
	option channel '40'
	option country 'CZ'
	option txpower '18'
	option htmode 'VHT80'
	option disabled '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'owrt'
	option encryption 'none'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/ahb/18100000.wmac'
	option band '2g'
	option country 'CZ'
	option txpower '20'
	option htmode 'HT20'
	option disabled '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'owrt'
	option encryption 'none'

# iwinfo
phy0-ap0  ESSID: "owrt"
          Access Point: 30:B5:C2:7F:43:69
          Mode: Master  Channel: 40 (5.200 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 18 dBm  Link Quality: 63/70
          Signal: -47 dBm  Noise: -103 dBm
          Bit Rate: 650.0 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 168C:003C 0000:0000 [Qualcomm Atheros QCA9880]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "owrt"
          Access Point: 30:B5:C2:7F:43:6A
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -89 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros QCA9550]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

[... 286 lines cut ...]
Fri Nov 29 17:00:05 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:05 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:10 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:10 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:15 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:15 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:16 2024 daemon.info hostapd: phy2-ap0: STA aa:aa:aa:aa:aa:aa IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Nov 29 17:00:16 2024 daemon.debug hostapd: phy2-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DEAUTHENTICATE.indication(aa:aa:aa:aa:aa:aa, 2)
Fri Nov 29 17:00:16 2024 daemon.debug hostapd: phy2-ap0: STA aa:aa:aa:aa:aa:aa MLME: MLME-DELETEKEYS.request(aa:aa:aa:aa:aa:aa)

< --- STA finally connected after 30s! --- >

Fri Nov 29 17:00:20 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:20 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:25 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:25 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:30 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:30 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:35 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:35 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
Fri Nov 29 17:00:40 2024 daemon.notice procd: /etc/rc.d/S95done: + test -d /sys/class/net/phy0-ap1
Fri Nov 29 17:00:40 2024 daemon.notice procd: /etc/rc.d/S95done: + sleep 5
```

Time for `tcpdump` and wireshark to get some traces now?  Any other ideas what to look for?