openwrt/openwrt #17036: build: kmods missing from package index

efahl, 2024-11-21:
### Describe the bug

In 23.05 and earlier, the `kmod-*` packages are listed in the arch-specific package list, and reflected into its corresponding `index.json`.  For example, see this and filter for `kmod-`

https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/index.json

In 24.10, the kmods are no longer present in the package list or index.json file.  See

https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/x86/64/packages/index.json

This is also the case in `apk`-based main SNAPSHOT, probably a different code path to the same bad result. https://downloads.openwrt.org/snapshots/targets/x86/64/packages/index.json

Also...
The `kmods` directory has it's own separate package list, which has traditionally been partially redundant with the full arch-specific package list.  But, the kmod-specific package list is not included in `feeds.conf`, so package managers won't know about them.

For an example:
  https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/x86/64/kmods/6.6.60-1-cccdb39f40ab0d2ddc0a71725516d8a8/

### OpenWrt version

24.10-SNAPSHOT

### OpenWrt release

r28032-1abaf051d2

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2024-11-21:
@aparcar @Ansuel @robimarko 

I'm currently looking at `scripts/ipkg-make-index.sh` on 23.05 branch, but my superficial glance at it doesn't reveal anything different for 24.10.  Do any of you recall anything that might cause the kmods to not be included?

For the SNAPSHOT, I'm pretty sure it's the `ls` vs `find` in the `CONFIG_USE_APK` code in `package/Makefile`.  I'll see what I can do there.

efahl, 2024-11-21:
Ok, now that I move up the directory tree a bit more, I see it's not the index generation, but the kmods themselves are missing from the packages/ dir.  So there's a cp or mv or something missing in the kmod build step???

In other words:
kmods here: https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages/
no kmods:  https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/x86/64/packages/

robimarko, 2024-11-21:
Kmods are in dedicated kmod dir:
https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/x86/64/kmods/

efahl, 2024-11-21:
Yeah, that separate kmods/*/index list has always been there, and ignored apparently as it was a subset of what was in packages/index

The root issue I'm having is that in the ASU server we have always just grabbed
https://downloads.openwrt.org/snapshots/targets/x86/64/packages/index.json
for all of the arch-specific packages.  This still works fine for 23.05 and earlier, whereas now we also need to grab

https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/<kernel-ver.hash>/index.json

I wonder how I can figure out the name of specific kmod subdir needed to special case it (esp. so as we don't even know the kernel version when we generate the request)?  It would be nice if there were a `targets/x86/64/feeds.conf` equivalent to the one in `packages` to tell us this.

https://downloads.openwrt.org/releases/24.10-SNAPSHOT/packages/x86_64/feeds.conf

efahl, 2024-11-22:
> Maybe you can get the kernel version from the filename of kernel-*.apk or kernel-*.ipk package in the TARGET PACKAGES dir.

We don't know the file name to be able to download it, chicken and egg problem.

The data actually can be found in an extremely convoluted way right now from what's in `version.buildinfo`, then using that to figure out the manifest, then parsing that to get the version~hash-rev, to create the path for the kmods dir.  Here's a mockup in sh, would actually be deployed in Python and require a ton more parsing and logic due to the limited info in the API request, but this was "easy".

```
# Example api requests
#   GET https://sysupgrade.openwrt.org/json/v1/snapshots/targets/x86/64/index.json
#  GET https://sysupgrade.openwrt.org/json/v1/releases/24.10-SNAPSHOT/targets/x86/64/index.json

target="x86/64"
if false; then
    release="24.10-snapshot"
    reldir="releases/24.10-SNAPSHOT"
else
    release="SNAPSHOT"
    reldir="snapshots"
fi
#release="23.05.5"
#reldir="releases/23.05.5"

echo $release

path="https://downloads.openwrt.org/$reldir/targets/$target/"

if [[ "$release" < "24.10" ]]; then  # This comparison is fake, but you get the idea.
    # Just return the index.
    curl -s $path/packages/index.json | wc
else
    # Get the old packages index...
    curl -s $path/packages/index.json | wc

    # Then merge kmods index into it.
    revcode=$(curl -s $path/version.buildinfo)
    targetd=$(echo $target | tr / -)
    if [ "$reldir" = "snapshots" ]; then
        manifest="$path/openwrt-$targetd.manifest"
    else
        manifest="$path/openwrt-${release}-${revcode}-${targetd}.manifest"
    fi

    kver=$(
        curl -s $manifest |
        awk -F'[ -]' '/kernel -/ {
            sub(/r/, "", $5);
            sub(/~/, "-"$5"-", $4);
            print $4
        }'
    )

    curl -s $path/kmods/$kver/index.json | wc
fi
```

efahl, 2024-11-22:
Here's the current ASU server implementation: https://github.com/openwrt/asu/blob/main/asu/main.py#L76

aparcar, 2024-11-22:
@efahl I did not fully follow the process but I think it's related to https://github.com/openwrt/buildbot/commit/ec2bd8a07e88f4e251fccb99142bd46c816d62ec

efahl, 2024-11-22:
Yup, https://github.com/openwrt/buildbot/commit/ec2bd8a07e88f4e251fccb99142bd46c816d62ec is the one, @robimarko pointed to the same one, too.

@dangowrt @ynezz What are the chances of just getting a symlink in the kmods directory from a fixed name like `current` or `latest` or something, updating when that build-specific directory is created...

At each of `https://downloads.blah.blah/targets/x86/64/`:

`ln -s  kmods/6.6.60-1-cccdb39f40ab0d2ddc0a71725516d8a8/  kmods/current/`

Then on the ASU server I can just do this:
```
GET upstream_url/.../packages/index.json       # Currently what we're doing.
GET upstream_url/.../kmods/current/index.json  # New addition.
```

Ansuel, 2024-11-22:
Honestly it's easy to do but it would introduce random complications... We
currently ship in packages the Linux kernel package with latest version so
that can be abused currently, parsed and get the Linux version. But
honestly, also this method doesn't sound so robust.

efahl, 2024-11-22:
> Honestly it's easy to do but it would introduce random complications...

Just curious what you're thinking here, I'm not familiar enough with buildbot stuff to see any issues and would like to learn.

How about this, solve it in buildroot rather than buildbot?  I could add another field to the `index.json` file itself (as far as I know, the ASU server is the sole consumer of these, so should be completely benign). I assume the information is floating around somewhere when [`package/Makefile`](https://github.com/openwrt/openwrt/blob/main/package/Makefile#L135) is run...
```
{
    "architecture": "x86_64",
    "kernel": "6.6.60-1-cccdb39f40ab0d2ddc0a71725516d8a8",    <<< New field.
    "packages": {
        "arptables-nft": "1.8.10-r1",
        "atm-aread": "2.5.2-r8",
...
```

closed 2024-11-26 by openwrt-bot