openwrt/openwrt #17038: Failsafe Shell Error: Unable to Obtain Session ID (SID) for Root User in OpenWrt

ZhadowValker, 2024-11-22:
### Describe the bug

When attempting to run a failsafe shell on a console in an OpenWrt environment, 
users encountered the error message: "login: can't get SID for root."

This issue arose after incorporating the following commit: 
[git.openwrt.org Git - openwrt/openwrt.git/commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=1310e4f1aecef6c2c1b68227216ba7cc04006002)

The problem was traced back to modifications in the login.sh script. 
The error message originates from the BusyBox login utility, specifically at line 192 in login.c, 
where it fails to obtain a Session ID (SID) for the root user.

Could you please provide guidance or a solution to resolve this issue?

### OpenWrt version

openwrt24.10 branch

### OpenWrt release

24.10

### OpenWrt target/subtarget

ALL

### Device

ALL

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Vinod067, 2024-11-25:
> Invalid Target/Subtarget reported. `ALL` Is this from a supported device?

Yes we are using a supported device

Vinod067, 2024-11-25:
> Invalid Release reported. `24.10` Is this from a clean repository?

Yes it's clean repository used for test

istvanhenkel, 2025-03-20:
Hi, I ran into the same problem. The following solution worked for me: in make menuconfig in Global build settings - Enable SELinux - default SELinux type - targeted. I hope it helps.

robbins, 2025-09-05:
Also encountering this issue. Can confirm changing the policy to `targeted` from `dssp` allows me to login, but I'd prefer to fix and use the OpenWRT policy.

Somewhat strangely, setting `enforcing=0` doesn't fix the issue, but disabling SELinux entirely does.

Vinod067, 2025-09-05:
If we want to try with selinux enablement ?