openwrt/openwrt #17039: ax23 problem with firware version 23.05.5

TomTom-42, 2024-11-22:
### Describe the bug

Reboot not working with this firmware version

command used
sleep 70 && touch /etc/banner && reboot

when reboot executed, stay blocked with only powered led light on
other led are off
also AP/rooter is not working,  no wifi, no ping on and no ssh accessible.
the only way is to remove the plug some secondes to restart the device.

rollback the device to the 23.05.4  firmware resolve the problem.

### OpenWrt version

23.0.5

### OpenWrt release

23.0.5

### OpenWrt target/subtarget

23.0.5

### Device

AX23

### Image kind

Official downloaded image

### Steps to reproduce

AX23 with firmware 23.0.5
execute command sleep 70 && touch /etc/banner && reboot

### Actual behaviour

Device not responding after reboot command

### Expected behaviour

Device is working well after reboot command

### Additional info

works well with firmware 23.05.4

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

TomTom-42, 2024-11-22:
Sorry I meant version concerned is 23.05.5  not 23.0.5

emixam360, 2025-03-05:
Hello
I am facing the same issue on 24.10.0
Unfortunatelly I can't connect to serial to see what happens during the failed reboot

xMikooo, 2025-05-16:
I also have this error as I use many ax23 routers in my network. The boot process will happen as usual, then hang at power LED and will stay like that forever. I have connected serial connection to one router running snapshot firmware and factory reset (happens at every openwrt release >= 23.05.5). I thought the router might kernel panic or hang at uboot somewhere, but no. It boots normally, you can access the console, the interfaces supposedly have IP addresses. However no ping and any kind of access is possible. Below is the bootlog, I'm not competent enough to troubleshoot further. This issue makes it not practical to use such router in a setup without backup power. In my case i have to reboot the router around 5 times to make it boot properly.
`U-Boot 1.1.3 (Feb 16 2022 - 16:26:34)

Board: Ralink APSoC DRAM:  128 MB
relocate_code Pointer at: 87f58000

Config XHCI 40M PLL
******************************
Software System Reset Occurred
******************************
flash manufacture id: 20, device id 40 18
Warning: un-recognized chip ID, please update bootloader!
*** Warning - bad CRC, using default environment

============================================
Ralink UBoot Version: 5.0.0.0
--------------------------------------------
ASIC MT7621A DualCore (MAC to MT7530 Mode)
DRAM_CONF_FROM: Auto-Detection
DRAM_TYPE: DDR3
DRAM bus: 16 bit
Xtal Mode=3 OCP Ratio=1/3
Flash component: SPI Flash
Date:Feb 16 2022  Time:16:26:34
============================================
THIS IS uboot
icache: sets:256, ways:4, linesz:32 ,total:32768
dcache: sets:256, ways:4, linesz:32 ,total:32768

 ##### The CPU freq = 880 MHZ ####
 estimate memory size =128 Mbytes

Press '4' or 't' to break the booting process

Press 'x' to enter recovery web server                                                                                                                                                     0
nm_init:791
nm_initFwupPtnStruct:276
nm_lib_readPtnTable:738
[NM_Debug](nm_lib_readPtnTable) 00743: NM_PTN_TABLE_BASE = 0xfe0000
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 1054, requried len = 8192
[NM_Debug](nm_lib_readPtnTable) 00751: Reading Partition Table from NVRAM ... OK

[NM_Debug](nm_lib_readPtnTable) 00759: Parsing Partition Table ... OK

[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
factory boot check integer ok.

3: System Boot system code via Flash.
## Booting image at bc040000 ...
   Image Name:   MIPS OpenWrt Linux-6.6.89
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    3403996 Bytes =  3.2 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
[NM_Error](nm_api_checkTestMode) 00417: factory boot check testmode flag is not 1.

### test_mode=disable,boot_args:console=ttyS1,115200 root=/dev/mtdblock2 rootfstype=squashfs init=/etc/preinit mtdparts=raspi:256k(uboot),4096k(uImage),11968k@0x440000(rootfs),64k@0xff0000(ART) mem=128M test_mode=disable
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 6.6.89 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 14.2.0 r29528-ffde9a9fe9) 14.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri May  9 14:38:19 2025
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[... 308 lines cut ...]
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2223:51ff:feee:eaa1/64 scope link
       valid_lft forever preferred_lft forever
3: wan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 20:23:51:ee:ea:a2 brd ff:ff:ff:ff:ff:ff
4: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
6: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
7: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
8: br-lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether 20:23:51:ee:ea:a1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd71:f41c:aa8f::1/60 scope global tentative noprefixroute
       valid_lft forever preferred_lft forever
root@OpenWrt:~#
`