openwrt/openwrt #17041: Duplicate dhcp host configuration hangs startup.

aerobiotic, 2024-11-22:
### Describe the bug

Use the uci command line interface to define static IPs for MAC addresses. If you duplicate an entry, the router won't be reachable for further configuration after a reboot. 

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1200AC

### Image kind

Official downloaded image

### Steps to reproduce

uci add dhcp host
uci set dhcp.@host[-1].name="EQUIP-001"
uci set dhcp.@host[-1].mac="b0:7f:b9:f5:b5:57"
uci set dhcp.@host[-1].ip="192.168.0.173"
uci set dhcp.@host[-1].dns="1"
uci commit dhcp
service dnsmasq restart

uci add dhcp host
uci set dhcp.@host[-1].name="EQUIP-001"
uci set dhcp.@host[-1].mac="B0:7F:B9:F5:B5:57"
uci set dhcp.@host[-1].ip="192.168.0.173"
uci set dhcp.@host[-1].dns="1"
uci commit dhcp
service dnsmasq restart

reboot

### Actual behaviour

The devices that are supposed to get an IP address assigned won't; it's as if DHCP is not reachable. 
I tried to ssh into the router, but it's not reachable.

### Expected behaviour

When a duplicate dhcp array item is detected at start-up, then don't crash or lockup.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-22:
Nothing crashes or locks up. Check logread after second +masq restart.

loopy321, 2024-11-23:
I just had this problem occur on 23.05.2 and was _shocked_ to eventually figure out that dnsmasq fails so _miserably_ when two hosts have the same ip.  There should be some failover process here instead of bringing down the whole network.
This was the extent of the logs:
```
daemon.crit dnsmasq[1]: duplicate dhcp-host IP address x.x.x.x at line 40 of /var/etc/dnsmasq.conf.cfg01411c
daemon.crit dnsmasq[1]: FAILED to start up
```

brada4, 2024-11-23:
It should be dhcp-host=MAC,MAC,MAC,IP , not two entries. The config/dhcp can not reflect that right.

brada4, 2024-11-23:
i.e MAC should be of `list` type and reconstruct multiple MAC-s for an IP
https://openwrt.org/docs/guide-user/base-system/dhcp#static_leases

brada4, 2024-11-23:
It is here. History says code was added 10 years ago, with @hauke and @systemcrash being recently commiting other code nearby.
https://github.com/openwrt/openwrt/blob/52b6c9247997e51a97f13bb9e94749bc34e2d52e/package/network/services/dnsmasq/files/dnsmasq.init#L369

systemcrash, 2024-11-23:
Congratulations on your discovery of the multi Mac single IP syntax, where it very explicitly says in the man page that it’s dangerous:

As a special case, in DHCPv4, it is possible to include more than one hardware address. eg: --dhcp-host=11:22:33:44:55:66,12:34:56:78:90:12,192.168.0.2 This allows an IP address to be associated with multiple hardware addresses, and gives dnsmasq permission to abandon a DHCP lease to one of the hardware addresses when another one asks for a lease. Beware that this is a dangerous thing to do, it will only work reliably if only one of the hardware addresses is active at any time and there is no way for dnsmasq to enforce this. It is, for instance, useful to allocate a stable IP address to a laptop which has both wired and wireless interfaces.

===

I have multiple entries that share the same IP but not MAC. If you’ve any good ideas for a preventive fix, patches are welcome.

aerobiotic, 2024-11-23:
> Nothing crashes or locks up. Check logread after second +masq restart. 

after second +masq restart.  - Is this automatically happening during a timeout? 

Are you saying that if I hit the reset button on the router a couple of times, I can then SSH into it? For me, the router was not reachable by SSH, so no access to logread.

There is no Luci either since I set up the router for shared SSH key login. Then, I made the HTTP server listen on localhost instead of 0.0.0.0. I then used the SSH tunnel to reach the UI. Since there is no SSH, there is no Luci.

brada4, 2024-11-23:
DHCP server is not starting, you can log in from static IP inside LAN

aerobiotic, 2024-11-23:
> DHCP server is not starting, you can log in from static IP inside LAN

@brada4 I see. So, I should have assigned a static IP to the computer I was using to configure the router. It was dependent on DHCP to get an IP address in order to access the LAN. This fooled me into thinking SSH was down, too. Thanks for that. This just leaves the need to harden the DHCP init process from the configuration issue I mentioned above.

aerobiotic, 2024-11-23:
I found a second way to break this; in my original example, the MAC addresses are the same—only the case is lower in one and upper in the other. I stumbled onto this when defining a series of static IPs for my equipment. All other entries used upper-case MAC addresses, but one was represented in lowercase. I assumed that using the same hostname as when I added the lower-case version would overwrite the previous one. But it does not work like that, and I should have done a uci delete and then added the definition again with the uppercase MAC address. 

So, I did the long-press reset on the router to clear the configuration. Since I had made notes and documented my uci commands in a document, it was going to be easy to get going again. 

I was researching what was on my LAN segment and added static host entries for each one. During that process, I copied and pasted a section of the uci commands to save typing when configuring the next device. So, after repairing the configuration from my notes, I broke it again as this was in there: 

`uci add dhcp host
uci set dhcp.@host[-1].name="AMCREST-silver"
uci set dhcp.@host[-1].mac="3C:EF:8C:94:30:D9"
uci set dhcp.@host[-1].ip="192.168.0.172"
uci set dhcp.@host[-1].dns="1"
uci commit dhcp
service dnsmasq restart

uci add dhcp host
uci set dhcp.@host[-1].name="AMCREST-silver"
uci set dhcp.@host[-1].mac="3C:EF:8C:94:30:D9"
uci set dhcp.@host[-1].ip="192.168.0.172"
uci set dhcp.@host[-1].dns="1"
uci commit dhcp
service dnsmasq restart`

I had not gotten around to editing the second AMCREST-silver to AMCREST-black and updating the MAC and IP address, so I had to reconfigure the router for a third time. 

I suspect that the fix might be to do a case-insensitive check on the MAC and skip duplicate definitions. So, the first one wins. For sure, log an error or warning in the logs so that a human can address this bad configuration. 

This should work for the laptop scenario since the MAC address for the LAN and the MAC address for the wireless interface are different.