openwrt/openwrt #17050: [pciutils]ERROR: package/feeds/packages/kmod failed to build.

libiunc, 2024-11-23:
### Describe the bug

When configuring **CONFIG_PACKAGE_pciutils=y** and using single thread, the error is as follows:

make[3]: Entering directory '/media/lib/LuckyE/opn/openwrt/feeds/packages/utils/kmod'
rm -f /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/.built
touch /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " make --jobserver-auth=3,4 -C /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ; make[4]: Entering directory '/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32' make --no-print-directory all-recursive Making all in . Making all in libkmod/docs make[6]: Nothing to be done for 'all'. Making all in man make[6]: Nothing to be done for 'all'. make[4]: Leaving directory '/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32' CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " make -C /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

MediaTek ARM / Filogic 8x0 (MT798x)

### Device

GL.iNet GL-MT3000

### Image kind

Self-built image

### Additional info

gcc 13.3
Binutils 2.43.1
llvm 19(for enable bridger)

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_glinet_gl-mt3000=y
CONFIG_TARGET_DEVICE_PACKAGES_mediatek_filogic_DEVICE_glinet_gl-mt3000=""
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_WEBSOCKET=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH="/usr/bin/clang"
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_DWARVES=y
CONFIG_EXPERIMENTAL=y
# CONFIG_GDB is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_IMAGEOPT=y
CONFIG_KERNEL_ARM_PMU=y
CONFIG_KERNEL_ARM_PMUV3=y
CONFIG_KERNEL_BPF_EVENTS=y
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CIFS_ACL is not set
CONFIG_KERNEL_DEBUG_INFO_BTF=y
CONFIG_KERNEL_DEBUG_INFO_BTF_MODULES=y
[... 444 lines cut ...]
CONFIG_SINGBOX_WITH_QUIC=y
CONFIG_SINGBOX_WITH_REALITY_SERVER=y
CONFIG_SINGBOX_WITH_UTLS=y
CONFIG_SINGBOX_WITH_WIREGUARD=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=160
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_LLVM_HOST=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_ZRAM_DEF_COMP_LZORLE=y
CONFIG_ZSTD_OPTIMIZE_O3=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

libiunc, 2024-11-23:
Noticed that Bananapi BPi-R64 (mediatek/mt7622) and StarFive VisionFive2 v1.3b (starfive/generic jh7110) also have the same problem

robimarko, 2024-11-23:
Please post the whole compilation of pciutils, because the snippet you posted is just flags, there is no error in it

Spudz76, 2024-11-24:
I had a build error in `kmod-32` as well (different platform, no reconfiguration), solved by nuking its `build_dir`, try:
`rm -r build_dir/target-aarch64_cortex-a53_musl/kmod-32`
And then run `make` again.  I doubt the addition of `pciutils` had anything to do with it.

Specific error on my tree was:
```
for tool in insmod lsmod rmmod depmod modprobe modinfo; do \
        ln -s kmod /media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/$tool; \
done
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/insmod': File exists
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/lsmod': File exists
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/rmmod': File exists
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/depmod': File exists
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/modprobe': File exists
ln: failed to create symbolic link '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/ipkg-install/usr/bin/modinfo': File exists
make[7]: *** [Makefile:2680: install-exec-hook] Error 1
make[6]: *** [Makefile:2554: install-exec-am] Error 2
make[5]: *** [Makefile:2440: install-am] Error 2
make[4]: *** [Makefile:1849: install-recursive] Error 1
make[4]: Leaving directory '/media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32'
make[3]: *** [Makefile:96: /media/wrtsrc/openwrt/build_dir/target-mipsel_24kc_musl/kmod-32/.built] Error 2
make[3]: Leaving directory '/media/wrtsrc/openwrt/feeds/packages/utils/kmod'
time: package/feeds/packages/kmod/compile#0.36#0.10#0.42
    ERROR: package/feeds/packages/kmod failed to build.
make[2]: *** [package/Makefile:185: package/feeds/packages/kmod/compile] Error 1
```

libiunc, 2024-11-24:
> Please post the whole compilation of pciutils, because the snippet you posted is just flags, there is no error in it

`
make[3]: Entering directory '/media/lib/LuckyE/opn/openwrt/feeds/packages/utils/kmod'
rm -f /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/.built
touch /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " make  -C /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[4]: Entering directory '/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32'
make --no-print-directory all-recursive
Making all in .
Making all in libkmod/docs
make[6]: Nothing to be done for 'all'.
Making all in man
make[6]: Nothing to be done for 'all'.
make[4]: Leaving directory '/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32'
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/media/lib/LuckyE/opn/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " make -C /media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32=kmod-32 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" DESTDIR="/media/lib/LuckyE/opn/openwrt/build_dir/target-aarch64_cortex-a53_musl/kmod-32/ipkg-install"   install;
[... cut ...]
make[2]: Leaving directory '/media/lib/LuckyE/opn/openwrt'
make[1]: *** [package/Makefile:179: /media/lib/LuckyE/opn/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/lib/LuckyE/opn/openwrt'
make: *** [/media/lib/LuckyE/opn/openwrt/include/toplevel.mk:248：world] 错误 2

________________________________________________________
Executed in  280.63 secs    fish           external
   usr time  244.91 secs  421.00 micros  244.91 secs
   sys time   86.45 secs    0.00 micros   86.45 secs
 `

libiunc, 2024-11-24:
On the platforms I have (jh7110 mt7981/7986/7622), as long as **pciutils** kmod-32 is selected, an error will be reported. At that time, I thought there was a problem with pciutils, but now it seems that there is a problem with kmod.

Spudz76, 2024-11-24:
Well I have `pciutils` selected, had the same crash in `kmod-32` and manually spot-cleaning the build_dir fixed it for me.  Only happened after recent `main` branch update.

ptpt52, 2024-11-26:
same issue report for me

robimarko, 2024-11-26:
I cannot reproduce this, probably a one-off that will be magically fixed after cleaning

ptpt52, 2024-11-26:
> I cannot reproduce this, probably a one-off that will be magically fixed after cleaning

possible reproduce steps:
build config for ramips-mt7621 and then build it again.

robimarko, 2024-11-26:
I aint going to build MT7261 for this as package is universal, I cannot reproduce it on qualcommax no matter how times I rebuilt

robimarko, 2024-11-26:
Yeah, enabling stuff like that did not work.

Spudz76, 2024-11-26:
When it failed for me I changed nothing.  Deleting the build_dir for kmod-32 fixed it fine (manual spot-clean).  Suspect some recent change in kmod-32.

ptpt52, 2024-11-26:
build failed here: https://github.com/x-wrt/build-release/actions/runs/12002975064/job/33496476091

Spudz76, 2024-11-27:
> build failed here: https://github.com/x-wrt/build-release/actions/runs/12002975064/job/33496476091

Useless without `make -j1 V=sc` output

LynxChaus, 2024-11-27:
It fails only when rebuild whole tree without cleaning.

initzhu, 2024-11-27:
> It fails only when rebuild whole tree without cleaning.

I can reproduce it no matter whether I re-pull the source code for compilation or execute make distclean

ptpt52, 2024-12-06:
a similar issue also happens on `ERROR: package/feeds/packages/i2c-tools failed to build.`

LynxChaus, 2024-12-06:
> a similar issue also happens on `ERROR: package/feeds/packages/i2c-tools failed to build.`

i2c-tools need patch to avoid build pyton and smbus module when it not selected in config.
https://gist.github.com/LynxChaus/396c5a6f0b6f6693bf6bc912e43b81a0

ptpt52, 2024-12-10:
@robimarko  take a look at the failed log

Spudz76, 2024-12-10:
I fixed mine similarly but by inserting 
```
 		$(LN_S) -f kmod $(DESTDIR)$(bindir)/$$tool; \
```