openwrt/openwrt #17071: QCN9074 on x86 machine. Card does not function at all

CursedSilicon, 2024-11-25:
### Describe the bug

Hi all, 

I am trying to bring up a QCN9074 chipset based Wi-Fi card (specifically a [Wallys DR9074 6E](https://www.wallystech.com/Network_Card/DR9074-6E-Wifi6E-Qualcomm-QCN9074-Single-Band-6GHz-4T4R-M-2-E-Key-Interface-802-11ax.html))

It's a Wi-Fi 6E compatible chip with an M.2 connector and an accompanying PCI-E 1x slot adapter. 

I have it plugged in to an x86_64 PC running the current latest OpenWRT snapshot at time of writing (r28050+73-be952e98bc) with Kernel 6.6.61 (built November 17th 2024 @ 18:29)

I've successfully installed the relevant packages (`kmod-ath11k-pci` and `ath11k-firmware-qcn9074`) along with hostapd, iw etc.

`dmesg` does not seem to report anything unusual. However I'm unable to initialize any kind of Wi-Fi activity from the device. 

Looking at it via LuCI the "Maximum transmit power" field appears to be broken. Reporting options of either `driver default` or `255 dBm (-2147483648 mW)` 

Both options report `- Current power: 0 dBm` despite this.

I've read reports of Reddit of users backporting support to kernel 5.10 and I can see that there is at least one other device in OpenWRT's table of hardware (the `"prpl Foundation Haze WNC LVRP`)" that supports this same chipset currently. 

I have also seen additional comments that Virtualization can interfere with the card. However this device is running bare-metal on a 6th gen "Skylake" intel system.

### OpenWrt version

r28050+73-be952e98bc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

HP HP ProDesk 400 G4 SFF

### Image kind

Official downloaded image

### Steps to reproduce

1. Install QCN 9074 in device 

2. Install relevant packages `kmod-ath11k-pci` `ath11k-firmware-qcn9074` `iw` `hostapd` `luci` and others

3. Reboot 

4. Attempt to create a Wi-Fi AP via LuCI 

### Actual behaviour

1. Install QCN 9074 in device 

2. Install relevant packages `kmod-ath11k-pci` `ath11k-firmware-qcn9074` `iw` `hostapd` `luci` and others

3. Reboot 

4. Attempt to create a Wi-Fi AP via LuCI 

5. Nothing happens. No changes are reported via `dmesg` or any other mechanism

### Expected behaviour

1. Install QCN 9074 in device 

2. Install relevant packages `kmod-ath11k-pci` `ath11k-firmware-qcn9074` `iw` `hostapd` `luci` and others

[... 670 lines cut ...]
[    8.075728] mhi mhi0: Requested to power ON
[    8.075904] mhi mhi0: Power on setup success
[    8.190371] mhi mhi0: Wait for device to enter SBL or Mission mode
[    8.484472] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.498355] ath11k_pci 0000:02:00.0: chip_id 0x0 chip_family 0x0 board_id 0xa2 soc_id 0xffffffff
[    8.498753] ath11k_pci 0000:02:00.0: fw_version 0x290b8862 fw_build_timestamp 2024-09-23 10:51 fw_build_id
[    9.088760] pps pps0: new PPS source ptp0
[    9.088990] ixgbe 0000:01:00.0: registered PHC device on eth0
[    9.212045] 8021q: adding VLAN 0 to HW filter on device eth0
[    9.212529] br-lan: port 1(eth0) entered blocking state
[    9.212781] br-lan: port 1(eth0) entered disabled state
[    9.213039] ixgbe 0000:01:00.0 eth0: entered allmulticast mode
[    9.213826] ixgbe 0000:01:00.0 eth0: entered promiscuous mode
[    9.346653] pps pps1: new PPS source ptp1
[    9.346889] ixgbe 0000:01:00.1: registered PHC device on eth1
[    9.470677] 8021q: adding VLAN 0 to HW filter on device eth1
[   10.171698] ath11k_pci 0000:02:00.0: htt event 48 not handled
[   12.992686] ixgbe 0000:01:00.0 eth0: NIC Link is Up 1 Gbps, Flow Control: RX/TX
[   12.993100] br-lan: port 1(eth0) entered blocking state
[   12.993466] br-lan: port 1(eth0) entered forwarding state
[   13.712659] ixgbe 0000:01:00.1 eth1: NIC Link is Up 1 Gbps, Flow Control: RX/TX`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

CursedSilicon, 2024-11-25:
Additional:

lspci -vvv output 

[lspci.txt](https://github.com/user-attachments/files/17896423/lspci.txt)

CursedSilicon, 2024-11-25:
`r28050+73-be952e98bc` should correspond to the latest `auc` installed trunk build

brada4, 2024-11-25:
Please describe the 73 patches on top of current snapshot?

CursedSilicon, 2024-11-25:
> Please describe the 73 patches on top of current snapshot?

I do not have any out of tree patches applied to this snapshot. To my understanding, `r28050+73-be952e98bc` is an official snapshot from downloads.openwrt.org. I can upload the `generic-ext4-combined-efi.img.gz` file for confirmation as well if desired.

>What does logread report?

Snipped relevant output (remainder of lastlog is mostly DHCP messages after `dmesg` from bootup)

`Sun Nov 24 17:59:33 2024 daemon.notice netifd: radio0 (6125): WARNING: Variable 'data' does not exist or is not an array/object`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: Set new config for phy phy0:`
`Sun Nov 24 17:59:34 2024 daemon.notice netifd: radio0 (6125): Command failed: Not found`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: Restart interface for phy phy0`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 stationary_ap=1 #num_global_macaddr=1 channel=acs_survey  interface=phy0-ap0 bssid=c4:4b:d1:d0:00:e0 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 auth_algs=1 wpa=0 ssid=OpenWrt bridge=br-lan wds_bridge= snoop_iface=br-lan qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56 #default_macaddr nas_identifier=c44bd1d000e0  (phy phy0) --> new PHY`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE`
`Sun Nov 24 17:59:34 2024 daemon.err hostapd: Hardware does not support configured mode`
`Sun Nov 24 17:59:34 2024 daemon.warn hostapd: phy0-ap0: IEEE 802.11 Hardware does not support configured mode (1) (hw_mode in hostapd.conf)`
`Sun Nov 24 17:59:34 2024 daemon.err hostapd: Could not select hw_mode and channel. (-2)`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: phy0-ap0: AP-DISABLED`
`Sun Nov 24 17:59:34 2024 daemon.err hostapd: phy0-ap0: Unable to setup interface.`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.722837] br-lan: port 2(phy0-ap0) entered blocking state`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.723085] br-lan: port 2(phy0-ap0) entered disabled state`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.723333] ath11k_pci 0000:02:00.0 phy0-ap0: entered allmulticast mode`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.723673] ath11k_pci 0000:02:00.0 phy0-ap0: entered promiscuous mode`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.725639] ath11k_pci 0000:02:00.0 phy0-ap0: left allmulticast mode`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.725928] ath11k_pci 0000:02:00.0 phy0-ap0: left promiscuous mode`
`Sun Nov 24 17:59:34 2024 kern.info kernel: [  917.726295] br-lan: port 2(phy0-ap0) entered disabled state`
`Sun Nov 24 17:59:34 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING`
`Sun Nov 24 17:59:34 2024 daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started`
`Sun Nov 24 17:59:34 2024 daemon.notice hostapd: hostapd.add_iface failed for phy phy0 ifname=phy0-ap0`
`Sun Nov 24 17:59:34 2024 daemon.notice netifd: Wireless device 'radio0' is now up`
`Sun Nov 24 17:59:36 2024 daemon.info dnsmasq[1]: read /etc/hosts - 6 names`
`Sun Nov 24 17:59:36 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names`
`Sun Nov 24 17:59:36 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses`
`Sun Nov 24 17:59:55 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf`
`Sun Nov 24 17:59:55 2024 daemon.notice hostapd: Reloaded settings for phy phy0`
`Sun Nov 24 17:59:55 2024 daemon.notice netifd: Wireless device 'radio0' is now up`

CursedSilicon, 2024-11-25:
/etc/config/wireless

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:1d.0/0000:02:00.0'
        option cell_density '0'
        option country 'US'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option network 'lan'

```

iw dev

```

phy#0
        Interface phy0-ap0
                ifindex 6
                wdev 0x2
                addr c4:4b:d1:d0:00:e0
                type AP
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

```

iw phy

```

Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 16
        max scan IEs length: 440 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * managed
                 * AP
                 * monitor
                 * mesh point
        Band 4:
                HE Iftypes: managed
                        HE MAC Capabilities (0x000b9a181040):
                                +HTC HE Supported
                                TWT Requester
                                Dynamic BA Fragementation Level: 1
                                BSR
                                Broadcast TWT
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                RX Control Frame to MultiBSS
                                A-MSDU in A-MPDU
                                OM Control UL MU Data Disable RX
                        HE PHY Capabilities: (0x1c604c89ffdb839c110c00):
                                HE40/HE80/5GHz
[... 321 lines cut ...]

interface=phy0-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
auth_algs=1
wpa=0
ssid=OpenWrt
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=c4:4b:d1:d0:00:e0
#default_macaddr

```

robimarko, 2024-11-25:
You are missing channel property in the config, no wonder it does not work

CursedSilicon, 2024-11-25:
I do not have the ability to set a channel via LuCI (see screenshot)

[image]

I tried inputting the required sections to `/etc/config/wireless` using my existing OpenWRT router as a reference for syntax

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:1d.0/0000:02:00.0'
        option cell_density '0'
        option country 'US'
        option channel '6'
        option band '2g'
        option cell_density '0'
        option htmode 'HT40'
        option txpower '28'
        option legacy_rates '1'

```

This did not appear to change anything in LuCI, nor did the card begin to operate after restarting the system.

robimarko, 2024-11-25:
Your card can only do 6GHz channels, 2GHz won't work.

As for LuCI, 6GHz support was added but I have no idea how it works as I dont have any HW supporting it.

CursedSilicon, 2024-11-25:
I don't have any hardware to reference that can operate on 6Ghz under OpenWRT. Testing it with 5Ghz should be functional according to the data sheet. The below configuration enabled the "Operating Frequency" selection but otherwise is identical to the previous screenshot

[image]

```
        option channel '36'
        option band '5g'
        option htmode 'HE160'
        option cell_density '0'
        option country 'US'
```

robimarko, 2024-11-25:
Look, there are multiple QCN9074 models, some support 2GHz only, some 2 and 5, some are 5 only, some 5 and 6, and some 6GHz only.

CursedSilicon, 2024-11-25:
Yes. The one I have appears to be the 6Ghz only model (likely a mistake from the retailer who I purchased it from)

I do have a compatible `RZ616` Wi-Fi chipset in my laptop that I can test pure 6Ghz only operation with. However I'm unclear if "6Ghz only" chips (or any kind of 'single band only' chips) are supported or not

robimarko, 2024-11-25:
Yes, single band radios are supported

testuser7, 2024-11-28:
> /etc/config/wireless
> 
> ```
> config wifi-iface 'wifinet0'
>         option device 'radio0'
>         option mode 'ap'
>         option ssid 'OpenWrt'
>         option encryption 'none'
>         option network 'lan'
> ```

You need to use WPA3 (`sae` option) for encryption. For 6GHz this is required.

testuser7, 2024-11-28:
> As for LuCI, 6GHz support was added but I have no idea how it works as I dont have any HW supporting it.

There is a problem with `iwinfo`, which is probably used in LuCI to retrieve values ​​and does not work in the 6 GHz band:
- https://github.com/openwrt/openwrt/issues/16357
- https://github.com/openwrt/openwrt/issues/16548

testuser7, 2024-12-06:
@CursedSilicon can you check if your card works with BDF from this PR: https://github.com/openwrt/openwrt/pull/16731 ?

CursedSilicon, 2024-12-22:
@testuser7 Sorry for the delay. Will test this later/tomorrow and report back

CursedSilicon, 2024-12-23:
Update: The machine it was in has suffered hardware failure. Will need to resume testing once I replace it