openwrt/openwrt #17075: mvebu does not boot

borkra, 2024-11-25:
### Describe the bug

Router does not boot

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

mvebu

### Device

WRT1900ACv2

### Image kind

Self-built image

### Steps to reproduce

Turn the power on

### Actual behaviour

Router does not start

### Expected behaviour

Router starting

### Additional info

/bin/sh: can't open '/etc/preinit': No such file or directory
open: No such file or directory
open: No such file or directory

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-11-25:
Please post the whole bootlog, otherwise its impossible to even guess.

How did you acquire the image that was flashed?

borkra, 2024-11-25:
I built the image myself. 

/etc/preinit is not installed on the image, even though /sbin/init requires it.

You do not need bootlog for this, just look in the ./build_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu after build it's not there

borkra, 2024-11-25:
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.63 (boris@boris-pc) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 14.2.0 r28116-cd92cbddf8) 14.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Nov 25 11:20:59 2024
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=50c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Linksys WRT1900AC v2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18452 r8192 d22508 u49152
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock7 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
[    0.000000] Bootloader command line (ignored): console=ttyS0,115200 root=/dev/mtdblock7 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
[    0.000000] Unknown kernel command line parameters "earlyprintk mangled_fs=jffs2 mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508460K/524288K available (6144K kernel code, 632K rwdata, 2168K rodata, 1024K init, 231K bss, 15828K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x06070000 -> 0x16070000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] sched_clock: 64 bits at 800MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000011] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xb881274fa3, max_idle_ns: 440795210636 ns
[    0.000025] Switching to timer-based delay loop, resolution 1ns
[    0.000153] Ignoring duplicate/late registration of read_current_timer delay
[    0.000156] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000297] Calibrating delay loop (skipped), value calculated using timer frequency.. 1600.00 BogoMIPS (lpj=8000000)
[    0.000304] CPU: Testing write buffer coherency: ok
[    0.000317] CPU0: Spectre v2: using BPIALL workaround
[    0.000321] pid_max: default: 32768 minimum: 301
[    0.002836] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.002841] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.005454] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005728] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=2.
[    0.005769] Setting up static identity map for 0x100000 - 0x100054
[    0.005844] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.005924] mvebu-pmsu: Initializing Power Management Service Unit
[    0.006001] rcu: Hierarchical SRCU implementation.
[    0.006003] rcu: 	Max phase no-delay instances is 1000.
[    0.006140] smp: Bringing up secondary CPUs ...
[    0.006297] Booting CPU 1
[    0.006417] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006423] CPU1: Spectre v2: using BPIALL workaround
[    0.006470] smp: Brought up 1 node, 2 CPUs
[    0.006474] SMP: Total of 2 processors activated (3200.00 BogoMIPS).
[    0.006478] CPU: All CPU(s) started in SVC mode.
[    0.009154] VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
[    0.009207] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.009215] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[... cut ...]
[    3.324871] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
[    3.331982] clk: Disabling unused clocks
[    3.336029] Waiting 1 sec before mounting root device...
[    4.411873] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
[    4.419626] Freeing unused kernel image (initmem) memory: 1024K
[    4.425708] Run /sbin/init as init process
/bin/sh: can't open '/etc/preinit': No such file or directory
open: No such file or directory
open: No such file or directory

robimarko, 2024-11-25:
Please post your diffconfig as well

borkra, 2024-11-25:
The problem is `base-files` package now depend on `KERNEL_NAMESPACES` config but nobody force selects it. Probably need to use `select` not `depend`

robimarko, 2024-11-25:
It is selected by default if small-flash is not set, then `KERNEL_NAMESPACES` is not selected but base-files also wont pull-in ujail

borkra, 2024-11-25:
Unfortunatly, that is not, that is how base-files look in menuconfig. 
If SMALL_FLASH or KERNEL_NAMESPACES not selected base_files is off and no  mention of ujail

[image]

borkra, 2024-11-25:
DEFAULT_base-files does not build base-fies

borkra, 2024-11-25:
Here is final definition of base-file config, as you can see SMALL_FLASH or KERNEL_NAMESPACES required

```
	config PACKAGE_base-files
		tristate "base-files................................... Base filesystem for OpenWrt"
		default y if DEFAULT_base-files
		default m if ALL||ALL_NONSHARED
		depends on !(!SMALL_FLASH) || KERNEL_NAMESPACES
		select PACKAGE_busybox if !SELINUX && PACKAGE_busybox-selinux<PACKAGE_base-files
		select PACKAGE_busybox-selinux if SELINUX
		select PACKAGE_fstools
		select PACKAGE_fwtool
		select PACKAGE_jsonfilter
		select PACKAGE_libc
		select PACKAGE_netifd
		select PACKAGE_openwrt-keyring if SIGNED_PACKAGES
		select PACKAGE_procd if !SELINUX && PACKAGE_procd-selinux<PACKAGE_base-files
		select PACKAGE_procd-seccomp if USE_SECCOMP
		select PACKAGE_procd-selinux if SELINUX
		select PACKAGE_procd-ujail if !SMALL_FLASH
		select PACKAGE_ubi-utils if NAND_SUPPORT
		select PACKAGE_usign if SIGNED_PACKAGES
		help
		 This package contains a base filesystem and system scripts for OpenWrt.
```

robimarko, 2024-11-25:
Please post your diffconfig

borkra, 2024-11-25:
Commit that broke it is: ac640718aa0ceae55969bb9e7e45d00bb7bc228a by @ynezz

commit 341cc047 (2024-11-26): Revert "target,base-files: unify handling of procd-ujail"

This reverts commit ac640718aa0ceae55969bb9e7e45d00bb7bc228a as it
removes base-files package when KERNEL_NAMESPACES is deselected, as
base-files depends on procd-ujail which needs KERNEL_NAMESPACES.

Fixes: #17075
Link: https://github.com/openwrt/openwrt/pull/17081
Link: https://github.com/openwrt/openwrt/pull/17081
PR #17081 (2024-11-26): Revert "target,base-files: unify handling of procd-ujail"
This reverts commit ac640718aa0ceae55969bb9e7e45d00bb7bc228a as it removes base-files package when KERNEL_NAMESPACES is deselected, asi base-files depends on procd-ujail which needs KERNEL_NAMESPACES.

Fixes: #17075
closed 2024-11-26 by openwrt-bot