openwrt/openwrt #17076: [mediatek/filogic][BPI-R4] Kernel can catch soft lockup condition during CPU freq switch under significant IO load

N-Storm, 2024-11-25:
### Describe the bug

Run into this issue at first while testing NVMe performance with fio on my custom build - on a sequential read having queue depth setting (`--iodepth`) > 4 (starting to happen really often on values 16-64) it wasn't stable. Sometimes it runs stable, giving me ~850 MiB/s reads. Sometimes read speed drops significantly (~110-120 MiB/s) and whole system becomes unresponsive. Pings won't reply, SSH sessions timeouts and even serial console (UART connection) outputs characters very slowly. One of CPU cores are 100% busy doing IRQ handling.

Was able to isolate and reproduce condition with upstream OpenWRT. Had this happen on my build at first, because I had ondemand cpu scaling governor enabled. These lockups can occur when CPU frequency are switched. Happens even with default "userspace" governor if switching speed manually. 

Tested with non-modified [build from downloads.openwrt.org](https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/mediatek/filogic/): 
```
bananapi_bpi-r4-sdcard.img.gz	6e5c83b05fa04c1673822eda65c7e4de67636bd0ea1f5b86e49f625483da6268	32511.9 KB	Sun Nov 24 16:13:22 2024
```

Apk-based snapshots are affected as well. Happens when two conditions met:

1. IO load having some queue depth running, i.e. fio doing a simple read test;
2. CPU frequency switch occurs. 

So far it seems that it doesn't matters from/to which frequency it switches. Sometimes it goes fine, but in some cases frequency switch immediately gets CPU to to that state and it stucks there while fio running the load. Once fio finishes doing the test (i.e. no more IO load), everything returns back to normal until next time those conditions are met.

### OpenWrt version

r28033-c279ca8b79

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

BananaPI BPI-R4 + BE14000 WiFi module + NVMe drive installed

### Image kind

Official downloaded image

### Steps to reproduce

I've tested under these conditions, so before the issue could be tested under others to narrow-down the list, these can be considered as prerequisites:

1. BPI-R4 with BE14000 WiFI7 module & NVMe SSD installed in M.2 slot designated for NVMe (back of the board, same side with the WiFi module).
2. Have all 3 AP per band (2.4/5/6GHz) running. Not sure that this are required, but seems like it contributes to the chances getting lockup (sounds logical - more hardware interrupts / driver handing / etc).
3. GPT partition on the NVMe, only 1 of the partitions mounted where fio testfile are located. Ext4 formatted.

Steps:
1. Install `fio` package. Others, like kmod-nvme or e2fsprogs should be already included with the "official" OpenWRT image I guess, but should be installed if not included with custom image.
2. Format a partition on the NVMe to ext4 fs and mount it to /mnt with the default options: `mount -t ext4 /dev/nvme0n1p1 /mnt`
3. Open 2 ssh terminals to the BPI-R4. Commands below could be run in the backgroud or screen, but I prefer to watch each running in it's own session.
4. In one session, run the sequential read test on NVMe with fio: `fio --filename=/mnt/testfile.dat --size=200MB --numjobs=4 --time_based --group_reporting --name=throughput-test-job --eta-newline=1 --direct=1 --rw=read --bs=256k --ioengine=libaio --iodepth=64 --runtime=120`
4. By the same time on a second terminal (or use screen/etc) run this simple one-liner to manually switch CPU frequency every second, by cycling through them: `i=0; while true; do sleep 1; case "$i" in 0) F=800000;; 1) F=1100000;; 2) F=1500000;; 3) F=1800000;; 4) F=800000; i=0;; esac; echo "Setting CPU Freq to $F"; echo $F > /sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed; i=$(($i+1)); done`
5. Watch until terminals suddenly stops to update and host becomes unresponsive. Repeat steps 3-4 if it won't happen during 

### Actual behaviour

Soft kernel/CPU lockup.

### Expected behaviour

Normal operation

### Additional info

Must be something related to IRQ/DMA handing. I can observe a picture like this in atop when this happens:
[image]

But this happens only when CPU frequency changes. If it's static, no matter which, even with lowest 800 MHz settings I'm getting stable read speeds without any lockups and CPU usage within reasonable bounds. So it's not a load that BPI-R4/mt7988 hardware can't handle.

`/proc/interrupts` doesn't seems to show any weird values. NVMe interrupts happens but I won't call it something extraordinary with a maximum delta of 2743. 

Feel free to ask me to gather more stats which might help.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

N-Storm, 2024-11-25:
Short video to demonstrate the issue:

https://github.com/user-attachments/assets/6361d346-1fc3-4f52-811f-1299980412c6

N-Storm, 2024-12-05:
Running on kernel with performance counters & some of the debug options enabled. Recording a trace with `perf sched` might give a hint on what's causing those "hiccups". Here is the latency analysis when everything went fine (i.e. nearly constant read speed and no freezes):

```
 -------------------------------------------------------------------------------------------------------------------------------------------
  Task                  |   Runtime ms  | Switches | Avg delay ms    | Max delay ms    | Max delay start           | Max delay end          |
 -------------------------------------------------------------------------------------------------------------------------------------------
  ksoftirqd/1:21        |     42.251 ms |       99 | avg:   7.535 ms | max:  35.976 ms | max start:  1446.122265 s | max end:  1446.158241 s
  ksoftirqd/2:26        |     29.219 ms |       96 | avg:   6.988 ms | max:  36.040 ms | max start:  1442.638158 s | max end:  1442.674198 s
  ksoftirqd/3:31        |     37.213 ms |      104 | avg:   6.041 ms | max:  40.016 ms | max start:  1452.090296 s | max end:  1452.130312 s
  kworker/1:4-eve:924   |      2.908 ms |      127 | avg:   1.237 ms | max:  20.053 ms | max start:  1442.838150 s | max end:  1442.858203 s
  kworker/2:6-mm_:941   |      2.742 ms |      155 | avg:   0.843 ms | max:  16.083 ms | max start:  1446.330192 s | max end:  1446.346275 s
  kworker/3:9-mm_:201   |      2.934 ms |      135 | avg:   0.797 ms | max:  20.060 ms | max start:  1451.150258 s | max end:  1451.170318 s
  ksoftirqd/0:15        |      6.301 ms |      210 | avg:   0.478 ms | max:  23.968 ms | max start:  1453.474442 s | max end:  1453.498409 s
  atop:(2)              |   7315.935 ms |      304 | avg:   0.206 ms | max:   1.642 ms | max start:  1441.110154 s | max end:  1441.111796 s
  napi/mtk_eth-5:1009   |     32.750 ms |     1771 | avg:   0.164 ms | max:  20.513 ms | max start:  1441.194345 s | max end:  1441.214857 s
  napi/mtk_eth-6:1010   |     38.325 ms |     1276 | avg:   0.144 ms | max:  16.933 ms | max start:  1441.018691 s | max end:  1441.035624 s
  kworker/0:0-eve:8     |      2.966 ms |      161 | avg:   0.142 ms | max:   8.004 ms | max start:  1453.490298 s | max end:  1453.498302 s
  migration/0:17        |      0.000 ms |      258 | avg:   0.052 ms | max:   1.406 ms | max start:  1441.034123 s | max end:  1441.035529 s
  migration/2:25        |      0.000 ms |       29 | avg:   0.044 ms | max:   0.775 ms | max start:  1440.030091 s | max end:  1440.030867 s
  blockd:5260           |      0.292 ms |        8 | avg:   0.037 ms | max:   0.119 ms | max start:  1454.302281 s | max end:  1454.302400 s
  sshd-session:(2)      |    365.356 ms |     1595 | avg:   0.032 ms | max:   0.974 ms | max start:  1443.441415 s | max end:  1443.442389 s
  migration/3:30        |      0.000 ms |       30 | avg:   0.029 ms | max:   0.187 ms | max start:  1445.234177 s | max end:  1445.234363 s
  kcompactd0:39         |      0.753 ms |       79 | avg:   0.027 ms | max:   0.149 ms | max start:  1443.510158 s | max end:  1443.510307 s
  mt76-tx phy0:2527     |      1.252 ms |       28 | avg:   0.022 ms | max:   0.063 ms | max start:  1440.634324 s | max end:  1440.634388 s
  kworker/u8:2-ev:45    |     74.006 ms |     3362 | avg:   0.022 ms | max:   2.666 ms | max start:  1441.357935 s | max end:  1441.360601 s
  kworker/u8:0-ev:11    |     86.256 ms |     3931 | avg:   0.022 ms | max:   2.158 ms | max start:  1441.020037 s | max end:  1441.022195 s
  procd:1               |      6.036 ms |       48 | avg:   0.022 ms | max:   0.233 ms | max start:  1442.045254 s | max end:  1442.045487 s
  kworker/u8:4-ph:9169  |    184.942 ms |     4420 | avg:   0.022 ms | max:   3.340 ms | max start:  1445.147022 s | max end:  1445.150362 s
  kworker/0:1H-kb:48    |      0.528 ms |        2 | avg:   0.021 ms | max:   0.028 ms | max start:  1449.718226 s | max end:  1449.718255 s
  odhcpd:4135           |      1.236 ms |       41 | avg:   0.020 ms | max:   0.083 ms | max start:  1443.606141 s | max end:  1443.606224 s
  netifd:3948           |      8.774 ms |       40 | avg:   0.020 ms | max:   0.066 ms | max start:  1476.875748 s | max end:  1476.875814 s
  wpa_supplicant:3880   |      6.540 ms |       44 | avg:   0.020 ms | max:   0.122 ms | max start:  1445.798183 s | max end:  1445.798305 s
  napi/phy0-8:2520      |     48.235 ms |      504 | avg:   0.020 ms | max:   2.186 ms | max start:  1437.400031 s | max end:  1437.402216 s
  migration/1:20        |      0.000 ms |       36 | avg:   0.020 ms | max:   0.055 ms | max start:  1444.030144 s | max end:  1444.030198 s
  kworker/1:0H-kb:23    |      0.179 ms |        1 | avg:   0.019 ms | max:   0.019 ms | max start:  1443.574156 s | max end:  1443.574176 s
  kworker/u8:1-ev:35    |    136.078 ms |     2067 | avg:   0.018 ms | max:   2.105 ms | max start:  1473.499766 s | max end:  1473.501871 s
  rcu_sched:16          |     68.081 ms |     5312 | avg:   0.017 ms | max:   4.021 ms | max start:  1437.398115 s | max end:  1437.402136 s
  hostapd:3881          |     80.010 ms |      648 | avg:   0.017 ms | max:   0.166 ms | max start:  1443.442029 s | max end:  1443.442195 s
  logd:3069             |      5.361 ms |       41 | avg:   0.017 ms | max:   0.040 ms | max start:  1441.852378 s | max end:  1441.852418 s
  ash:9223              |     20.777 ms |       40 | avg:   0.016 ms | max:   0.031 ms | max start:  1437.904136 s | max end:  1437.904167 s
[... cut ...]
  ksoftirqd/1:21        |      0.031 ms |        2 | avg:   0.013 ms | max:   0.017 ms | max start:  2282.878093 s | max end:  2282.878109 s
  napi/mtk_eth-6:1010   |      0.105 ms |        5 | avg:   0.011 ms | max:   0.014 ms | max start:  2281.284975 s | max end:  2281.284989 s
 -----------------------------------------------------------------------------------------------------------------
  TOTAL:                |  95241.086 ms |    23172 |
 ---------------------------------------------------
  INFO: 0.005% context switch bugs (1 out of 21918)
```

btv, 2024-12-10:
Out of pure curiosity, what happens when you install the BFQ module and apply it to NVME that is hosting / ?
https://openwrt.org/packages/pkgdata/kmod-iosched-bfq

N-Storm, 2024-12-13:
Tried BFQ. As I was expecting, nothing really changed. Got same hiccups and didn't looked any further:

```
root@OpenWrt:/mnt# cat /sys/class/block/nvme0n1/queue/scheduler
none mq-deadline kyber [bfq]
root@OpenWrt:/mnt# fio --filename=/mnt/testfile.dat --size=200MB --numjobs=4 --time_based --group_reporting --name=throughput-test-job --eta-newline=1 --direct=1 --rw=read --bs=256k --ioengine=libaio --iodepth=64 --runtime=20
throughput-test-job: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=64
...
fio-3.37
Starting 4 threads
Jobs: 4 (f=4): [R(4)][14.3%][r=829MiB/s][r=3316 IOPS][eta 00m:18s]
Jobs: 4 (f=4): [R(4)][19.0%][r=788MiB/s][r=3152 IOPS][eta 00m:17s]
Jobs: 4 (f=4): [R(4)][23.8%][r=850MiB/s][r=3398 IOPS][eta 00m:16s]
Jobs: 4 (f=4): [R(4)][30.0%][r=820MiB/s][r=3281 IOPS][eta 00m:14s]
Jobs: 4 (f=4): [R(4)][35.0%][r=850MiB/s][r=3398 IOPS][eta 00m:13s]
Jobs: 4 (f=4): [R(4)][40.0%][r=850MiB/s][r=3399 IOPS][eta 00m:12s]
Jobs: 4 (f=4): [R(4)][45.0%][r=850MiB/s][r=3398 IOPS][eta 00m:11s]
Jobs: 4 (f=4): [R(4)][50.0%][r=344MiB/s][r=1375 IOPS][eta 00m:10s]
Jobs: 4 (f=4): [R(4)][60.0%][r=849MiB/s][r=3395 IOPS][eta 00m:08s]
Jobs: 4 (f=4): [R(4)][70.0%][r=849MiB/s][r=3395 IOPS][eta 00m:06s]
Jobs: 4 (f=4): [R(4)][75.0%][r=849MiB/s][r=3396 IOPS][eta 00m:05s]
Jobs: 4 (f=4): [R(4)][80.0%][r=850MiB/s][r=3399 IOPS][eta 00m:04s]
Jobs: 4 (f=4): [R(4)][85.0%][r=835MiB/s][r=3338 IOPS][eta 00m:03s]
Jobs: 4 (f=4): [R(4)][95.0%][r=439MiB/s][r=1756 IOPS][eta 00m:01s]
Jobs: 4 (f=4): [R(4)][100.0%][r=102MiB/s][r=406 IOPS][eta 00m:00s]
Jobs: 4 (f=3): [R(2),f(1),R(1)][100.0%][r=128MiB/s][r=511 IOPS][eta 00m:00s]
```

N-Storm, 2024-12-14:
Trying to do some kernel profiling / tracing, which I'm not familiar with at all, but from my unexperienced look it must be an issue in MTK frequency scaling driver. At the first glance it looks like it polls/waits for some events/state change and propagates changes on others which causes a lot of IRQs overloading or somehow causing other IRQs routines getting delayed too much (which in turn might trigger other effects when kernel tries to resolve these, but that's just a guess).

At least that's that I've concluded from my amateur "research". These are the parts of ftrace reports I guess might be of interest.

Parts of the trace-cmd ftrace (function graph) report when CPU freq scaling triggered every second and I've experienced freezes (ash-nnnnn task are the one which had frequency change loop running):

```console
...
task: kworker/0:0-8
  Event: func: process_one_work() (10) Total: 2955383 Avg: 295538 Max: 1046692(ts:4434.720144) Min:71846(ts:4422.611521)
  Event: func: generic_handle_domain_irq() (7) Total: 1630155 Avg: 232879 Max: 449923(ts:4430.713977) Min:95461(ts:4433.625423)
  Event: func: do_interrupt_handler() (6) Total: 1208538 Avg: 201423 Max: 452923(ts:4430.713978) Min:97307(ts:4433.625424)
  Event: func: gic_handle_irq() (6) Total: 1199000 Avg: 199833 Max: 451384(ts:4430.713977) Min:96462(ts:4433.625423)
  Event: func: free_ioctx() (1) Total: 1010538 Avg: 1010538 Max: 1010538(ts:4434.720119) Min:1010538(ts:4434.720119)
  Event: func: mtk_hw_reset_monitor_work() (3) Total: 951615 Avg: 317205 Max: 534307(ts:4430.714006) Min:55615(ts:4423.540699)
  Event: func: aio_free_ring() (1) Total: 931230 Avg: 931230 Max: 931230(ts:4434.720042) Min:931230(ts:4434.720042)
  Event: func: handle_percpu_devid_irq() (5) Total: 722076 Avg: 144415 Max: 184154(ts:4434.719627) Min:93153(ts:4433.625422)
  Event: func: truncate_setsize() (1) Total: 705923 Avg: 705923 Max: 705923(ts:4434.719819) Min:705923(ts:4434.719819)
  Event: func: arch_timer_handler_phys() (5) Total: 704307 Avg: 140861 Max: 180154(ts:4434.719624) Min:90923(ts:4433.625421)
  Event: func: truncate_pagecache() (1) Total: 701615 Avg: 701615 Max: 701615(ts:4434.719817) Min:701615(ts:4434.719817)
  Event: func: hrtimer_interrupt() (5) Total: 696307 Avg: 139261 Max: 178231(ts:4434.719623) Min:90154(ts:4433.625421)
  Event: func: truncate_inode_pages_range() (1) Total: 657000 Avg: 657000 Max: 657000(ts:4434.719796) Min:657000(ts:4434.719796)
  Event: func: __hrtimer_run_queues() (5) Total: 583615 Avg: 116723 Max: 149308(ts:4434.719604) Min:76692(ts:4433.625412)
  Event: func: mtk_r32() (54) Total: 541770 Avg: 10032 Max: 462000(ts:4430.713983) Min:1000(ts:4423.540662)
  Event: func: tick_sched_timer() (5) Total: 535692 Avg: 107138 Max: 137154(ts:4434.719597) Min:71077(ts:4433.625409)
  ...
  
```

And same tasks / parts when no CPU frequency scaling took place, but fio was running with the same workload (and worked without any freezes / slowdowns):

```
task: kworker/0:0-8
  Event: func: process_one_work() (10) Total: 2955383 Avg: 295538 Max: 1046692(ts:4434.720144) Min:71846(ts:4422.611521)
  Event: func: generic_handle_domain_irq() (7) Total: 1630155 Avg: 232879 Max: 449923(ts:4430.713977) Min:95461(ts:4433.625423)
  Event: func: do_interrupt_handler() (6) Total: 1208538 Avg: 201423 Max: 452923(ts:4430.713978) Min:97307(ts:4433.625424)
  Event: func: gic_handle_irq() (6) Total: 1199000 Avg: 199833 Max: 451384(ts:4430.713977) Min:96462(ts:4433.625423)
  Event: func: free_ioctx() (1) Total: 1010538 Avg: 1010538 Max: 1010538(ts:4434.720119) Min:1010538(ts:4434.720119)
  Event: func: mtk_hw_reset_monitor_work() (3) Total: 951615 Avg: 317205 Max: 534307(ts:4430.714006) Min:55615(ts:4423.540699)
  Event: func: aio_free_ring() (1) Total: 931230 Avg: 931230 Max: 931230(ts:4434.720042) Min:931230(ts:4434.720042)
  Event: func: handle_percpu_devid_irq() (5) Total: 722076 Avg: 144415 Max: 184154(ts:4434.719627) Min:93153(ts:4433.625422)
  Event: func: truncate_setsize() (1) Total: 705923 Avg: 705923 Max: 705923(ts:4434.719819) Min:705923(ts:4434.719819)
  Event: func: arch_timer_handler_phys() (5) Total: 704307 Avg: 140861 Max: 180154(ts:4434.719624) Min:90923(ts:4433.625421)
  Event: func: truncate_pagecache() (1) Total: 701615 Avg: 701615 Max: 701615(ts:4434.719817) Min:701615(ts:4434.719817)
  Event: func: hrtimer_interrupt() (5) Total: 696307 Avg: 139261 Max: 178231(ts:4434.719623) Min:90154(ts:4433.625421)
  Event: func: truncate_inode_pages_range() (1) Total: 657000 Avg: 657000 Max: 657000(ts:4434.719796) Min:657000(ts:4434.719796)
  Event: func: __hrtimer_run_queues() (5) Total: 583615 Avg: 116723 Max: 149308(ts:4434.719604) Min:76692(ts:4433.625412)
  Event: func: mtk_r32() (54) Total: 541770 Avg: 10032 Max: 462000(ts:4430.713983) Min:1000(ts:4423.540662)
  Event: func: tick_sched_timer() (5) Total: 535692 Avg: 107138 Max: 137154(ts:4434.719597) Min:71077(ts:4433.625409)
```

But capturing every function call is too much, so I've run another trace with a function filter set to 'mtk*' and doing freq. switching + fio reports this:

```
task: ash-10746
  Event: func: mtk_cpufreq_set_target() (3) Total: 13012180339 Avg: 4337393446 Max: 13004736262(ts:2896.702172) Min:3211616(ts:2881.692034)
  Event: func: mtk_pll_round_rate() (6) Total: 13003141341 Avg: 2167190223 Max: 13002521339(ts:2896.700131) Min:113538(ts:2882.692884)
[... cut ...]
Event:       func:  mtk_pll_round_rate()      (6)     Total:  13.01G   (  13003141341  )  Avg:  2167190223  Max:  13002521339(ts:2896.700131)  Min:113538(ts:2882.692884)13003141341   (  13003141341  )
Event:       func:  mtk_cpufreq_set_target()  (3)     Total:  13.02G   (  13012180339  )  Avg:  4337393446  Max:  13004736262(ts:2896.702172)  Min:3211616(ts:2881.692034)13012180339  (  13012180339  )
sum(Total):  -      -                         -       -       26.02G   (  26015321680  )
```

I can upload perf/trace-cmd dumps or decoded text reports. Or run more traces with other parameters if someone with more experience than me could shed a light on this.