openwrt/openwrt #17077: ethtool --cable-test-tdr devname fails with Timeout kernel message on Huawei AP5030DN and MikroTik RB5009UPr+S+IN

tim-seoss, 2024-11-25:
### Describe the bug

`ethtool --cable-test-tdr eth1` on Huawei AP5030DN and MikroTik RB5009UPr+S+IN fail to give full output and log:

`Timeout while waiting for cable test to finish` to kernel log.

Observed on both 23.05.5 and snapshot.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ath79/generic

### Device

Huawei AP5030DN

### Image kind

Official downloaded image

### Steps to reproduce

Install ethtool-full package, and then:
`ethtool --cable-test-tdr eth1`

### Actual behaviour

ethtool does not report TDR data, and in kernel debug messages:

`Timeout while waiting for cable test to finish`

### Expected behaviour

Full TDR data as per https://lpc.events/event/4/contributions/436/attachments/217/388/cable-diagnostics.pdf

This works with patch see link below.

### Additional info

Additional bug report from fbl here: https://forum.openwrt.org/t/mikrotik-rb5009upr-s-in-rb5009ug-with-poe/204897/20

I have a patch in development (draft patch in forum thread and below with help from upstream developer Andrew Lunn).  I'll aim to feed it upstream first once finished (first read delay timing may need reducing so that it does not slow down faster hardware - I only have a Huawei AP5030DN to test on and newer PHY chips may be ready sooner).

```
Index: linux-6.6.61/drivers/net/phy/marvell.c
===================================================================
--- linux-6.6.61.orig/drivers/net/phy/marvell.c
+++ linux-6.6.61/drivers/net/phy/marvell.c
@@ -2043,13 +2043,17 @@ static int marvell_vct5_wait_complete(st
        int i;
        int val;
 
-       for (i = 0; i < 32; i++) {
+       usleep_range(9500, 10000);
+
+       for (i = 0; i < 100; i++) {
                val = __phy_read(phydev, MII_VCT5_CTRL);
                if (val < 0)
                        return val;
 
                if (val & MII_VCT5_CTRL_COMPLETE)
                        return 0;
+
+               usleep_range(800, 1200);
        }
 
        phydev_err(phydev, "Timeout while waiting for cable test to finish\n");
```

Alternate debug version of patch:

```
Index: linux-6.6.61/drivers/net/phy/marvell.c
===================================================================
--- linux-6.6.61.orig/drivers/net/phy/marvell.c
+++ linux-6.6.61/drivers/net/phy/marvell.c
@@ -38,6 +38,9 @@
 #include <asm/irq.h>
 #include <linux/uaccess.h>
 
+/* For timing debugging only TODO remove */
+#include <linux/time.h>
+
 #define MII_MARVELL_PHY_PAGE           22
 #define MII_MARVELL_COPPER_PAGE                0x00
 #define MII_MARVELL_FIBER_PAGE         0x01
@@ -2042,14 +2045,24 @@ static int marvell_vct5_wait_complete(st
 {
        int i;
        int val;
+       u64 entrystamp, exitstamp;
+       entrystamp = ktime_get_ns();
+
+       usleep_range(9500, 10000);
+       /* msleep(10); */
 
-       for (i = 0; i < 32; i++) {
+       for (i = 0; i < 100; i++) {
                val = __phy_read(phydev, MII_VCT5_CTRL);
                if (val < 0)
                        return val;
 
-               if (val & MII_VCT5_CTRL_COMPLETE)
+               if (val & MII_VCT5_CTRL_COMPLETE) {
+                       exitstamp = ktime_get_ns();
+                       phydev_warn(phydev, "Got VCT5 ctrl data after polling %d times\n", i);
+                       phydev_warn(phydev, "Got VCT5 after %llu ns\n", exitstamp - entrystamp);
                        return 0;
+               }
+               usleep_range(800, 1200);
        }
 
        phydev_err(phydev, "Timeout while waiting for cable test to finish\n");

```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tim-seoss, 2024-11-25:
To clarify:

The PHY in use on a  here is a Marvell 88E1512 (88E1510 family) - this PHY includes TDR cable test support in the mainline Linux kernel.  As of time of writing only this PHY and derived/related Marvell Ethernet PHYs (integrated or discrete) include cable test TDR support in the mainline and OpenWrt kernels.

Other devices which this bug might impact include:

- MicroTik RB5009UG
- Turis Omnia
- Solidrun Mochabin
- Clearfog Pro and Base
- Buffalo LS421DE
- Fortinet FortiGate 50E / 51E / 52E and FortiWiFi 50E-2R
- Ctera C200 V2
- D-Link DWR-118-A2
- Shuttle OMNINAS KD22

tim-seoss, 2024-11-26:
Looks issue is confirmed with Turis Omnia too: https://forum.turris.cz/t/ethernet-cable-test-with-ethtool-cable-test-and-cable-test-tdr/20732/2

tim-seoss, 2024-11-30:
OpenWrt forum thread here:

https://forum.openwrt.org/t/ethernet-cable-test-with-raw-tdr-data-reporting-bug-fix-testers-needed/216724

Sm00shed, 2026-03-15:
Hi @tim-seoss,

I extended marvell_vct5_wait_complete() with timing measurements and tested on a MikroTik RB5009UPr+S+IN - measured ~645ms / ~205 iterations until completion. The current 100 iteration limit is not sufficient for this device - at least 250 iterations would be needed to avoid timeout.

Happy to test any updated version.