openwrt/openwrt #17095: AppleTalk DDP multicast broken - traffic not forwarded to atalkd socket

APCCV, 2024-11-27:
### Describe the bug

Netatalk v4 restores AppleTalk functionality. atalkd operates without issues on 23.05.
AppleTalk relies on a non IP protocol called DDP, layer 3, similar to IP.

On Snapshot or 24.10, DDP traffic received by OpenWrt can be seen on br-lan with tcpdump. However, it is not received by the atalkd daemon. The switch is either dropping or not flooding broadcasts like this one:
`<MAC> -> 09:00:07:ff:ff:ff (oui Unknown), 802.3, length 36: LLC, dsap SNAP (0xaa) Individual, ssap SNAP (0xaa) Command, ctrl 0x03: oui Ethernet (0x000000), ethertype Appletalk ARP (0x80f3), length 28: aarp probe 6400.206 tell 6400.206`

Identical setup works OK in 23.05.

This is a DSA issue.
A custom image built off commit https://github.com/openwrt/openwrt/commit/337e36e0ef988d850105cfa5f7588e36d90aef2b (Sep 30, 2023, conversion to DSA) does not work.
Immediately preceding commit https://github.com/openwrt/openwrt/commit/86dadeba482e2ed41f1ccc95fc7739d85a5709c0 works.

Issue tested both on ipq608x/generic (C2600) and ath79/generic (DIR 835).

### OpenWrt version

r28197-5695267847

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Official downloaded image

### Steps to reproduce

Two devices (two running atalkd daemons) required on same broadcast domain so they exchange traffic. An OpenWrt device plus atalkd running on a PC on the LAN is also an option but netatalk v4 software for the PC OS has to be sourced from netatalk's repo and built. DDP is routed fine through wireless so a PC as a client is handy.

Setup:
- install an official recent snapshot build (kmod-appletalk has to be available)
- `apk add netatalk` and `apk add kmod-appletalk` (add tcpdump for good measure)
- execute `atalk -d` on both devices (it should load the appletalk kernel module automatically, but if daemon exits with an error, modprobe it and try again)

### Actual behaviour

The two atalkd daemons will ignore each other. Exploratory packages will be displayed and activity stops.
```
root@TestWrt-65:~# atalkd -d
lo -phase 2 -net 0 -addr 0.0 -caddr 0.0
br-lan -phase 2 -net 65280-65534 -addr 0.0 -caddr 65280.0
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
packet from 65280.124 on br-lan (12) 6 (9)
06 05 00 00 00 00 00 00                         	........        
(end)
---NO MORE ACTIVITY
```
In this example` 65280.124` is an autoconfigured address for the device.
After three probes, both atalkd daemons stop posting packages to stdout because they cannot hear each other and go dormant. No packages but from `65280.124` (in this example) are displayed by this atalkd. Identical behaviour observed on second device but with a different address.
File sharing functions (out of scope of this report on how to test that) will not work.

`tcpdump -i br-lan -e | grep Appletalk` (if tcpdump is installed) will display packets from both nodes, for example:
```
00:27:03.652198 <remoteMAC> (oui Unknown) > 09:00:07:ff:ff:ff (oui Unknown), 802.3, length 36: LLC, dsap SNAP (0xaa) Individual, ssap SNAP (0xaa) Command, ctrl 0x03: oui Ethernet (0x000000), ethertype Appletalk ARP (0x80f3), length 28: aarp probe 6400.206 tell 6400.206
00:27:06.982423 <localMAC> (oui Unknown) > 09:00:07:ff:ff:ff (oui Unknown), 802.3, length 37: LLC, dsap SNAP (0xaa) Individual, ssap SNAP (0xaa) Command, ctrl 0x03: oui Appletalk (0x080007), pid Appletalk (0x809b), length 29: 65280.124.rtmp > 0.rtmp:  at-rtmp 16
```
But again, packages from <remoteMAC> above will not be displayed to stdout because atalkd is not receiving them.

### Expected behaviour

The two atalkd will react to each other's presence. One of them will take an active role. It will display packages from both instances. (`6400.147` and `1000.148` below)
```
....
packet from 6400.147 on br-lan (4fa) 1 (16)
01 19 00 08 93 19 00 80 19 00 82 03 E8 80 03 E8 	................
82                                              	.               
(end)
packet from 1000.148 on br-lan (4fa) 1 (16)
01 03 E8 08 94 03 E8 80 03 E8 82 19 00 80 19 00 	................
82                                              	.               
(end)
....
```

`tcpdump -i br-lan -e | grep Appletalk` (if tcpdump is installed) will display packets from both nodes and some of these (actual data packets, not ARP discoveries) will result in output in stdout. Activity will be displayed every few seconds.

### Additional info

This setup works in 23.05 provided a custom image is built so the kmod-appletalk package is generated.
Works on custom builds up to commit prior to conversion to DSA. Stops working on commit that converted device to DSA.

Issue also experienced with an ath79 device (DIR-835). This is not device specific.

A working 23.05 device, or a PC with a netatalk v4 build in the same broadcast domain will see packets from the two faulty devices and respond to them, which proves that all atalkd instances are generating traffic and traffic is being picked up by atalkd instances in working images. tcpdump will also show traffic from all devices.

Note: previously opened issue https://github.com/openwrt/openwrt/issues/16983 but it referenced a custom built image. The appletalk kmod is now available in snapshot so closed that one and opened this one instead.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-11-27:
@APCCV: Good investigation!

APCCV, 2024-12-04:
Discussion here.
https://forum.openwrt.org/t/dsa-and-appletalk-multicast-issues/216494/74

APCCV, 2024-12-10:
PR to address issue:
https://github.com/openwrt/openwrt/pull/17220

Neustradamus, 2024-12-28:
@APCCV: Good job and congratulations for your contribution in Linux Kernel!
- https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=a024e377efed

Linked to:
- https://lore.kernel.org/netdev/173532783475.575952.12564839557235746896.git-patchwork-notify@kernel.org/
- https://github.com/openwrt/openwrt/pull/17220

APCCV, 2024-12-28:
Patch has been merged to main. Thanks!!
Done!

closed 2024-12-28 by APCCV

Neustradamus, 2024-12-28:
@APCCV commits merged by @robimarko:
- Main: https://github.com/openwrt/openwrt/commit/da7ab64f1f6d72597a1c8a8bb3b57409227bc635
- 24.10: https://github.com/openwrt/openwrt/commit/76a546a83a68459b88a14b3382a58740daf5bf00