openwrt/openwrt #17101: extroot disk won't mount as overlay.

PanosChtz, 2024-11-28:
### Describe the bug

Downloaded the syslog image from the firmware selector. I only added `block-mount kmod-fs-ext4 e2fsprogs parted` as custom packages in the firmware selector so I can work with the extroot configuration as described in https://openwrt.org/docs/guide-user/additional-software/extroot_configuration .
After following the steps and rebooting, the external disk is not mounted as overlay.
Logs as follows.

```
root@OpenWrt:~# grep -e /overlay /etc/mtab
/dev/loop0 /overlay f2fs rw,lazytime,noatime,background_gc=on,nodiscard,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=reuse,checkpoint_merge,fsync_mode=posix,discard_unit=block 0 0
overlayfs:/overlay / overlay rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work,xino=off 0 0
root@OpenWrt:~# df /overlay /
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/loop0             7581376    196424   7384952   3% /overlay
overlayfs:/overlay     7581376    196424   7384952   3% /
root@OpenWrt:~# block info; uci show fstab; logread | sed -n -e "/- preinit -/,/- init -/p"
/dev/loop0: UUID="17d5ea7a-1dd2-11b2-ba42-36b7c3b05f2b" LABEL="rootfs_data" VERSION="1.16" MOUNT="/overlay" TYPE="f2fs"
/dev/mmcblk0p7: UUID="e11b49de-30b96036-93f56fe9-fcef9bbb" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/sda1: UUID="1a73fab8-e130-4578-9532-f9020ddf5418" LABEL="extroot" VERSION="1.0" TYPE="ext4"
fstab.@global[0]=global
fstab.@global[0].anon_swap='0'
fstab.@global[0].anon_mount='0'
fstab.@global[0].auto_swap='1'
fstab.@global[0].auto_mount='1'
fstab.@global[0].delay_root='5'
fstab.@global[0].check_fs='0'
fstab.@mount[0]=mount
fstab.@mount[0].target='/overlay'
fstab.@mount[0].uuid='17d5ea7a-1dd2-11b2-ba42-36b7c3b05f2b'
fstab.@mount[0].enabled='0'
fstab.@mount[1]=mount
fstab.@mount[1].target='/rom'
fstab.@mount[1].uuid='e11b49de-30b96036-93f56fe9-fcef9bbb'
fstab.@mount[1].enabled='0'
fstab.@mount[2]=mount
fstab.@mount[2].target='/mnt/sda1'
fstab.@mount[2].uuid='7502d4b0-7de2-4f8e-8ed6-530a322a371e'
fstab.@mount[2].enabled='0'
fstab.extroot=mount
fstab.extroot.uuid='1a73fab8-e130-4578-9532-f9020ddf5418'
fstab.extroot.target='/overlay'
fstab.rwm=mount
fstab.rwm.device='/dev/loop0'
fstab.rwm.target='/rwm'
Mon Sep 23 12:36:20 2024 user.info kernel: [    2.566907] init: - preinit -
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.722813] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.731298] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.732376] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.746651] mt7530-mdio mdio-bus:1f lan1: configuring for phy/2500base-x link mode
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    2.828696] random: procd: uninitialized urandom read (4 bytes read)
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.907974] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.948174] hub 2-1:1.0: USB hub found
Mon Sep 23 12:36:20 2024 kern.info kernel: [    2.952701] hub 2-1:1.0: 4 ports detected
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.087848] usb 1-2: new high-speed USB device number 2 using xhci-mtk
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.270772] hub 1-2:1.0: USB hub found
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.274766] hub 1-2:1.0: 4 ports detected
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.617847] usb 1-2.4: new high-speed USB device number 3 using xhci-mtk
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.793695] usb-storage 1-2.4:1.0: USB Mass Storage device detected
Mon Sep 23 12:36:20 2024 kern.info kernel: [    3.800190] scsi host0: usb-storage 1-2.4:1.0
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    4.810323] scsi 0:0:0:0: Direct-Access     Sabrent                   1.00 PQ: 0 ANSI: 6
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    4.842719] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    4.852192] sd 0:0:0:0: [sda] Write Protect is off
Mon Sep 23 12:36:20 2024 kern.debug kernel: [    4.856997] sd 0:0:0:0: [sda] Mode Sense: 37 00 00 08
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    4.863733] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Mon Sep 23 12:36:20 2024 kern.info kernel: [    4.932639]  sda: sda1
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    4.945248] sd 0:0:0:0: [sda] Attached SCSI disk
Mon Sep 23 12:36:20 2024 user.info kernel: [    4.995542] mount_root: loading kmods from internal overlay
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.004965] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.012431] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.135227] block: attempting to load /etc/config/fstab
Mon Sep 23 12:36:20 2024 user.err kernel: [    5.141222] block: unable to load configuration (fstab: Entry not found)
Mon Sep 23 12:36:20 2024 user.err kernel: [    5.147951] block: no usable configuration
Mon Sep 23 12:36:20 2024 kern.info kernel: [    5.152422] loop0: detected capacity change from 0 to 15178752
Mon Sep 23 12:36:20 2024 kern.info kernel: [    5.247844] loop0: detected capacity change from 15178752 to 15166848
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    5.307657] F2FS-fs (loop0): Mounted with checkpoint version = 8f5dccf
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.314502] mount_root: loading kmods from internal overlay
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.323704] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.332265] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.415538] block: attempting to load /etc/config/fstab
Mon Sep 23 12:36:20 2024 user.err kernel: [    5.420822] block: unable to load configuration (fstab: Entry not found)
Mon Sep 23 12:36:20 2024 user.err kernel: [    5.427528] block: no usable configuration
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.432239] mount_root: switching to f2fs overlay
Mon Sep 23 12:36:20 2024 kern.warn kernel: [    5.437608] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
Mon Sep 23 12:36:20 2024 user.debug kernel: [    5.448648] urandom-seed: Seeding with /etc/urandom.seed
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.470343] procd: - early -
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.473266] procd: - watchdog -
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.996528] procd: - watchdog -
Mon Sep 23 12:36:20 2024 user.info kernel: [    5.999954] procd: - ubus -
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    6.016336] random: ubusd: uninitialized urandom read (4 bytes read)
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    6.054392] random: ubusd: uninitialized urandom read (4 bytes read)
Mon Sep 23 12:36:20 2024 kern.notice kernel: [    6.060954] random: ubusd: uninitialized urandom read (4 bytes read)
Mon Sep 23 12:36:20 2024 user.info kernel: [    6.068303] procd: - init -

```
In the boot log I see `block: unable to load configuration (fstab: Entry not found)` which I do not know how to fix. I tried `uci set fstab.@global[0].delay_root="15"` or adding `export PREINIT=1` and `mount_root` to rc.local with no success.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-28:
It visibly complains about wrong /etc/config/fstab , try to fix or recreate that from scratch using luci.

PanosChtz, 2024-12-02:
I saw this issue which seems to be related, albeit for a different device

https://github.com/openwrt/openwrt/issues/7352

GWustenstein, 2024-12-16:
Been beating my head on this BPI-R3 trying to mount /dev/mmcblk0p6 (6.8g formatted to ext4) to /overlay, nothing seems to work, either using uci or luci.  If I do mount it, df -h shows it mounted, but the available space shows the default installation size (164M of the 435M the install reserves).  Help appreciated!