openwrt/openwrt #17103: image build fails when apk can't delete a package which hasn't been built (validly)

aimacintyre, 2024-11-28:
### Describe the bug

when building images for devices with DEVICE_PACKAGES references in image recipes that are prefixed with "-" (to _NOT_ include said packages), if the referenced packages are not otherwise built into the base+packages rootfs apk aborts with a "no package found" error when attempting to remove the package during creation of the base+packages rootfs.

### OpenWrt version

r28203-f4492b6b1c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM Fritz7490

### Image kind

Self-built image

### Steps to reproduce

check out main branch
update and install package feeds
copy diffconfig file as .config
`make defconfig`

### Actual behaviour

Build fails - end of log (from `make -j1 V=sc`):
```
{...}
/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/staging_dir/host/bin/mksquashfs4 /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/root-lantiq /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/root.squashfs -nopad -noappend -root-owned -comp xz -Xpreset 9 -Xe -Xlc 0 -Xlp 2 -Xpb 2  -b 256k -p '/dev d 755 0 0' -p '/dev/console c 600 0 0 5 1' -no-xattrs
Pseudo file "dev" exists in source filesystem "/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/root-lantiq/dev".
Ignoring, exclude it (-e/-ef) to override.
Parallel mksquashfs: Using 12 processors
Creating 4.0 filesystem on /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/root.squashfs, block size 262144.

[|                                                               ]   0/762   0%
[===============================================================-] 762/762 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 3735.81 Kbytes (3.65 Mbytes)
	25.36% of uncompressed filesystem size (14731.11 Kbytes)
Inode table size 7886 bytes (7.70 Kbytes)
	21.91% of uncompressed inode table size (35998 bytes)
Directory table size 10974 bytes (10.72 Kbytes)
	50.06% of uncompressed directory table size (21921 bytes)
Number of duplicate files found 7
Number of inodes 1080
Number of files 753
Number of fragments 56
Number of symbolic links 212
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 114
Number of hard-links 0
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
rm -rf /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/target-dir-a5ccb0e7
cp -fpR /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/root.orig-lantiq /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/target-dir-a5ccb0e7
IPKG_INSTROOT=/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/target-dir-a5ccb0e7 /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/staging_dir/host/bin/fakeroot /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/staging_dir/host/bin/apk --root /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/target-dir-a5ccb0e7 --keys-dir /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490 --no-logfile --preserve-env --no-scripts --repositories-file /dev/null --repository file:///home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/staging_dir/packages/lantiq/packages.adb del kmod-owl-loader
ERROR: No such package: kmod-owl-loader
make[5]: *** [/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/include/image.mk:375: target-dir-a5ccb0e7] Error 1
make[5]: Leaving directory '/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/target/linux/lantiq/image'
make[4]: *** [Makefile:20: install] Error 2
make[4]: Leaving directory '/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/target/linux/lantiq'
make[3]: *** [Makefile:12: install] Error 2
make[3]: Leaving directory '/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/target/linux'
time: target/linux/install#29.23#9.49#32.39
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/install] Error 1
make[2]: Leaving directory '/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490'
make[1]: *** [target/Makefile:26: /home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490'
make: *** [/home/andymac/Dev/OpenWrt/main/openwrt-f4492b6b1c-fb7490/include/toplevel.mk:248: world] Error 2
```

### Expected behaviour

build successfully completes

### Additional info

The image recipe for the avm_fritz7490 device has a DEVICE_PACKAGES entry of `-kmod-owl-loader`; this package not being required on this device though normally included on other AVM devices in the same subtarget.  If the build .config is generated from a diffconfig only including devices not requiring this package (see diffconfig below) the .config will have `# CONFIG_PACKAGE_kmod-owl-loader is not set` so the package won't be built.

### Diffconfig

```text
CONFIG_TARGET_lantiq=y
CONFIG_TARGET_lantiq_xrx200=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_lantiq_xrx200_DEVICE_avm_fritz7490-micron=y
CONFIG_TARGET_DEVICE_lantiq_xrx200_DEVICE_avm_fritz7490=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-11-28:
Please try:
https://github.com/openwrt/openwrt/pull/17106

aimacintyre, 2024-11-29:
@robimarko #17106 works for me to fix this

commit 55dfe3fc (2024-11-29): image: apk: dont fail on removing a package that doesnt exist

Currently, in case when a package in the device package list is marked with
the "-"" prefix meaning that its supposed to be removed for that device but
it was not actually previously built it will cause the APK to error out it
and thus image generation will also fail.

So, lets do what OPKG build step does and supress the error with the "-"
prefix in make.

Fixes: #17103
Fixes: d788ab376f85 ("build: add APK package build capabilities")
Link: https://github.com/openwrt/openwrt/pull/17106
PR #17106 (2024-11-29): image: apk: dont fail on removing a package that doesnt exist
Currently, in case when a package in the device package list is marked with the "-" prefix meaning that its supposed to be removed for that device but it was not actually previously built it will cause the APK to error out it and thus image generation will also fail.

So, lets do what OPKG build step does and supress the error with the "-" prefix in make.

Fixes: #17103
Fixes: d788ab376f85 ("build: add APK package build capabilities")
closed 2024-11-29 by openwrt-bot