openwrt/openwrt #17108: apk: scripts tar file a lot bigger than it needs to be

efahl, 2024-11-28:
### Describe the bug

I just noticed this one and thought it would be good to log it here before I forget.

The `scripts.tar` file is mostly null padding.  Can we gzip it?  Or somehow get tar to reduce block size or ???  Seems like an easy 250k (or more) savings...

```
# ll /lib/apk/db/scripts.tar
-rw-r--r--    1 root     root        311296 Nov 27 00:42 /lib/apk/db/scripts.tar

# tar tvf /lib/apk/db/scripts.tar | awk '{x += $3} END {print x" bytes"}'
50450 bytes
```

### OpenWrt version

r28200-c857145e03

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-11-28:
I guess we could make APK compress it since its already supports gzip AFAIK

efahl, 2025-11-05:
Just saw this again when reviewing some other things, and somehow it appears to be even worse than before.  It's not so bad when it's in squashfs `/rom`, but the first time you add or remove a package, it goes into `/overlay` and explodes...

```
$ cat /lib/apk/db/scripts.tar | wc -c
624640

$ gzip -c /lib/apk/db/scripts.tar | wc -c
24973
```

robimarko, 2025-11-07:
Yeah, please open an issue there, there is likely some workaround

efahl, 2025-11-09:
https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11154

efahl, 2025-11-14:
@fabled got us an implementation in record time (👍🏼), so I've been doing preliminary testing on both host and x86 device.  (Preliminary as in: I've only applied the specific changes as a patch without any other changes since our last version bump to apk-tools-3.0.0_rc6; everything is working as expected.)

https://gitlab.alpinelinux.org/alpine/apk-tools/-/commit/012cdcfdf981453b572a70c9568e8df6d0fe0ca1

@robimarko 
Just one question for you:
Should we wait for rc9 and then grab that, or should I update apk-tools now to this commit?

This commit missed the apk-tools rc8 tag by a few days, so looking at their cadence, might be a month or two (or a lot less, it all depends) for the rc9...

Subsequent to this making it into the code, changes to our build are pretty trivial, I've got it working with a one-liner at
https://github.com/openwrt/openwrt/blob/971c058cf55f85a295378031ba7f7435cb2a0f6a/include/rootfs.mk#L94
Simply zip scripts.tar in the rootfs after we've finished creating it:
```
                done; \
-               if [ -z "$(CONFIG_USE_APK)" ]; then \
+               if [ -n "$(CONFIG_USE_APK)" ]; then \
+                       gzip -9 ./lib/apk/db/scripts.tar; \
+               else \
                        $(if $(IB),,awk -i inplace \
```

robimarko, 2025-11-15:
@efahl Why not simply backport this change and get this into OpenWrt ASAP as this is a problem for the next release as well

efahl, 2025-11-15:
Will do, PR incoming...

efahl, 2025-11-15:
> this is a problem for the next release as well

Ha, I forgot to hit "Comment" yesterday, when I was going to say: Yes, my major concern is that 25.x would sprint ahead and this would get missed, then I'd have to explain why removing a package caused "disk full"(!) to someone.