openwrt/openwrt #17112: "Luci Network->Associated Stations" and "iwinfo assoclist" include non-authenticated stations

joe-sydney, 2024-11-29:
### Describe the bug

In the process of tracking down stations that appear in Luci "Network->Associated Stations" as well as "winfo assoclist", I found that non-authenticated stations were also included. 

It seems like some rogue stations were trying to connect to my AP, and although they were not authenticated (I also changed Wifi WPA2 passwords on the AP but the same behaviour occured). 

Is this expected? I was expecting that only after the station had successfully been authenticated (with the right password), would it be considered associated and appear as an Associated Station and in the assoclist.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAX202

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-11-29:
It is certainly outside activity, not OpenWRT on its own.
There are some benign explanations too, like dictionary access point name and some client of other AP with same name saved is nearby.

Best would be to discuss this in forums.

 minimal checklist:
- make log larger luci/system/system - change 64 to like 1024
- add new WPA3 or WPA2+CCMP and setting at least optional 11w in with random MAC and new, non-dictionary name and relatively long password neither related to you or the place without connecting it to LAN network, do not enable unless you found all parameters and are sure they are set.
- ? does it start immediately get associations? YES? somebody or something is up to no good.
- No hits?  - connect new AP to network, connect it from at least couple of clients and disconnect old AP from network, maybe remove.
- you can use luci-app-attendedsysupgrade to easily upgrade to 23.05.5

brada4, 2024-11-29:
It is outside interference, not OpenWRT's own making. Will not go with update, I am just trying to help you figure out if display is caused by evil or carelessness.