openwrt/openwrt #17119: "Segmentation fault" error when flashing openWRT factory image to RT-AX54 (ASUS RTAX-1800S)

zuritero, 2024-11-30:
### Describe the bug

Hello,
I attempted to flash the openWRT factory firmware via mtd-write, but I get a "Segmentation fault" message in the terminal. This is on a brand new ASUS RT-AX1800S unit.

Some details:
Model: ASUS RT-AX1800S V2
PN: 901G08X0-MA1C0V
openWRT Factory code: openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin

### OpenWrt version

Unable to run

### OpenWrt release

Unable to run

### OpenWrt target/subtarget

Unable to run

### Device

Unable to run

### Image kind

Official downloaded image

### Steps to reproduce

copy the firmware file from computer to router
`cat openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory | ssh admin@device "cat > /root/openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory"`

ssh in to router
`shh admin@192.168.50.1`

Flash the firmware
`mtd-write -i openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory -d Kernel`

### Actual behaviour

Message "Segmentation fault" is displayed on the terminal a few seconds after executing `mtd-write` command

### Expected behaviour

First time doing this (flashing firmware), so expect some message indicating success, and also openWRT firmware to load next time router is power cycled.

### Additional info

My understanding is that the RT-AX1800S is supported by openWRT AX54 firmware. This is a brand new router, not sure if it comes with a different memory chip which may not be compatible with the released of openWRT mentioned

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jdmonin, 2024-12-07:
fyi @zuritero , I think I bought the same router as you; the retail box says it's `RT-AX1800S V2  H/W Ver: A1` and it looks like they changed the type of CPU architecture, this openwrt image probably won't work at all

zuritero, 2024-12-07:
> fyi @zuritero , I think I bought the same router as you; the retail box says it's `RT-AX1800S V2 H/W Ver: A1` and it looks like they changed the type of CPU architecture, this openwrt image probably won't work at all

Hi @jdmonin, indeed the box says that. Unfortunately, you don't know which version you may get when ordering online (e.g. amazon). I don't remember either reading anything specific about V1 vs V2 in the openWRT site for this router. Thank you for the feedback.

jdmonin, 2024-12-08:
Same here; Amazon doesn't warn you. I thought I was buying a very supported device based on all the info on its openWRT device page. Ah well :)

theofficialgman, 2026-01-07:
The PR for 25.12 was merged and a new pre-release has been cut that includes support https://github.com/openwrt/openwrt/releases/tag/v25.12.0-rc2

https://firmware-selector.openwrt.org/?version=25.12.0-rc2&target=mediatek%2Ffilogic&id=asus_rt-ax57m