openwrt/openwrt #17126: armsr: 24.10 SNAPSHOT combined image is using ext4 instead of f2fs

AlbrechtL, 2024-11-30:
### Describe the bug

I realized that the image [`generic-squashfs-combined.img.gz`](https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/armsr/armv8/openwrt-24.10-snapshot-r28155-17bdccb4a5-armsr-armv8-generic-squashfs-combined.img.gz) is using ext4 as writable filesystem. In the 23.05 and the master branch, the image `generic-squashfs-combined.img.gz` is using f2fs.

I don't know if this behavior is intended, so I'm reporting it :-).

### OpenWrt version

r28155-17bdccb4a5

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

armsr/armv8

### Device

QEMU QEMU Virtual Machine

### Image kind

Official downloaded image

### Steps to reproduce

Download  [`generic-squashfs-combined.img.gz`](https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/armsr/armv8/openwrt-24.10-snapshot-r28155-17bdccb4a5-armsr-armv8-generic-squashfs-combined.img.gz), boot it and type `mount`.

### Actual behaviour

```
root@OpenWrt:~# mount
[...]
/dev/loop0 on /overlay type ext4 (rw,noatime)
[...]
```

### Expected behaviour

```
root@OpenWrt:~# mount
[...]
/dev/loop0 on /overlay type f2fs
[...]
```

### Additional info

I discovered this issue because my script https://github.com/AlbrechtL/openwrt-docker/blob/master/src/mount_openwrt_squashfs_combined.sh was not working properly.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2024-11-30:
How large is your rootfs(_data) partition? Filesystem is supposed to be selected based on the available space for the rootfs_data partition: if it's smaller than 100 MiB ext4 will be used, else f2fs.

AlbrechtL, 2024-12-01:
Thanks @KanjiMonster for the explanation.

Following you will find the sizes. For some reason the squashfs filesystem and the writable filesystem are located in the same partition.

**OpenWrt 24.10 SNAPSHOT r28155-17bdccb4a5**
[openwrt-24.10-snapshot-r28155-17bdccb4a5-armsr-armv8-generic-squashfs-combined.img](https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/armsr/armv8/openwrt-24.10-snapshot-r28155-17bdccb4a5-armsr-armv8-generic-squashfs-combined.img.gz)

```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.3M      4.3M         0 100% /rom
efivars                 255.9K      5.9K    250.0K   2% /sys/firmware/efi/efivars
tmpfs                   113.6M      1.1M    112.4M   1% /tmp
/dev/loop0               88.0M      1.3M     79.7M   2% /overlay
overlayfs:/overlay       88.0M      1.3M     79.7M   2% /
tmpfs                   512.0K         0    512.0K   0% /dev
```
```
root@OpenWrt:~# fdisk -l /dev/vda
GPT PMBR size mismatch (475679 != 475710) will be corrected by write.
The backup GPT table is not on the end of the device.
Disk /dev/vda: 232.28 MiB, 243564032 bytes, 475711 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: C3F7A169-25A7-0A0F-F23C-0323455A5800

Device       Start    End Sectors  Size Type
/dev/vda1      512 262655  262144  128M EFI System
/dev/vda2   262656 475647  212992  104M Linux filesystem
/dev/vda128     34    511     478  239K BIOS boot

Partition table entries are not in disk order.
```

**OpenWrt 23.05.5**
[openwrt-23.05.5-armsr-armv8-generic-squashfs-combined.img](https://archive.openwrt.org/releases/23.05.5/targets/armsr/armv8/openwrt-23.05.5-armsr-armv8-generic-squashfs-combined.img.gz)
```
root@OpenWrt:/# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.0M      4.0M         0 100% /rom
tmpfs                   114.2M     32.0K    114.2M   0% /tmp
/dev/loop0               98.1M     38.2M     59.9M  39% /overlay
overlayfs:/overlay       98.1M     38.2M     59.9M  39% /
tmpfs                   512.0K         0    512.0K   0% /dev
```
```
root@OpenWrt:/# fdisk -l /dev/vda
GPT PMBR size mismatch (475679 != 475710) will be corrected by write.
The backup GPT table is not on the end of the device.
Disk /dev/vda: 232.28 MiB, 243564032 bytes, 475711 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 14ACDE48-0F72-A275-153D-6740D828F000

Device       Start    End Sectors  Size Type
/dev/vda1      512 262655  262144  128M EFI System
/dev/vda2   262656 475647  212992  104M Linux filesystem
/dev/vda128     34    511     478  239K BIOS boot

Partition table entries are not in disk order.
```

**OpenWrt master r28214-5a4eb56a7b**
[openwrt-armsr-armv8-generic-squashfs-combined.img](https://downloads.openwrt.org/snapshots/targets/armsr/armv8/openwrt-armsr-armv8-generic-squashfs-combined.img.gz)

```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.0M      4.0M         0 100% /rom
efivars                 255.9K      5.9K    250.0K   2% /sys/firmware/efi/efivars
tmpfs                   113.0M      1.4M    111.5M   1% /tmp
/dev/loop0               98.1M     39.9M     58.2M  41% /overlay
overlayfs:/overlay       98.1M     39.9M     58.2M  41% /
tmpfs                   512.0K         0    512.0K   0% /dev
```
```
root@OpenWrt:~# fdisk -l /dev/vda
GPT PMBR size mismatch (475679 != 475710) will be corrected by write.
The backup GPT table is not on the end of the device.
Disk /dev/vda: 232.28 MiB, 243564032 bytes, 475711 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 3C5488D9-39E9-C1CF-C55D-1FEA5E9DA300

Device       Start    End Sectors  Size Type
/dev/vda1      512 262655  262144  128M EFI System
/dev/vda2   262656 475647  212992  104M Linux filesystem
/dev/vda128     34    511     478  239K BIOS boot

Partition table entries are not in disk order.
```

AlbrechtL, 2024-12-03:
Now I think how it happens. Since the total disk image size is fixed and thus new software, the squashfs part becomes bigger the remaining disk space is lower. Because the master image doesn't include LuCI there is more than 100 MB space left and f2fs is used. Same for 23.05 because the software components are smaller there. In 24.10 we reached the kipping point where the remaining disk size is lower than 100 MB.

Maybe it is worth to think about to increase the total image size. I think for x86_64 and armsr based systems the flash or hard disk size doesn't so much. As long as we are lower than 1 GB we are fine I think.