openwrt/openwrt #17131: VLAN | Wireless client traffic (except BOOTP/DHCP) cannot pass uplink interface and vice versa

codiflow, 2024-12-01:
### Describe the bug

I own a [TP-Link EAP615-Wall v1](https://openwrt.org/toh/tp-link/eap615-wall) with the [official and up-to-date 23.05.4 image](https://openwrt.org/releases/23.05.4) of OpenWRT.

First of all: I tried **many different setups** and variants for hours now and the setup itself did **never** work.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link EAP615-Wall v1

### Image kind

Official downloaded image

### Steps to reproduce

I **factory resetted** the device.

I added all physical network ports (`lan0` - `lan3`) to a bridge (`br-router`) and activated VLAN filtering:
[image]

Then I configured the automatically added `br-router.10` device and disabled IPv6 there.

Then I set up the interface `lan` and assigned the `br-router.10` interface to it. The protocol is set to `DHCP client`. IPv6 is disabled here as well and the firewall zone `lan` is assigned.
=> This is my **management interface** for the firewall.

I then added another interface `wifi` and assigned the `br-router.10` interface to it as well. The protocol is set to `Unmanaged`.IPv6 is disabled here as well and also the firewall zone `lan` is assigned.
=> This is the interface the wireless clients should use (could have been the `lan` interface as well but for the ease of understanding I've choosen this way)

Now I added a wireless network and hooked it up to the interface `wifi`.

The EAP615-Wall is **only routing the connected devices**, there is **no DHCP/DNS** or anything else configured on the device.

Now **all physically connected devices** are directly getting their IP address according to the configured VLAN from the uplink main router, can ping it and vice versa. So everything is fine here. The same goes for the EAP615-Wall itself.

BUT:

### Actual behaviour

**All wireless clients** are only able to ping the EAP615-Wall and other directly connected devices within the same VLAN connected to the EAP615-Wall. The upstream connection to the main router **is not working** properly for wireless clients. The strange part is that the wireless clients **get their IP configuration** from the main router using this connection and **this part is working** (it's the only part). I already checked firewalls in between but as the wired clients are using the same connection there's no traffic being blocked on main router side.

### Expected behaviour

Traffic is being forwarded for the wireless clients like it is for the wired clients in the same VLAN.

### Additional info

## What I've tried so far
- Ping from **wireless client** to **EAP615-Wall** ✅
- Ping from **wireless client** to other **wired clients** in same VLAN connected to EAP615-Wall ✅
- Ping from other **wired clients** in same VLAN connected to EAP615-Wall to **wireless client** ✅
- Ping from **EAP615-Wall** to **wireless client** ✅
- Ping from **wireless client** to **main router** ❌
- Ping from **wireless client** to other wired clients in same VLAN **not** connected to **EAP615-Wall** ❌
- Ping from **main router** to **wireless client** ❌

Switching the OpenWRT firewall "off", changing firewall zones, accepting ingress and forwarding traffic in general, rebooting the router after the configuration was finished. All this led to **no changes in behaviour**.

I then started to monitor whats going on at interface side using `tcpdump`:

On the main routers interface (the one where all vlans are aggregated) I was using this command:
`tcpdump -v -i <INTERFACE_NAME> ether host <MAC_ADDRESS_OF_WIRELESS_CLIENT>`

Results:
The only packets arriving there were BOOTP/DHCP (UDP) from the wireless client. This tells me the connection and the VLAN setup must be correct and working. **Am I right here?** 🤔

On the EAP615-Walls bridge `br-router` I was using this command:
`tcpdump -v -i br-router port not <SSH_PORT_TO_WIRELESS_CLIENT> and ether host <MAC_ADDRESS_OF_WIRELESS_CLIENT> and ip`

> Note: As I'm one of the wired clients connected to the EAP615-Wall I can connect to the wireless client via SSH and had to exclude the packets from the output.

Result:
Here I see all ping packets to other wired clients in same VLAN connected to **EAP615-Wall** and to and from the EAP615-Wall itself.

But if I ping the main router or some targets in the internet, I **don't get an output on any of the tcpdump sessions**.

### Conclusion
There must be some kind of "packet forward" issue within OpenWRT as the traffic never leaves OpenWRT. The same goes for ingress traffic from the main router to devices **connected wirelessly** to EAP615-Wall. The default routes are being set correctly on the wireless client side as the DHCP leases have been assigned from the main router.

I found some issues about this which made me think about this being a bug:
https://github.com/openwrt/openwrt/issues/16314
https://github.com/openwrt/openwrt/issues/14473

### Another note
I don't know why some of these packets are being logged on EAP615-Wall side from time to time – I cannot reproduce this as neither `dig @<IP-OF-MAIN-ROUTER> openwrt.org` produces such output nor does `netcat -u  <MAIN-ROUTER-IP> 53`

```
02IP (tos 0x0, ttl 64, id 43348, offset 0, flags [DF], proto UDP (17), length 67)
    <WIRELESS-CLIENT-IP>.41800 > <MAIN-ROUTER-IP>.53: 63596+ A? 0.debian.pool.ntp.org. (39)
IP (tos 0x0, ttl 64, id 43349, offset 0, flags [DF], proto UDP (17), length 67)
    <WIRELESS-CLIENT-IP>.41800 > <MAIN-ROUTER-IP>.53: 15528+ AAAA? 0.debian.pool.ntp.org. (39)
IP (tos 0x0, ttl 64, id 10990, offset 0, flags [DF], proto TCP (6), length 60)
    <WIRELESS-CLIENT-IP>.39774 > <ANOTHER-MAIN-ROUTER-IP>.3551: Flags [S], cksum 0x84ce (correct), seq 660595127, win 64240, options [mss 1460,sackOK,TS val 506829142 ecr 0,nop,wscale 7], length 0
IP (tos 0x0, ttl 64, id 10991, offset 0, flags [DF], proto TCP (6), length 60)
    <WIRELESS-CLIENT-IP>.39774 > <ANOTHER-MAIN-ROUTER-IP>.3551: Flags [S], cksum 0x8096 (correct), seq 660595127, win 64240, options [mss 1460,sackOK,TS val 506830222 ecr 0,nop,wscale 7], length 0
IP (tos 0x0, ttl 64, id 43350, offset 0, flags [DF], proto UDP (17), length 67)
    <WIRELESS-CLIENT-IP>.41800 > <MAIN-ROUTER-IP>.53: 63596+ A? 0.debian.pool.ntp.org. (39)
IP (tos 0x0, ttl 64, id 43351, offset 0, flags [DF], proto UDP (17), length 67)
    <WIRELESS-CLIENT-IP>.41800 > <MAIN-ROUTER-IP>.53: 15528+ AAAA? 0.debian.pool.ntp.org. (39)
IP (tos 0x0, ttl 64, id 10992, offset 0, flags [DF], proto TCP (6), length 60)
    <WIRELESS-CLIENT-IP>.39774 > <ANOTHER-MAIN-ROUTER-IP>.3551: Flags [S], cksum 0x7876 (correct), seq 660595127, win 64240, options [mss 1460,sackOK,TS val 506832302 ecr 0,nop,wscale 7], length 0
IP (tos 0x0, ttl 64, id 1306, offset 0, flags [DF], proto UDP (17), length 62)
    <WIRELESS-CLIENT-IP>.45144 > <MAIN-ROUTER-IP>.53: 55973+ A? api.telegram.org. (34)
IP (tos 0x0, ttl 64, id 1307, offset 0, flags [DF], proto UDP (17), length 62)
    <WIRELESS-CLIENT-IP>.45144 > <MAIN-ROUTER-IP>.53: 40343+ AAAA? api.telegram.org. (34)
IP (tos 0x0, ttl 64, id 26942, offset 0, flags [DF], proto TCP (6), length 60)
    <WIRELESS-CLIENT-IP>.39776 > <ANOTHER-MAIN-ROUTER-IP>.3551: Flags [S], cksum 0x4ff1 (correct), seq 2518848582, win 64240, options [mss 1460,sackOK,TS val 506834143 ecr 0,nop,wscale 7], length 0
IP (tos 0x0, ttl 64, id 26943, offset 0, flags [DF], proto TCP (6), length 60)
    <WIRELESS-CLIENT-IP>.39776 > <ANOTHER-MAIN-ROUTER-IP>.3551: Flags [S], cksum 0x4be2 (correct), seq 2518848582, win 64240, options [mss 1460,sackOK,TS val 506835182 ecr 0,nop,wscale 7], length 0
```

The mentioned packets here which are heading to the main router are never shown on main router side in tcpdump.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

codiflow, 2024-12-01:
Here are some additional config file dumps:

`/etc/config/firewall`

```
config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option synflood_protect '1'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	list network 'wifi'
	list network 'lan'
```

`/etc/config/network`

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd4a:8d83:9bf6::/48'
	option packet_steering '1'

config device
	option type 'bridge'
	option name 'br-router'
	option ipv6 '0'
	list ports 'lan0'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'

config bridge-vlan
	option device 'br-router'
	option vlan '10'
	list ports 'lan0:t*'
	list ports 'lan2'
	list ports 'lan3'

config bridge-vlan
	option device 'br-router'
	option vlan '9'
	list ports 'lan0:t'
	list ports 'lan1'

config interface 'wifi'
	option proto 'none'
	option device 'br-router.10'
	option type 'bridge'

config device
	option name 'br-router.10'
	option type '8021q'
	option ifname 'br-router'
	option vid '10'
	option ipv6 '0'

config interface 'lan'
	option proto 'dhcp'
	option device 'br-router.10'

config route
	option interface 'lan'
	option target '0.0.0.0/0'
	option gateway '<MAIN-ROUTER-IP>'
	option disabled '1'

config device
	option name 'br-router.9'
	option type '8021q'
	option ifname 'br-router'
	option vid '9'
	option ipv6 '0'

config route
	option interface 'wifi'
	option target '0.0.0.0/0'
	option gateway '<MAIN-ROUTER-IP>'
	option disabled '1'
```

`/etc/config/wireless`

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option country 'DE'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option mode 'ap'
	option ssid '<MY-SSID>'
	option encryption 'sae-mixed'
	option key '<MY-PASSPHRASE>'
	option wpa_disable_eapol_key_retries '1'
	option network 'wifi'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option mode 'ap'
	option ssid '<MY-SSID>'
	option encryption 'sae-mixed'
	option key '<MY-PASSPHRASE>'
	option wpa_disable_eapol_key_retries '1'
	option network 'wifi'
```

e9hack, 2024-12-03:
Can you show the output of `brctl show` and `bridge vlan`?

codiflow, 2024-12-03:
Sure.

`brctl show`

```
bridge name	bridge id		STP enabled	interfaces
br-router		7fff.788cb564213c	no		lan2
							lan0
							phy0-ap0
							lan3
							lan1
```

`bridge vlan`

`-ash: bridge: not found`

e9hack, 2024-12-04:
I compile the images by myself. I don't know which setting enables the bridge command.
I'm using a completely different setup but becomes the same result.

It looks like that ether hostapd or some other component is not able to set vlan ids correctly.

The output of both commands shows the following:

```
brctl show | grep 201
br-guest1               7fff.xxxxxxxxxxxx       no              2G-ap2.201
```

```
bridge vlan | grep 201
br-guest1         201
2G-ap2.201        1 PVID Egress Untagged
eth0.201          1 PVID Egress Untagged
```

I'm using a hotplug script which corrects the vlan ids from 1 to 201 which results in this:

```
bridge vlan | grep 201
br-guest1         201
2G-ap2.201        201 PVID Egress Untagged
eth0.201          201 PVID Egress Untagged
```

codiflow, 2024-12-04:
After that I disabled the static IPv4 route and rebooted the router. After the reboot the routes were just like they should be – set up automatically:

[image]

Maybe it was my fault because I had to set up the static route during my tests and after disabling the static route (which overlaps with the one being automatically set) it removes this one without further notice 😅

But this is just an unrelated side story – should not have anything to do with my wireless connection problems.

codiflow, 2024-12-16:
According to a hint from the IRC channel #openwrt-devel I installed `OpenWrt 23.05.5, r24106-10cc5fcd00` from [here](https://mirror-03.infra.openwrt.org/releases/23.05.5/targets/ramips/mt7621/openwrt-23.05.5-ramips-mt7621-tplink_eap615-wall-v1-squashfs-sysupgrade.bin) which is not listed on the official device page (yet).

But unfortunately **nothing has changed** so far. Maybe I will also try 24.10-rc2 but currently I'm totally in the dark with debugging this. So every hint is appreciated. 🙏

codiflow, 2024-12-16:
OK now it's getting interesting.

I just took the risk to install `24.10.0-rc2` from remote (yeah, device still available 🥳) and the result of `bridge vlan` now looks different:

```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
br-router         9
                  10
br-wifi           1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
```

If that's the "real" setting I understand why it does not work. I'll keep everyone posted here about my progress.

codiflow, 2024-12-16:
So it seems like after upgrading to 24.10-rc2 a new bridge called `br-wifi` was created **automatically**:

`brctl show`
```
bridge name	bridge id		STP enabled	interfaces
br-router		7fff.788cb564213c	no		lan2
							lan0
							lan3
							lan1
br-wifi		7fff.788cb564213c	no		phy0-ap0
```

But also after configuring everything according to the values above nothing has changed unfortunately. ☹️

`bridge vlan`
```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
br-router         9
                  10
phy0-ap0          10 PVID Egress Untagged
```

I also tried to use `br-wifi` instead of `br-router.10` as device for the interface `wifi` but the problem persists:

`bridge vlan`
```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
br-router         9
                  10
br-wifi           10
phy0-ap0          10 PVID Egress Untagged
```

codiflow, 2024-12-16:
All the following steps are done on 24.10-rc2:

Something seems to be completely weird with configuring networks 🤪🤪🤪

I tried to configure a setting on the device page in device `phy0-ap0` which is the wireless device. After the device has been configured, it **vanished** from `brctl show`:

```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
br-router         9
                  10
```

Maybe this is normal (i dunno), so I **unconfigured** the device again and applied. My understanding would be the device now reappears, but that was not the case.

So I changed the device used on interface `wifi` from `br-router.10` to `br-router.9`. Now the output of `brctl show` showed the device again:

```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
phy0-ap0          9 PVID Egress Untagged
br-router         9
                  10
```

And after changing it to the right value `br-router.10` it became more weird:

```
port              vlan-id  
lan0              9
                  10 PVID
lan3              10 PVID Egress Untagged
lan2              10 PVID Egress Untagged
lan1              9 PVID Egress Untagged
phy0-ap0          9 Egress Untagged
                  10 PVID Egress Untagged
br-router         9
                  10
```

Now the wireless device is internally "connected" to both VLANs but that is not how I configured it in the GUI.

Seems like this is a bug in 24.10-rc2 because this even removing the interface `wifi` does not change this.

codiflow, 2025-04-26:
I bought a new **GL.iNET GL-MT6000 Flint 2** (with most recent vanilla OpenWrt 24.10.1 installed) and configured it the same way like the [TP-Link EAP615-Wall v1](https://openwrt.org/toh/tp-link/eap615-wall).

Result: **Same issue.**

So it must be either a wrong configuration / a bug in OpenWRT or a general issue with my network.

The latter is not very likely as the wired clients including VLAN work like the should.

**But**: Seems like I found a pattern which could lead to the real issue:

All **wired clients** are configured like this and they work just fine:

`Network` => `Devices` => `br-router` => `Configure...` => `Bridge VLAN filtering`

![](https://private-user-images.githubusercontent.com/43847817/391299544-af5c2915-7417-4dd7-af01-639248a1f760.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU2NzkzOTQsIm5iZiI6MTc0NTY3OTA5NCwicGF0aCI6Ii80Mzg0NzgxNy8zOTEyOTk1NDQtYWY1YzI5MTUtNzQxNy00ZGQ3LWFmMDEtNjM5MjQ4YTFmNzYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI2VDE0NTEzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhMTY4NGViZjA4ODA2YmQ0ZTQ3MDIyNWEzOTRlNTg1M2Y0MmY0OTViOTNiNzk4NzEzNzYxMjBiOGNkZWNjZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NR3QXyFkiPgzw6pPikn1dR50ZOgKkemLPMdo7DJaO3k)

**BUT** the TP-Link EAP615-Wall v1 itself (so the CPU or what it is called in OpenWRT business) **AND** the wirelessly attached clients are using another device and they **don't work**. So the router **cannot** resolve domains, **cannot** ping stuff which is outside of itself (for example my whole network), BUT they all **DHCP leases** from outside just fine.

They are all using the device `br-router.10` which is a VLAN device for VLAN 10:

[image]

[image]

[image]

Maybe this helps with tracking down what the issue is?

codiflow, 2025-04-26:
Turned out that the cause was a **MAC Address filter** configured on the switch a longer time ago for this specific port 🤦‍♂️🤦‍♂️🤦‍♂️

When security drives you mad and brings up problems which would have never been there without it 😅

Thanks for all who helped with this non-issue! 💪

closed 2025-04-26 by codiflow