openwrt/openwrt #17139: [Snapshot] AX3000T no CPU clock management

erayrafet, 2024-12-01:
### Describe the bug

I have the Xiaomi AX3000T and there is no CPU clock management. The CPU directories in /dev do not exist as btop doesn't display clock speeds either. Because of that, the CPU is always hovering at about 60C which is quite hot on the outside as well. Seems like cpufreq isn't enabled.

https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/1782?u=erayrafet

https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/1783?u=erayrafet

[image]
[image]

### OpenWrt version

r28217-af611bce44

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Self-built image

### Steps to reproduce

No special steps necessary. Snapshot builds are from the Firmware Selector.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2024-12-02:
It not supported

RomanovDanila, 2024-12-02:
On SDM662 SOC The difference between 2x cortex-A53@1.3 and cortex-A53@0.6 is around 50mWt

erayrafet, 2024-12-02:
Well, this chip is cooking itself.

RomanovDanila, 2024-12-02:
Also, overclocking mt7981 cores from 1.3 to 1.7 doesn't change temperatures

lukasz1992, 2024-12-02:
MT7988 include driver to change cpu frequency, but there is no such for mt7981 available.

kszaq, 2024-12-03:
Possible duplicate of https://github.com/openwrt/openwrt/issues/13376

RomanovDanila, 2024-12-06:
@libiunc It could be changed in BL2. But there is no difference between idle temps at 850 and 1700MHz