openwrt/openwrt #17142: [24.10.0-rc1] memcpy warning in drv_mei_cpe on ipq40xx VRX518 (Fritzbox 7520)

octomike, 2024-12-02:
### Describe the bug

boot shows Warning and trace for `memcpy: detected field-spanning write` in `linux-ipq40xx_generic/dsl_cpe_mei-ugw_8.5.2.10/src/drv_mei_cpe_msg_process.c:3570`

### OpenWrt version

r28158-d276b4c91a

### OpenWrt release

24.10.0-rc1

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530

### Image kind

Self-built image

### Steps to reproduce

Build 24.10.0-rc1 with patch from #15421 (Fritzbox 7520 modem hack discussed in PR)

### Additional info

```
------------[ cut here ]------------
WARNING: CPU: 2 PID: 1712 at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/dsl_cpe_mei-ugw_8.5.2.10/src/drv_mei_cpe_msg_process.c:3570 MEI_IoctlCmdMsgWrite+0x290/0x2c8 [drv_mei_cpe]
memcpy: detected field-spanning write (size 4) of single field "pDestPtr" at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/dsl_cpe_mei-ugw_8.5.2.10/src/drv_mei_cpe_msg_process.c:3570 (size 2)
Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat(O) clip drv_dsl_cpe_api(O) drv_mei_cpe(O) br2684 vrx518_tc(O) atm vrx518(O) drv_ifxos(O) sha512_arm md5 ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug(O) crc32c_generic
CPU: 2 PID: 1712 Comm: evnthnd Tainted: G           O       6.6.63 #0
Hardware name: Generic DT based system
 unwind_backtrace from show_stack+0x10/0x14
 show_stack from dump_stack_lvl+0x40/0x4c
 dump_stack_lvl from __warn+0x94/0xbc
 __warn from warn_slowpath_fmt+0xf8/0x15c
 warn_slowpath_fmt from MEI_IoctlCmdMsgWrite+0x290/0x2c8 [drv_mei_cpe]
 MEI_IoctlCmdMsgWrite [drv_mei_cpe] from MEI_IoctlMsgSend+0x68/0x2b8 [drv_mei_cpe]
 MEI_IoctlMsgSend [drv_mei_cpe] from MEI_InternalMsgSend+0x48/0x50 [drv_mei_cpe]
 MEI_InternalMsgSend [drv_mei_cpe] from MEI_InternalSendMessage+0x58/0x84 [drv_mei_cpe]
 MEI_InternalSendMessage [drv_mei_cpe] from MEI_VRX_DSM_StatsGet+0x60/0x9c [drv_mei_cpe]
 MEI_VRX_DSM_StatsGet [drv_mei_cpe] from MEI_VRX_DSM_FwStatsCheck+0x58/0xc0 [drv_mei_cpe]
 MEI_VRX_DSM_FwStatsCheck [drv_mei_cpe] from MEI_DEV_IoctlFirmwareDownload+0xc84/0x1b64 [drv_mei_cpe]
 MEI_DEV_IoctlFirmwareDownload [drv_mei_cpe] from MEI_InternalFirmwareDownload+0x2c/0x34 [drv_mei_cpe]
 MEI_InternalFirmwareDownload [drv_mei_cpe] from DSL_DRV_DEV_FwDownload+0x23c/0x820 [drv_dsl_cpe_api]
 DSL_DRV_DEV_FwDownload [drv_dsl_cpe_api] from DSL_DRV_AutobootLoadFirmware+0x240/0x624 [drv_dsl_cpe_api]
 DSL_DRV_AutobootLoadFirmware [drv_dsl_cpe_api] from DSL_DRV_IoctlHandleHelperCall+0x8c/0x14c [drv_dsl_cpe_api]
 DSL_DRV_IoctlHandleHelperCall [drv_dsl_cpe_api] from DSL_DRV_IoctlHandle+0x4c8/0xb94 [drv_dsl_cpe_api]
 DSL_DRV_IoctlHandle [drv_dsl_cpe_api] from DSL_DRV_Ioctls+0x7c/0xb8 [drv_dsl_cpe_api]
 DSL_DRV_Ioctls [drv_dsl_cpe_api] from sys_ioctl+0x25c/0xc0c
 sys_ioctl from ret_fast_syscall+0x0/0x54
Exception stack(0xc355dfa8 to 0xc355dff0)
dfa0:                   00000008 b6e549bc 00000008 c0447201 b6e549bc 00000000
dfc0: 00000008 b6e549bc 0003ac3c 00000036 00000003 00000008 00000001 b6f2ccd0
dfe0: 00039e54 b6e54860 00013e3c b6ee566c
---[ end trace 0000000000000000 ]---
```

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_avm_fritzbox-7530=y
CONFIG_ATH10K_LEDS=y
CONFIG_BIND_ENABLE_DOH=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
CONFIG_KERNEL_IO_URING=y
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
# CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED is not set
CONFIG_OPENSSH_LIBFIDO2=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_ath10k-firmware-qca4019=m
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-full-htt=m
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-htt=m
CONFIG_PACKAGE_bind-client=m
CONFIG_PACKAGE_bind-dig=m
CONFIG_PACKAGE_bind-libs=m
CONFIG_PACKAGE_bind-nslookup=m
CONFIG_PACKAGE_bmon=m
CONFIG_PACKAGE_collectd=m
CONFIG_PACKAGE_collectd-mod-conntrack=m
CONFIG_PACKAGE_collectd-mod-cpu=m
CONFIG_PACKAGE_collectd-mod-cpufreq=m
CONFIG_PACKAGE_collectd-mod-df=m
CONFIG_PACKAGE_collectd-mod-disk=m
CONFIG_PACKAGE_collectd-mod-dns=m
CONFIG_PACKAGE_collectd-mod-entropy=m
CONFIG_PACKAGE_collectd-mod-ethstat=m
CONFIG_PACKAGE_collectd-mod-exec=m
CONFIG_PACKAGE_collectd-mod-interface=m
CONFIG_PACKAGE_collectd-mod-ipstatistics=m
CONFIG_PACKAGE_collectd-mod-iwinfo=m
CONFIG_PACKAGE_collectd-mod-load=m
CONFIG_PACKAGE_collectd-mod-memory=m
CONFIG_PACKAGE_collectd-mod-network=m
CONFIG_PACKAGE_collectd-mod-ping=m
CONFIG_PACKAGE_collectd-mod-sqm=m
CONFIG_PACKAGE_collectd-mod-vmem=m
CONFIG_PACKAGE_collectd-mod-wireless=m
CONFIG_PACKAGE_confuse=m
CONFIG_PACKAGE_diffutils=m
CONFIG_PACKAGE_git=m
CONFIG_PACKAGE_git-http=m
CONFIG_PACKAGE_hostapd-openssl=m
CONFIG_PACKAGE_htop=m
CONFIG_PACKAGE_iftop=m
CONFIG_PACKAGE_iperf3=m
CONFIG_PACKAGE_iptables-mod-ipopt=m
CONFIG_PACKAGE_iptables-nft=m
CONFIG_PACKAGE_kmod-ath10k=m
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-ifb=m
CONFIG_PACKAGE_kmod-ipt-core=m
CONFIG_PACKAGE_kmod-ipt-ipopt=m
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=m
CONFIG_PACKAGE_kmod-nf-ipt=m
CONFIG_PACKAGE_kmod-nft-compat=m
CONFIG_PACKAGE_kmod-sched-cake=m
CONFIG_PACKAGE_kmod-sched-core=m
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libatomic=m
CONFIG_PACKAGE_libcbor=m
CONFIG_PACKAGE_libcurl=m
CONFIG_PACKAGE_libevdev=m
CONFIG_PACKAGE_libfido2=m
CONFIG_PACKAGE_libiperf3=m
CONFIG_PACKAGE_libiptext=m
CONFIG_PACKAGE_libiptext-nft=m
CONFIG_PACKAGE_libiptext6=m
CONFIG_PACKAGE_libltdl=m
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=m
CONFIG_PACKAGE_libnl-core=m
CONFIG_PACKAGE_libnl-route=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_libopenssl-conf=m
CONFIG_PACKAGE_libopenssl-legacy=m
CONFIG_PACKAGE_liboping=m
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libpcre2=m
CONFIG_PACKAGE_libstdcpp=m
CONFIG_PACKAGE_libudev-zero=m
CONFIG_PACKAGE_liburcu=m
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-wolfssl=y
CONFIG_PACKAGE_libuv=m
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_libxtables=m
CONFIG_PACKAGE_nmap=m
CONFIG_PACKAGE_openssh-client=m
CONFIG_PACKAGE_openssh-keygen=m
CONFIG_PACKAGE_openssh-server=m
CONFIG_PACKAGE_openssh-sftp-client=m
CONFIG_PACKAGE_openssh-sftp-server=m
CONFIG_PACKAGE_sqm-scripts=m
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_tcpdump=m
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_vim-full=m
CONFIG_PACKAGE_vim-runtime=m
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-basic-wolfssl=y
CONFIG_PACKAGE_xtables-nft=m
CONFIG_PCRE2_JIT_ENABLED=y
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_WOLFSSL_HAS_NO_HW=y
# CONFIG_WPA_MBO_SUPPORT is not set
CONFIG_WPA_WOLFSSL=y
# CONFIG_HTOP_LMSENSORS is not set
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

janh, 2024-12-02:
Relevant discussion in the pull request that added support for kernel 6.1: https://github.com/openwrt/openwrt/pull/13332

To summarize: `memcpy` is used to copy data to multiple struct fields at once (`header.index`, `header.length`, `payload` - where `pDestPtr` is a pointer to the first one). This seems to work because there is no padding between these fields.

cybern0id, 2024-12-17:
This kernel warning also happens on the default openwrt.org build of 24.10.0-rc2 r28161-ea17e958b9 (not self-built or patched) on my Fritzbox 7530.
I have just today upgraded from 23.05.5 stable.
I also have two other kernel warnings relating to `backports-6.11.2/net/mac80211/main.c:272 0xbf2c8888 [mac80211@f459305b+0x77000]` and `hostapd`. None of the warnings seem to affect functionality over the past several hours of use (Wireguard, Wifi, VDSL, PBR).

fightforlife, 2025-02-02:
Can reproduce with RC7.
I am using the 7520 as modem, configured like here- Port 4 to router with VLAN 1 untagged for LAN and VLAN 7 tagged for DSL.
https://forum.openwrt.org/t/how-to-bridge-pppoe-dsl-connection-from-openwrt-to-openwrt/185816/14
```
[   29.452807] ------------[ cut here ]------------
[   29.452853] WARNING: CPU: 3 PID: 2088 at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/dsl_cpe_mei-ugw_8.5.2.10/src/drv_mei_cpe_msg_process.c:3570 0xbf11cc68 [drv_mei_cpe@105346d2+0x19000]
[   29.456526] memcpy: detected field-spanning write (size 4) of single field "pDestPtr" at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/dsl_cpe_mei-ugw_8.5.2.10/src/drv_mei_cpe_msg_process.c:3570 (size 2)
[   29.474705] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat(O) clip drv_dsl_cpe_api(O) ledtrig_usbport drv_mei_cpe(O) ledtrig_transient ledtrig_pattern ledtrig_oneshot ledtrig_activity br2684 vrx518_tc(O) atm vrx518(O) drv_ifxos(O) sha512_arm md5 ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug(O) crc32c_generic
[   29.544556] CPU: 3 PID: 2088 Comm: evnthnd Tainted: G           O       6.6.73 #0
[   29.566749] Hardware name: Generic DT based system
[   29.574133] Function entered at [<c030d1d8>] from [<c0308ec8>]
[   29.578814] Function entered at [<c0308ec8>] from [<c0a5056c>]
[   29.584630] Function entered at [<c0a5056c>] from [<c032165c>]
[   29.590446] Function entered at [<c032165c>] from [<c032177c>]
[   29.596261] Function entered at [<c032177c>] from [<bf11cc68>]
[   29.602084] Function entered at [<bf11cc68>] from [<bf11d040>]
[   29.607893] Function entered at [<bf11d040>] from [<bf118e9c>]
[   29.613710] Function entered at [<bf118e9c>] from [<bf118efc>]
[   29.619524] Function entered at [<bf118efc>] from [<bf12b244>]
[   29.625341] Function entered at [<bf12b244>] from [<bf12b2d8>]
[   29.631157] Function entered at [<bf12b2d8>] from [<bf12a168>]
[   29.636973] Function entered at [<bf12a168>] from [<bf119120>]
[   29.642789] Function entered at [<bf119120>] from [<bf15d4e8>]
[   29.648607] Function entered at [<bf15d4e8>] from [<bf146f40>]
[   29.654421] Function entered at [<bf146f40>] from [<bf1425c0>]
[   29.660237] Function entered at [<bf1425c0>] from [<bf145e10>]
[   29.666052] Function entered at [<bf145e10>] from [<bf149328>]
[   29.670741] qca8k-ipq4019 c000000.switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   29.671868] Function entered at [<bf149328>] from [<c04c6554>]
[   29.671878] Function entered at [<c04c6554>] from [<c0300040>]
[   29.677747] br-lan: port 4(lan4) entered blocking state
[   29.685930] Exception stack(0xc3515fa8 to 0xc3515ff0)
[   29.685944] 5fa0:                   00000008 b6e589bc 00000008 c0447201 b6e589bc 00000000
[   29.685958] 5fc0: 00000008 b6e589bc 0003ac3c 00000036 00000003 00000008 00000001 b6f30fb0
[   29.691850] br-lan: port 4(lan4) entered forwarding state
[   29.697650] 5fe0: 00039e54 b6e58860 00013e3c b6ee966c
[   29.729799] ---[ end trace 0000000000000000 ]---
```

janh, 2025-02-06:
I just sent a patch for this to the mailing list.

closed 2025-02-22 by openwrt-bot