openwrt/openwrt #17150: wifi-scripts with new ucode

rmandrad, 2024-12-03:
### Describe the bug

wifi scripts with new code fails while setting wifi

`Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): wifi-scripts: Starting
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): wifi-scripts: Starting
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): wifi-scripts: Starting
Tue Dec  3 05:22:44 2024 daemon.notice hostapd: Set new config for phy phy0.2:
Tue Dec  3 05:22:44 2024 daemon.notice hostapd: Set new config for phy phy0.1:
Tue Dec  3 05:22:44 2024 daemon.notice wpa_supplicant[2352]: Set new config for phy phy0.2
Tue Dec  3 05:22:44 2024 daemon.notice hostapd: Set new config for phy phy0.0:
Tue Dec  3 05:22:44 2024 daemon.notice wpa_supplicant[2352]: Set new config for phy phy0.1
Tue Dec  3 05:22:44 2024 daemon.notice wpa_supplicant[2352]: Set new config for phy phy0.0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): command failed: Invalid argument (-22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): command failed: Invalid argument (-22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): command failed: Invalid argument (-22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): Reference error: access to undeclared variable digest
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): In iface_roaming(), file /usr/share/ucode/wifi/ap.uc, line 332, byte 48:
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   called from function generate (/usr/share/ucode/wifi/ap.uc:459:22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   called from function setup_interface (/usr/share/ucode/wifi/hostapd.uc:531:58)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:558:106)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   called from function setup (./mac80211.sh:304:21)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   called from anonymous function (./mac80211.sh:339:14)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):  `    set_default(config, 'mobility_domain', substr(digest.md5(config.ssid), 0, 4));`
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):   Near here ----------------------------------------^
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): Reference error: access to undeclared variable digest
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): In iface_roaming(), file /usr/share/ucode/wifi/ap.uc, line 332, byte 48:
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   called from function generate (/usr/share/ucode/wifi/ap.uc:459:22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   called from function setup_interface (/usr/share/ucode/wifi/hostapd.uc:531:58)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:558:106)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   called from function setup (./mac80211.sh:304:21)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   called from anonymous function (./mac80211.sh:339:14)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):  `    set_default(config, 'mobility_domain', substr(digest.md5(config.ssid), 0, 4));`
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):   Near here ----------------------------------------^
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): Reference error: access to undeclared variable digest
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): In iface_roaming(), file /usr/share/ucode/wifi/ap.uc, line 332, byte 48:
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   called from function generate (/usr/share/ucode/wifi/ap.uc:459:22)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   called from function setup_interface (/usr/share/ucode/wifi/hostapd.uc:531:58)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:558:106)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   called from function setup (./mac80211.sh:304:21)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   called from anonymous function (./mac80211.sh:339:14)
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):  `    set_default(config, 'mobility_domain', substr(digest.md5(config.ssid), 0, 4));`
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):   Near here ----------------------------------------^
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739):
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): wifi-scripts: network is not present in the schema
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5740): wifi-scripts: Preparing interface: phy0.2-ap0 with MAC: xxxxxx
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): wifi-scripts: network is not present in the schema
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5738): wifi-scripts: Preparing interface: phy0.0-ap0 with MAC: xxx
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): wifi-scripts: network is not present in the schema
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5739): wifi-scripts: Preparing interface: phy0.1-ap0 with MAC: xxx
Tue Dec  3 05:22:44 2024 daemon.warn netifd: Wireless device 'radio0' setup failed, retry=3
Tue Dec  3 05:22:44 2024 daemon.warn netifd: Wireless device 'radio1' setup failed, retry=3
Tue Dec  3 05:22:44 2024 daemon.warn netifd: Wireless device 'radio2' setup failed, retry=3
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio1 (5757): wifi-scripts: Tearing down phy0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio0 (5756): wifi-scripts: Tearing down phy0
Tue Dec  3 05:22:44 2024 daemon.notice netifd: radio2 (5758): wifi-scripts: Tearing down phy0
Tue Dec  3 05:22:44 2024 daemon.notice hostapd: Set new config for phy phy0.1:
Tue Dec  3 05:22:44 2024 daemon.notice hostapd: Set new config for phy phy0.2:
Tue Dec  3 05:22:44 2024 daemon.notice wpa_supplicant[2352]: Set new config for phy phy0.1
[... 226 lines cut ...]
snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

mediatek/filogic

### Device

bananapi4

### Image kind

Self-built image

### Steps to reproduce

fails at start 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

blogic, 2024-12-03:
fixed in main branch

closed 2024-12-03 by blogic

sppmasterspp, 2025-01-09:
> fixed in main branch

I get this running on qualcommax/ipq807x

```
daemon.notice netifd: radio0 (17956): wifi-scripts: Tearing down phy0
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[2884]: Set new config for phy phy0
daemon.notice netifd: Wireless device 'radio0' is now down
daemon.notice netifd: radio0 (17971): wifi-scripts: Starting
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[2884]: Set new config for phy phy0
daemon.notice netifd: radio0 (17971): command failed: Not supported (-95)
daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
daemon.notice hostapd: Restart interface for phy phy0
daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211h=1 ieee80211d=1 spectrum_mgmt_required=0 beacon_int=100 noscan=1 airtime_mode=0 rssi_reject_assoc_rssi=0 rssi_ignore_probe_request=0 acs_exclude_dfs=0 enable_background_radar=1 min_tx_power=0 hw_mode=a ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=0 vht_capab=[RXLDPC][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-7991] he_bss_color=128 he_spr_sr_control=3 ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=0 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_twt_required=0 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=

kern.info kernel: [  241.633470] ath11k c000000.wifi phy0-ap0: entered allmulticast mode
daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.err hostapd: Interface initialization failed
daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED
daemon.notice hostapd: phy0-ap0: AP-DISABLED
daemon.err hostapd: phy0-ap0: Unable to setup interface.
daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
kern.info kernel: [  241.639271] ath11k c000000.wifi phy0-ap0: entered promiscuous mode
user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy0-ap0)
kern.info kernel: [  241.689270] ath11k c000000.wifi phy0-ap0: left allmulticast mode
kern.info kernel: [  241.689329] ath11k c000000.wifi phy0-ap0: left promiscuous mode
kern.info kernel: [  241.694472] br-lan: port 6(phy0-ap0) entered disabled state
daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started
daemon.notice hostapd: hostapd.add_iface failed for phy phy0 ifname=phy0-ap0
daemon.notice netifd: Wireless device 'radio0' is now up
daemon.notice netifd: radio0 (17971): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
daemon.notice netifd: radio0 (17971): wifi-scripts: network is not present in the schema
daemon.notice netifd: radio0 (17971): wifi-scripts: Preparing interface: phy0-ap0 with MAC: null
```