openwrt/openwrt #17152: OpenWrt extreme-networks ws-ap3825i ft_fixup_l2cache: FDT_ERR_NOTFOUND

Sidj77, 2024-12-03:
### Describe the bug

I uploaded 10 pieces, starting with the oldest:
openwrt-22.03.0-mpc85xx-p1020-extreme-networks_ws-ap3825i-initramfs-kernel
until for:
openwrt-23.05.5-mpc85xx-p1020-extreme-networks_ws-ap3825i-initramfs-kernel 
this one also:
extreme-networks_ws-ap3825i-initramfs-kernel.bin <
all of them die<ft_fixup_l2cache: FDT_ERR_NOTFOUND> (older ones load something and show openwrt, but after restart they die)
Each one gives this error (copy with putty in the appendix).

### OpenWrt version

All version - with newest also

### OpenWrt release

All version - with newest also

### OpenWrt target/subtarget

-

### Device

extreme-networks ws-ap3825i

### Image kind

Official downloaded image

### Steps to reproduce

setenv ramboot_openwrt "setenv ipaddr 192.168.1.22; setenv serverip 192.168.1.1; tftpboot 0x2000000 initramfs.bin; interrupts off; bootm start 0x2000000; bootm loados; fdt resize;fdt resize; fdt boardsetup; fdt resize; fdt boardsetup; fdt chosen; fdt resize; fdt chosen; bootm prep; bootm go;"

setenv boot_openwrt "cp.b 0xEC000000 0x2000000 0x2000000; interrupts off; bootm start 0x2000000; bootm loados; fdt resize; fdt boardsetup; fdt resize; fdt boardsetup; fdt chosen; fdt resize; fdt chosen; bootm prep; bootm go;"

setenv bootcmd "run boot_openwrt";

saveenv

run ramboot_openwrt

setenv ramboot_openwrt "setenv ipaddr 192.168.1.22; setenv serverip 192.168.1.66; tftpboot 0x2000000 initramfs.bin; interrupts off; bootm 0x2000000;"

setenv boot_openwrt "bootm 0xEC000000;"

setenv bootcmd "run boot_openwrt";

saveenv

run ramboot_openwrt

### Actual behaviour

Filename 'initramfs.bin'.
Load address: 0x2000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #######################
done
Bytes transferred = 7015568 (6b0c90 hex)
## Booting kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-5.15.167
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x020000ec
     Data Size:    7001092 Bytes = 6.7 MiB
     Architecture: PowerPC
     OS:           Linux
     Load Address: 0x01000000
     Entry Point:  0x01000000
     Hash algo:    crc32
     Hash value:   bfcef764
     Hash algo:    sha1
     Hash value:   52adf161a596cdaab746169b0e4b6b627615ef1f
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Flattened Device Tree from FIT Image at 02000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x026ad640
     Data Size:    12541 Bytes = 12.2 KiB
     Architecture: PowerPC
     Hash algo:    crc32
     Hash value:   ecfc0d58
     Hash algo:    sha1
     Hash value:   6d81d68bcfbf0113b69bb11ac38d7761ddf8959f
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x26ad640
   Loading Kernel Image ... OK
OK
   Loading Device Tree to 02ff9000, end 02fff0fc ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Sidj77, 2024-12-03:
> Invalid Version reported. `All version - with newest also` Is this from a clean repository?

Yes, info in post

Sidj77, 2024-12-03:
> Invalid Release reported. `All version - with newest also` Is this from a clean repository?

Yes, info in post

Sidj77, 2024-12-03:
> Invalid Target/Subtarget reported. `-` Is this from a supported device?

Yes, info in post: ap3825i

Adorfer, 2025-03-29:
for me it was broken in 24.x and 23, tested with 4 devices, none worked. 
but the version in 22.03.0 works for me
https://archive.openwrt.org/releases/22.03.0/targets/mpc85xx/p1020/openwrt-22.03.0-mpc85xx-p1020-extreme-networks_ws-ap3825i-initramfs-kernel.bin

jensgraef, 2025-04-18:
I've encountered the same issue with openwrt-24.10.0
22.03.0 works.
I've added an entry to the wiki page https://openwrt.org/toh/extreme_networks/ws-ap3825i#troubleshooting_ft_fixup_l2cachefdt_err_notfound to help others encountering this issue.

jbusuttil, 2026-09-03:
I've just been doing a fresh install of OpenWRT on AP3825i. I also found 24.10 triggered the same issue, so went back to 22.03.0, then sysupgrade'd to 24.10, then to 25.12.5 successfully using attended-sysupgrade. (Yes, the OpenWRT wiki still has 24.10 as max supported version, but it really is out of date.)

Note that on first reboot after flashing to 25.12.5, it can take 6-10 minutes to fully complete the upgrade, at which point the console looks like it's hung. Be patient :-)