openwrt/openwrt #17153: [24.10.0-rc1] Cannot install kernel modules

LaySoft, 2024-12-03:
### Describe the bug

I executed "opkg update", after I tried to install some kernel moduiles, but the following errors occured:
`Collected errors:
 * opkg_install_cmd: Cannot install package kmod-nf-nathelper.
 * opkg_install_cmd: Cannot install package kmod-ipt-raw.
 * opkg_install_cmd: Cannot install package kmod-usb-storage.
 * opkg_install_cmd: Cannot install package kmod-usb-storage-uas.
 * opkg_install_cmd: Cannot install package kmod-fs-ext4.
`

### OpenWrt version

24.10.0-rc1

### OpenWrt release

24.10.0-rc1

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX4200

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LaySoft, 2024-12-03:
I installed this firmware:
`https://downloads.openwrt.org/releases/24.10.0-rc1/targets/mediatek/filogic/openwrt-24.10.0-rc1-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin`

namiltd, 2024-12-04:
Fixed: https://github.com/openwrt/openwrt/commit/ea17e958b9933aaf3e4a390073025aa226fe91b7
OpenWrt 23.10.0-rc2 released: 66e76aa94f0508d63a75fac2739a5bc0edb021ec
PR to close.

closed 2024-12-05 by hauke