openwrt/openwrt #17157: OpenWrt 24.10.0-rc1.  Linksys EA7500v1: No Internet after clean install

openwrtuser, 2024-12-04:
### Describe the bug

Router: Linksys EA7500v1

After clean install of OpenWrt 24.10.0-rc1 from previous stable build (23.05.5) **there is no connection to the Internet.**

Reset to defaults either via Luci or via command line does not resolve this problem.

After downgrading back to version 23.05.5, Internet starts working as it should.

### OpenWrt version

OpenWrt 24.10.0-rc1 r28158-d276b4c91a / LuCI openwrt-24.10 branch 24.332.74596~80f18df

### OpenWrt release

OpenWrt 24.10.0-rc1 r28158-d276b4c91a / LuCI openwrt-24.10 branch 24.332.74596~80f18df

### OpenWrt target/subtarget

Target Platform	ipq806x/generic

### Device

Model	Linksys EA7500 V1 WiFi Router

### Image kind

Official downloaded image

### Steps to reproduce

Regular upgrade via Luci (with Force Upgrade option enabled),

or, via terminal:

ssh root@192.168.1.1
cd /tmp
wget https://downloads.openwrt.org/releases/24.10.0-rc1/targets/ipq806x/generic/openwrt-24.10.0-rc1-ipq806x-generic-linksys_ea7500-v1-squashfs-factory.bin
sysupgrade -F /tmp/openwrt-24.10.0-rc1-ipq806x-generic-linksys_ea7500-v1-squashfs-factory.bin

### Actual behaviour

Internet is not working.

### Expected behaviour

Internet should be working.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

openwrtuser, 2024-12-04:
Technical info from

http://192.168.1.1/cgi-bin/luci/

Status

System

Hostname	OpenWrt
Model	Linksys EA7500 V1 WiFi Router
Architecture	ARMv7 Processor rev 0 (v7l)
Target Platform	ipq806x/generic
Firmware Version	OpenWrt 24.10.0-rc1 r28158-d276b4c91a / LuCI openwrt-24.10 branch 24.332.74596~80f18df
Kernel Version	6.6.63
Local Time	2024-11-29 17:27:19
Uptime	0h 1m 55s
Load Average	0.12, 0.08, 0.03

openwrtuser, 2024-12-04:
Repository is clean.

The device is supported.

openwrtuser, 2024-12-04:
By the way, the same problem exists also with current SNAPSHOT version for this device - there is no Internet after clean install.

vincejv, 2024-12-04:
do you get your WAN IP via DHCP (IPOE) or PPPoE?

openwrtuser, 2024-12-04:
> do you get your WAN IP via DHCP (IPOE) or PPPoE?

Via DHCP.

aimacintyre, 2024-12-05:
See #16604.

openwrtuser, 2024-12-10:
Today I had a look at this file (etc/config/network) on my Netgear R6250 router (with the same OpenWRT version installed),
and now I think, there is **more correct way to fix this problem:**

1. Open file /etc/config/network and copy line 

	option macaddr 'xx:xx:xx:xx:xx:xx'

Insert it after the following two lines:

config device
	option name 'wan'

2. Now, delete the following section:

config device
	option name 'lan1'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan2'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan3'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan4'
	option macaddr 'xx:xx:xx:xx:xx:xx'

3. Save the file, and restart network (or reboot):

service network restart

Your Internet should now be working.

openwrtuser, 2024-12-10:
Since I also have Linksys EA8500, I tried this fix on it as well, and it works.

egc112, 2025-01-04:
Also ran into the same problem where all lan ports and wan port had the same MAC address after upgrading my EA8500 from 23.05.5 to 24.10-rc4, see https://forum.openwrt.org/t/build-for-netgear-r7800/316/3904?u=egc
Hope @Ansuel can have a look

mircea-vutcovici, 2025-01-26:
For fixing wan port on Linksys EA8500, after resetting config, I found a more simple approach.
First check that the mac is present for `wan` interface
```
root@OpenWrt:~# uci show network.@device[5]
network.cfg090f15=device
network.cfg090f15.name='wan'
network.@device[5].macaddr='aa:bb:cc:dd:ee:ff'
```
Then remove the hardcoded MAC address:
```
root@OpenWrt:~# uci del network.@device[5].macaddr
root@OpenWrt:~# uci changes
-network.cfg090f15.macaddr
root@OpenWrt:~# uci commit
```

hauke, 2025-02-02:
@robimarko  or @Ansuel  Could you please have a look at this problem. 
I also saw similar reports in the forum.

Ansuel, 2025-02-02:
mh strange... i guess the problem is present from the DSA switch... that complicates debugging it. But I should have a device like this... let me do 1-2 test

egc112, 2025-02-02:
Thanks fort looking into this. 
There is also this: https://forum.openwrt.org/t/openwrt-24-10-0-rc7-seventh-release-candidate/223006/57?u=egc  seems also DSA and switch related?

closed 2025-02-03 by openwrt-bot

nrobert13, 2026-03-08:
is this fixed on 25.12.0? I just flashed over a 24.10.5 with config save, and it still doesn't work. the wan interface is showing the same mac as the lan interfaces. I tried changing it to a unique mac in luci under devices, but the changed mac won't show up on the interface. also tried removing the mac from the device, still not working. tried also network restart. Could it be because I flashed with config save?

```
uci show network
network.@device[5]=device
network.@device[5].name='wan'
network.@device[5].ipv6='0'
```

```
 ip -d link show wan
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether c0:56:27:0d:7a:76 brd ff:ff:ff:ff:ff:ff promiscuity 0 allmulti 0 minmtu 68 maxmtu 9000
    dsa conduit eth0 addrgenmode eui64 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 gso_ipv4_max_size 65536 gro_ipv4_max_size 65536 portname p5 switchid 00000000 parentbus mdio_bus parentdev 37000000.mdio-mii:10
```

```
ifconfig wan
wan       Link encap:Ethernet  HWaddr C0:56:27:0D:7A:76
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1 errors:0 dropped:0 overruns:0 frame:0
          TX packets:665 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:60 (60.0 B)  TX bytes:230755 (225.3 KiB)

```

commit eba2fbf6 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839
commit cd747cb3 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839

(cherry picked from commit eba2fbf63814bbcf8c99c98177818476fca808a7)