openwrt/openwrt #17160: Wrong GPIO PIN on GL.iNet GL-A1300

issuex, 2024-12-04:
### Describe the bug

reset button not work on OpenWRT 23.05.5 . 
from my diag it work on pin 475 by run
```
echo 412 > /sys/class/gpio/export
echo "in" > /sys/class/gpio/gpio412/direction
echo 475 > /sys/class/gpio/export
echo "in" > /sys/class/gpio/gpio475/direction
```
475 seem work for reset switch and
412 seem work for toggle switch, 1=on 0=off

i wish someone help me to fix it in builder, and add support for toggle switch.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-A1300

### Image kind

Official downloaded image

### Steps to reproduce

just flash official image and reset button not work.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-04:
Both version 23.05 and 24.10 reset button settings are the same. Attach syslog from version 23.05.

issuex, 2024-12-04:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.167 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.3.0 r24106-10cc5fcd00) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Sep 23 12:34:46 2024
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: GL.iNet GL-A1300
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s19660 r8192 d21300 u49152
[    0.000000] pcpu-alloc: s19660 r8192 d21300 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=ubi root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 247204K/262144K available (6768K kernel code, 611K rwdata, 1020K rodata, 1024K init, 246K bss, 14940K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000001] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000024] Switching to timer-based delay loop, resolution 20ns
[    0.000319] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000349] CPU: Testing write buffer coherency: ok
[    0.000406] pid_max: default: 32768 minimum: 301
[    0.001248] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001280] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.003273] qcom_scm: convention: smc legacy
[    0.004502] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.004680] rcu: Hierarchical SRCU implementation.
[    0.005279] smp: Bringing up secondary CPUs ...
[    0.008792] smp: Brought up 1 node, 4 CPUs
[    0.008829] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.008845] CPU: All CPU(s) started in SVC mode.
[    0.015533] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.015702] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.015739] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.016123] pinctrl core: initialized pinctrl subsystem
[    0.018177] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.018608] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020112] thermal_sys: Registered thermal governor 'step_wise'
[    0.020545] cpuidle: using governor ladder
[    0.020606] cpuidle: using governor menu
[    0.049458] cryptd: max_cpu_qlen set to 1000
[    0.054298] usbcore: registered new interface driver usbfs
[    0.054411] usbcore: registered new interface driver hub
[    0.054478] usbcore: registered new device driver usb
[    0.054565] pps_core: LinuxPPS API ver. 1 registered
[    0.054580] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.054609] PTP clock support registered
[    0.056878] clocksource: Switched to clocksource arch_sys_counter
[    0.058289] NET: Registered PF_INET protocol family
[    0.058535] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.059524] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.059577] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.059602] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.059645] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[... 170 lines cut ...]
[   19.917651] ath: EEPROM indicates default country code should be used
[   19.917657] ath: doing EEPROM country->regdmn map search
[   19.917665] ath: country maps to regdmn code: 0x3a
[   19.917674] ath: Country alpha2 being used: US
[   19.917681] ath: Regpair used: 0x3a
[   19.934711] kmodloader: done loading kernel modules from /etc/modules.d/*
[   48.026561] ipqess-edma c080000.ethernet eth0: Link is Down
[... cut ...]
[   48.096937] br-lan: port 2(lan2) entered blocking state
[   48.101138] br-lan: port 2(lan2) entered disabled state
[   48.106836] device lan2 entered promiscuous mode
[   48.123433] qca8k-ipq4019 c000000.switch wan: configuring for phy/psgmii link mode
[   51.198830] qca8k-ipq4019 c000000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   51.198878] br-lan: port 1(lan1) entered blocking state
[   51.198900] br-lan: port 1(lan1) entered forwarding state
[   51.212138] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

issuex, 2024-12-04:
not sure why PIN on 23.* look far different. but 24-rc2 correct.

namiltd, 2024-12-04:
cat /sys/kernel/debug/gpio

namiltd, 2024-12-11:
PR https://github.com/openwrt/openwrt/pull/17228 accepted.
Test images from https://mirror-03.infra.openwrt.org/releases/23.05-SNAPSHOT/ which should contain this PR tomorrow.

pielgrzym, 2024-12-18:
All buttons work fine when using snapshot :) Thanks!

closed 2024-12-27 by issuex