openwrt/openwrt #17165: imagebuilder: "Cannot install package kmod-mt7921u"

nerdhardest, 2024-12-04:
### Describe the bug

Output from imagebuilder 24.10.0-rc1.

### OpenWrt version

24.10.0-rc1

### OpenWrt release

24.10.0-rc1

### OpenWrt target/subtarget

glinet_gl-ar750s-nor-nand

### Device

glinet_gl-ar750s-nor-nand

### Image kind

Official downloaded image

### Steps to reproduce

Try to build image for "24.10.0-rc1-ath79-nand" and include package "kmod-mt7921u".

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-04:
Ignore the bot, it has certain versioning and model expectations.

But can you please include the log that captures the error

nerdhardest, 2024-12-04:
Sounds like KA2107 is correct; this is likely a duplicate of 17146.

I'll wait for -rc2 to try again.

Thx.

hauke, 2024-12-05:
fixed in ea17e958b9933aaf3e4a390073025aa226fe91b7

closed 2024-12-05 by hauke