openwrt/openwrt #17166: Nexx WT3020F poor wifi performance

ViBE-HU, 2024-12-04:
### Describe the bug

I have two units of this router and tested both before my report. You can read more details [here](https://forum.openwrt.org/t/nexx-wt3020f-poor-wifi-performance/214470). Both behave the same. I tested with a few client machines and several OS'es. The client machines have Intel AX200 and AX211 cards. So my guess the experienced issues is are not hardware/client related.

With the stock firmware the wifi bandwidth is fine but lag spikes noticeable. I archived the firmwares [here](https://1drv.ms/f/c/5e4496dea959027f/En8CWanelkQggF7LAAAAAAABY-DtemxVZLN-KCucIKt67w?e=IK7GXy).

So my reference point is the stock performance. The stock firmware can reach ~94 Mbps up/down speed via wifi and as I mentioned above there is a noticeable stuttering. It only supports WPA and WPA2 encryptions. 

A few weeks ago I started to use this router again and tried some actual OpenWRT builds. I noticed that even if the wired performance still looks okay, but the wireless is not so good. With WPA2 encryption I reached about ~70 Mbps bandwidth and moderate stuttering. And with WPA3 encryption the bandwidth evaporates. The highest I reached is ~16 Mbps and the stuttering noticeable. Under heavy load it becomes useless.

### OpenWrt version

r27964-a0eafc3c77

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7620

### Device

Nexx WT3020 (8M)

### Image kind

Official downloaded image

### Steps to reproduce

1. Setup the router (as a router)
2. Setup wireless connection with WPA2 and/or WPA3 encryption
3. Connect to the router via wifi with a PC
4. Use that PC for online services/

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-12-04:
Since there are problems with the performance of 2.4GHz WiFi for Mediatek MT7620 with enabled 802.11w Management Frame Protection, use WPA2-PSK security.

brada4, 2024-12-05:
Do you think you will get better answers asking same q&estion repeatedly?

ViBE-HU, 2024-12-07:
> Do you think you will get better answers asking same q&estion repeatedly?

You advised that I should report the issue. So now it is not clear what was the point if everything is "fine" but still worth to report it.

brada4, 2024-12-07:
You kind of opened this 2 days after hardware limitation was brought to you in forum.