openwrt/openwrt #17167: [24.10.0-rc2] batman problems with VLAN on R7800

rmccask, 2024-12-04:
### Describe the bug

The R7800 moved from swconfig to dsa and needed to be reconfigured.  I setup batman and it worked properly connecting to my network on vlan 4 over WiFi.  After a reboot it lost access to batman and I have lots of log entries with:
Wed Dec  4 12:33:43 2024 kern.info kernel: [  684.643416] batman_adv: bat0: adding TT local entry xx:xx:xx:xx:xx:xx to non-existent VLAN 4

Restarting the lan interface, which is attached to device br-lan.4 fixes the problem until I reboot again.  Device br-lan has bat0 in bridge ports with bat0 getting VLAN 4 tagged on the VLAN filtering tab.

My image was built with:  make image PROFILE="netgear_r7800" PACKAGES="-libustream-mbedtls libustream-openssl -wpad-basic-mbedtls wpad-mesh-openssl luci-ssl-openssl luci-proto-batman-adv"

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

helljol, 2024-12-07:
I have the same issue i'm not able to get 802.11s mesh vlan working with batman_adv after upgrading my xiaomi ax3200s to 24.10.0-rc2.

I have not found any workaround to get it working for now

c-gabri, 2024-12-15:
Getting the same error here on a Zyxel NR7101 after upgrading from 23.05.5 to 24.10.0-rc2:

`bat0: adding TT local entry xx:xx:xx:xx:xx:xx to non-existent VLAN x`

Batman DHCP clients don't get an IP from the Batman DHCP server (the device that was upgraded). Had a perfectly working Batman setup with VLANs before. Device was already on DSA.
I have to upgrade to 24.10.0-rc2 because MbedTLS on 23.05.5 is too old and doesn't work with my OpenWisp server.

rmccask, 2024-12-24:
I have confirmed that the problem still exists on rc4.

mantheman, 2024-12-24:
I have never had any problems using batman-adv with my R7800s. It still works fine in 24.10.0-rc4. 
Maybe you can make a post in the forums with more detail. Also have you read the batmand-adv FAQ?

B.A.T.M.A.N. Advanced - VLAN questions[](https://www.open-mesh.org/doc/batman-adv/Faq.html#b-a-t-m-a-n-advanced-vlan-questions)
VLAN forwarding doesn’t work[](https://www.open-mesh.org/doc/batman-adv/Faq.html#vlan-forwarding-doesn-t-work)
Q: When I have a bridge which connects an ethernet devices with bat0, 802.1Q VLAN frames from the ethernet interfaces are not forwarded by batman-adv. Instead a warning like batman_adv: bat0: adding TT local entry xx:xx:xx:xx:xx:xx to non-existent VLAN 23 is printed periodically. How is it possible to forward these frames

A: batman-adv since 2014.0.0 is 802.1Q VLAN-aware. It is only able to forward VLAN frames when it knows about the VLAN. This can either be done by creating a 802.1Q VLAN device with the correct VID on top of the batadv (bat0) device:

ip link add link bat0 name bat0.23 type vlan id 23
Or in case of a VLAN-aware bridge, it is better to add the VLANs as required to the specific ports:

bridge vlan add vid 23 dev bat0
Note: Do not rely on VLAN packets being filtered when no VLAN is added on top of bat0. This is likely subject to change in the future.

rmccask, 2024-12-25:
After testing with rc4, I moved my R7800 back to 23.05.5 and it is using batman with 2 Belkin RT3200.  They have been using batman since 23.05.2.  I used non-batman mesh prior to that.  The RT3200 are already dsa and the R7800 config for 24.10.0 should be very similar to how they are currently setup.  The R7800 does work with batman on 24.10.0-rc4 so I think the configuration is correct but the link doesn't come up correctly after a reboot.  I can see that the bat0 device is receiving but not sending, TX stays zero.  Restarting the lan interface brings it back to working correctly until I reboot again.  I will try again but all the vlan and batman settings looked correct and it did function until I rebooted.

Yes, I did look at the FAQ and did a lot of research before I posted about this issue.  I don't think this is user error.  Two other people have added comments about problems with batman after upgrading to 24.10.0 so I don't think I am isolated in this but the fact you have it running is good news.

c-gabri, 2025-01-05:
The problem for me is solved by creating a different bridge for each VLAN and adding `bat0.<vid>` to each, instead of creating a single bridge and adding `bat0` marked tagged on all VLANs (VLAN filtering). Of course the other bridge ports must be set accordingly (`<port>.<vid>` for tagged, `<port>` for untagged).

Separate bridges is indeed the way it's done in most guides I've seen, but for some reason VLAN filtering was working on 23.05 and not on 24.10.

rmccask, 2025-01-06:
I tried creating a bridge for each vlan and that seems to be working.  I want my wlan to be all vlans, tagged, but that doesn't seem to be working so far.  I am hoping this will be fixed before 24.10.0 final but for now I have it running on this device.  Thanks, @c-gabri !

c-gabri, 2025-01-06:
@rmccask Not sure I understand: you still get VLANs with the separate bridges approach (tagged or untagged depending on how you add ports to the bridges), it's just a different way of defining them

rmccask, 2025-01-06:
When I tried to configure the wan interface to one of the new bridge devices earlier, it would time out and then I would roll it back.  I tried it 3 times with minor variations.  I was connected via one of the lan interfaces at the time.  I tried a few minutes ago and it worked but this time I was connecting via the batman interface over the wlan.  I also rebooted a couple of times and moved it back to the normal location.  Now it is all working though I haven't tested the wan interface to verify if all the vlans are present but they are configured to be.

JFroch, 2026-09-13:
@simonwunderlich do you know about this behaviour?

To sum it up: If a batman adv interface is (re)started as part of a VLAN aware bridge that has at least one DSA port member, the VLAN IDs that are tagged on the batman adv interface are not sent to/received by it and `adding TT local entry xx:xx:xx:xx:xx:xx to non-existent VLAN xx` messages show up in syslog, the translocal table also stays empty. I have added a `printk` to `batadv_meshif_create_vlan` in mesh-interface.c and while a start on a bridge without a DSA port printed the tagged VLAN IDs to syslog, on a bridge with DSA port it only printed `-1`.

Manually adding bat0.xx interfaces to the bridge and setting them untagged to the corresponding VLAN does work.

This was tested on v2026.2 and .3 (and openwrt snapshot)

simonwunderlich, 2026-09-17:
@JFroch this is actually a long standing limitation/design decision in batman-adv to only include VLANs which have been explicitly registered on the host system, e.g. by adding a bat0.xx interface. 

The original question sounds like bat0 gets a frame with a VLAN header on vlan 4, which it then reacts. That shouldn't happen in the first place if the VLAN was not intended to be used inside the network (and not just inside the switch to specify ports etc).

I've not gone through the whole ticket, but perhaps this helps explaining.

rmccask, 2026-09-18:
VLAN 4 is defined and the exact same setup works on other devices (RT3200 and OpenWrt One) so I don't think this is "by design".  I took my r7800 and changed the config around and it was working, then I rebooted and it broke again so I think the bug is still there with the latest 25.12 snapshots.  I didn't think to look for the error message or try to debug it this time and just restored the previous configuration.  I can reproduce it and get some fresh logs if anyone is interested.

JFroch, 2026-09-18:
@simonwunderlich thank you, does setting the VLAN as tagged on the bat0 interface like below also count as explicitly registering it, or is creating bat0.xx interfaces the only way it "should" work? 
Because this does work, but as soon as a DSA port is added (and the batman interfaces is restarted by whatever means), it stops working/the VLANs aren´t registered on the batman interface anymore.

```
config device 'device_lan'
        option name 'br-lan'
        option type 'bridge'
        list ports 'bat0'
        list ports 'internet'

config bridge-vlan 'bridge_vlan_1015'
        option device 'br-lan'
        option vlan '1015'
        list ports 'bat0:t'
        list ports 'internet:t'

config bridge-vlan 'bridge_vlan_1010'
        option device 'br-lan'
        option vlan '1010'
        list ports 'bat0:t'
        list ports 'internet:u*'
```

As many openwrt devices have built-in DSA switches for the LAN side (or even all ports) it would be nice to be able to bridge the batman interface to DSA ports without having to manually create a VLAN subinterface for each one.

In my case and as far as I understand it the other people in this thread, I want to transport the VLANs over the mesh and use the DSA ports as uplink/downlink ports, as they contain the networks for e.g. guests and iot devices.