openwrt/openwrt #17175: TP-Link C6v2 5GHz WiFi

zekica, 2024-12-05:
### Describe the bug

On TP-Link C6v2 5GHz wifi doesn't work unless /lib/firmware/ath10k/QCA9888/hw2.0/board.bin and board-2.bin are replaced with the versions from 23.05.5 and file /lib/firmware/ath10k/pre-cal-pci-0000:00:00.0.bin is copied from 23.05.5

When the files are copied, ath10k correctly initializes the 5GHz QCA9888 chip.

### OpenWrt version

24.10.0-rc2

### OpenWrt release

r28161-ea17e958b9

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C6v2 EU

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kofec, 2024-12-05:
@zekica please read: https://github.com/openwrt/openwrt/issues/14541

kofec, 2024-12-05:
Please read this comment: https://github.com/openwrt/openwrt/pull/14611#issuecomment-2000831458

castillofrancodamian, 2024-12-05:
@kofec I could never get those "instructions" to work.

castillofrancodamian, 2024-12-05:
> Please read this comment: [#14611 (comment)](https://github.com/openwrt/openwrt/pull/14611#issuecomment-2000831458)

I could never get those "instructions" to work.

zekica, 2024-12-05:
I have read everything from that discussion, but there is one thing that's not clear to me:

[@namiltd's comment ](https://github.com/openwrt/openwrt/pull/14611#issuecomment-2105932930) says that to generate board.bin, I should dump the "art" partition, offset 0x5000, count 0x2F20 (12064) and save that to board.bin and submit it to the ath10k-devel list.

But the code that actually extracts the "pre-cal" (possibly wrong name) actually uses `nvmem-cell-names = "pre-calibration"` from `nvmem-cells = <&precal_art_5000>,` and precal_art_5000 is a nvme cell `reg = <0x5000 0x2f20>` from the art partition.

If I understand correctly, ath10k loads "board.bin" if the exact variant is not in "board-2.bin", and the patch in pr #14611 also adds the option for ath10k to load the same nvme-cell directly - what commit 348e0ee157637a7b6da61d56638f9e843645d673 defines.

I can send the board.bin to upstream - fine. But I'm trying to understand what [this comment](https://github.com/openwrt/openwrt/pull/14611#issuecomment-1938473929) from @robimarko means.

nrpetonr, 2024-12-10:
TP-Link C6v2 5GHz not working for me either on 24.10.0-rc2

zekica, 2024-12-10:
The easiest way to **make it work for now** in the same way it was working before is to replace the script `/etc/hotplug.d/firmware/11-ath10k-caldata` with the following contents (or copy [the entire file](https://github.com/openwrt/openwrt/blob/openwrt-23.05/target/linux/ath79/generic/base-files/etc/hotplug.d/firmware/11-ath10k-caldata) from the 23.05 branch):

```
#!/bin/sh

[ -e /lib/firmware/$FIRMWARE ] && exit 0

. /lib/functions/caldata.sh
. /lib/functions/k2t.sh

board=$(board_name)

case "$FIRMWARE" in
"ath10k/pre-cal-pci-0000:00:00.0.bin")
	case $board in
	tplink,archer-c58-v1|\
	tplink,archer-c59-v1|\
	tplink,archer-c59-v2|\
	tplink,archer-c60-v1|\
	tplink,archer-c60-v2|\
	tplink,archer-c60-v3|\
	tplink,archer-c6-v2|\
	tplink,archer-c6-v2-us|\
	tplink,tl-wa1201-v2)
		caldata_extract "art" 0x5000 0x2f20
		ath10k_patch_mac $(macaddr_add $(mtd_get_mac_binary info 0x8) -1)
		ln -sf /lib/firmware/ath10k/pre-cal-pci-0000\:00\:00.0.bin \
			/lib/firmware/ath10k/QCA9888/hw2.0/board.bin
		;;
	esac
	;;
*)
	exit 1
	;;
esac
```

and reboot

yureiita, 2024-12-14:
I extracted boardfile from stock firmware (Archer C6(EU)_V2_1.3.7 Build 20220613) and generated board-2.bin using ath10k-bdencoder from https://github.com/qca/qca-swiss-army-knife

Attached is the board-2.bin, just copy it into `/lib/firmware/ath10k/QCA9888/hw2.0/` and the 5GHz wifi should be working again. Tested this on both 23.05.5 and 24.10.0-rc2.

[board-2.bin.zip](https://github.com/user-attachments/files/18134102/board-2.bin.zip)

```
% ath10k-bdencoder -i board-2.bin 

FileSize: 12148
FileCRC32: 3ce952ac
FileMD5: fd70953602e08a9069f60255a18cda37
BoardNames[0]: 'bus=pci,bmi-chip-id=0,bmi-board-id=20'
BoardLength[0]: 12064
BoardCRC32[0]: 38a08251
BoardMD5[0]: 6fddc1f03e8968fcfaf82215064bfed3
```

kofec, 2024-12-14:
Hello, I tried your version and:
from:
```
[   22.193586] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   22.247488] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   22.254303] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   25.759694] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   25.769282] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   25.783824] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 59e741e7
[   26.159934] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
[   26.687977] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   26.697764] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
[   26.716437] ath10k_pci 0000:00:00.0: could not probe fw (-12)
```
cp board-2.bin /lib/firmware/ath10k/QCA9888/hw2.0/board-2.bin
to:
[   56.560823] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   56.569076] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[   56.599387] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   56.610808] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
[   56.924756] ath10k_pci 0000:00:00.0: rts threshold -1
[   56.930424] ath10k_pci 0000:00:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   56.941905] ------------[ cut here ]------------
[   56.946690] WARNING: CPU: 0 PID: 1371 at backports-6.11.2/net/mac80211/main.c:272 ieee80211_do_open+0x600/0x804 [mac80211]
[   56.958407] Modules linked in: ath9k(O) ath9k_common(O) nft_fib_inet nf_flow_table_inet ath9k_hw(O) ath10k_pci(O) ath10k_core(O) ath(O) nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) rng_core nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat(O) nfsv3 nfs oid_registry lockd sunrpc grace dns_resolver sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac gpio_button_hotplug(O) crc32c_generic
```
[   57.021194] CPU: 0 PID: 1371 Comm: hostapd Tainted: G           O       6.6.63 #0
[   57.028928] Stack : 00000000 831d8e90 00000000 800c4c4c 00000000 00000004 00000000 00000000
[   57.037582]         82d1bb74 809d0000 808a0000 8074d268 81a40888 00000001 82d1bb18 d21b7c64
[   57.046238]         00000000 00000000 8074d268 82d1ba58 ffffefff 00000000 00000000 ffffffea
[   57.054893]         000000c4 82d1ba64 000000c4 8081b768 00000001 8074d268 00000000 82820968
[   57.063547]         00000009 831d8e90 00000000 83199b00 00000018 803c1ae8 00000000 809d0000
[   57.072203]         ...
[   57.074731] Call Trace:
[   57.077257] [<800667d4>] show_stack+0x28/0xf0
[   57.081785] [<8066c3b8>] dump_stack_lvl+0x38/0x60
[   57.086666] [<80084f90>] __warn+0xb0/0xe4
[   57.090815] [<80085020>] warn_slowpath_fmt+0x5c/0xac
[   57.095969] [<82820968>] ieee80211_do_open+0x600/0x804 [mac80211]
[   57.102592] [<82820b2c>] ieee80211_do_open+0x7c4/0x804 [mac80211]
[   57.109088] 
[   57.110700] ---[ end trace 0000000000000000 ]---
```

yureiita, 2024-12-16:
@kofec the logs indicate that board-2.bin is loaded after copying it.

before:
```
[   21.859867] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   21.912731] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   21.919590] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   24.539998] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   24.549580] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   24.564121] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 9a038867
[   24.952077] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
[   25.476180] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   25.485968] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
[   25.506764] ath10k_pci 0000:00:00.0: could not probe fw (-12)
```

after:
```
[   19.772744] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   19.827744] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   19.834539] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   22.490041] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   22.499624] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   22.514156] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 9a038867
[   22.889249] ath10k_pci 0000:00:00.0: board_file api 2 bmi_id 0:20 crc32 3ce952ac
[   24.676266] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   24.684079] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[   24.714535] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   24.725853] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12900 sram: 29508'
[   24.962189] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-nvmem max-sta 32 raw 0 hwcrypto 1
[   25.070213] ath: EEPROM regdomain: 0x0
[   25.070237] ath: EEPROM indicates default country code should be used
[   25.070243] ath: doing EEPROM country->regdmn map search
[   25.070257] ath: country maps to regdmn code: 0x3a
[   25.070264] ath: Country alpha2 being used: US
[   25.070270] ath: Regpair used: 0x3a
```

As for the warning, it's a different issue that may be related to https://github.com/openwrt/openwrt/issues/15959. According to https://github.com/openwrt/openwrt/issues/15959#issuecomment-2259603978, it's a harmless warning.

ashr123, 2025-02-08:
> I extracted boardfile from stock firmware (Archer C6(EU)_V2_1.3.7 Build 20220613) and generated board-2.bin using ath10k-bdencoder from https://github.com/qca/qca-swiss-army-knife
>
> Attached is the [...]

@yureiita So is this the solution?
should I do (lets say)

```bash
scp ~/Downloads/board-2.bin.zip root@openwrt.local:/lib/firmware/ath10k/QCA9888/hw2.0/
```

Reboot, and it will just work?

yureiita, 2025-02-09:
> [@yureiita](https://github.com/yureiita) So is this the solution? should I do (lets say)
> 
> ```
>scp ~/Downloads/board-2.bin.zip root@openwrt.local:/lib/firmware/ath10k/QCA9888/hw2.0/
>```
> Reboot, and it will just work?

@ashr123 yes but you need to unzip the `board-2.bin.zip` file first and then `scp` the `board-2.bin` file instead.

ashr123, 2025-02-09:
> @ashr123 yes but you need to unzip the `board-2.bin.zip` file first and then `scp` the `board-2.bin` file instead.

@yureiita See comment: https://github.com/openwrt/openwrt/issues/14541#issuecomment-2645915800

I think it will not help in my case for some reason

yureiita, 2025-02-09:
@ashr123 The above board file (https://github.com/openwrt/openwrt/issues/17175#issuecomment-2542733743) is repacked where the 'variant' part of the boardname is stripped using `ath10k-bdencoder` as mentioned in step 3 in https://github.com/openwrt/openwrt/issues/14541#issuecomment-2645640755. I've tested it on my Archer C6 V2 EU on v24.10.0 and can confirm that it works.

ashr123, 2025-02-09:
> [@ashr123](https://github.com/ashr123) The above board file ([#17175 (comment)](https://github.com/openwrt/openwrt/issues/17175#issuecomment-2542733743)) is repacked where the 'variant' part of the boardname is stripped using `ath10k-bdencoder` as mentioned in step 3 in [#14541 (comment)](https://github.com/openwrt/openwrt/issues/14541#issuecomment-2645640755). I've tested it on my Archer C6 V2 EU on v24.10.0 and can confirm that it works.
> 
> In your case, you were trying to use `board-2.bin` from v23.05.5 which will not work.

It worked!!!
Can you describe what did you do for it to work?

yureiita, 2025-02-11:
> [@yureiita](https://github.com/yureiita) Can you describe what did you do for it to work?

1. Download the latest stock firmware from TP-Link (Archer C6(EU)_V2_1.3.7 Build 20220613)
2. Extract the firmware using `binwalk`
3. Find the board file in the extracted firmware (`/lib/firmware/QCA9888/hw.2/boarddata_0.bin`)
4. In the OpenWrt's logs, find the board name that it's looking for
```
[   15.497925] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
```
5. Rename board file to the expected name (`bus=pci,bmi-chip-id=0,bmi-board-id=20.bin`)
6. On the same directory as the board file, create a `board-2.json` with containing the name of the board file
```json
[
    {
        "names": [
            "bus=pci,bmi-chip-id=0,bmi-board-id=20"
        ],
        "data": "bus=pci,bmi-chip-id=0,bmi-board-id=20.bin"
    }
]
```
7. Use `ath10k-bdencoder` to create `board-2.bin` file

xdevs23, 2025-02-16:
https://github.com/openwrt/openwrt/issues/17175#issuecomment-2542733743
> 
> Attached is the board-2.bin, just copy it into `/lib/firmware/ath10k/QCA9888/hw2.0/` and the 5GHz wifi should be working again. Tested this on both 23.05.5 and 24.10.0-rc2.
> 
> [board-2.bin.zip](https://github.com/user-attachments/files/18134102/board-2.bin.zip)

I can confirm this works for `OpenWrt 24.10.0-rc7 r28417-daef29c75d` on `TP-Link Archer C6 v2 (EU/RU/JP)`.
Hope it gets fixed in the final release, I really don't want to be copying files around in the future.

ashr123, 2025-02-16:
@xdevs23 but there is a final release: https://firmware-selector.openwrt.org/?version=24.10.0&target=ath79%2Fgeneric&id=tplink_archer-c6-v2

xdevs23, 2025-02-16:
> [@xdevs23](https://github.com/xdevs23) but there is a final release: https://firmware-selector.openwrt.org/?version=24.10.0&target=ath79%2Fgeneric&id=tplink_archer-c6-v2

Oh, my bad, that was an oversight on my part. I saw 24.10.0-rc7 and didn't see 24.10.0. So it's not fixed even though the stable version was released?

ashr123, 2025-02-16:
If I understand you correctly, see https://github.com/openwrt/openwrt/issues/17175#issuecomment-2646496499 & some previous comments

xdevs23, 2025-02-16:
> [@xdevs23](https://github.com/xdevs23) If I understand you correctly, see [#17175 (comment)](https://github.com/openwrt/openwrt/issues/17175#issuecomment-2646496499) & some previous comments

Well as I said I can confirm it works by replacing the board-2.bin file but I would prefer if it would just work out of the box.

SuperKoe, 2025-03-02:
> The easiest way to **make it work for now** in the same way it was working before is to replace the script `/etc/hotplug.d/firmware/11-ath10k-caldata` with the following contents (or copy [the [...]

Thanks, worked for my TP-Link Archer C6 v2 (EU/RU/JP) so simple!
[https://i.postimg.cc/X7g604d8/afbeelding.png](url)

tobrunn, 2025-04-02:
Just one question: Does 5GHz WiFi  now work with latest OpenWRT release 24.10.0 or still not?

I had a hard time to update / downgrade my Archer C6 routers after I realized that 5G not working and do not want to have the same fun again. I'am not very keen on fiddle around with release candidates or exctract and copy bin files around.

 It would also be really nice if this issue would be mentioned in the according device page as long as it is applicable in order to warn simple users not to update their router until it is finally fixed!
Thank you!

mcurly, 2025-04-04:
Hello, 

[https://github.com/openwrt/openwrt/issues/17175#issuecomment-2531723181](url)

Works for me on a different equipment. 

My router is a Xiaomi AC 2350.

It presented the same symptoms as the OP. 

More precisely, using the full script from [https://github.com/openwrt/openwrt/blob/openwrt-23.05/target/linux/ath79/generic/base-files/etc/hotplug.d/firmware/11-ath10k-caldata](url)

Although my **_VLAN_** setup had to be adapted, (more exactly the **_VAP_** part of it that was related to the way (intended way perhaps) that the (prevailing) **_Switch_** configuration demands. 

After having migrated from 23.05.5 to 24.10.0, not only was the AC wireless hindered but also, after having applied the above solution, out of the previously working **Network** bridge devices for each **_VLAN_** only one was working (the main **_VLAN_**), because it was correctly sitting on top of the correctly configured `bridge device` for the main **_VLAN_**. 

My main **_VLAN_** is "tagged" with number one (by default it's now `eth0.1`, (and the other two that I'm using are `eth0.20` and `eth0.30`).

Before doing the upgrade I had configured _**(virtual)**_ bridges **_(sub devices)_** that were used as underlying layers for their **Network interfaces** counterparts. 

For instance (maybe, wrongly, **before** the upgrade), I had the need to create **_VLAN_** devices, (under the **_Switch_** configuration page). After that, under the **_devices_** tab on the **_networks_** page, I had to create bridges for each of the created _**VLAN**_ "tags", and only then could they (the virtual _**Network devices**_) be used with (the creation and attribution of) the _**Network interfaces**_ to the previously created _**Network "objects"**_. 

After migration (upgrading), maybe predictably, there were already preconfigured _**VLAN**_ bridge devices under the _**devices**_ tab, with the **_same name_** as their **_Network interfaces_** "siblings" . By the way, I think that previously, the designation of the bridge devices at the **_Network devices_** tab, (or any other **device** name for any other **device type**), could not have the **_same name_** as a preexisting **_Network interface_** name (and vice-versa). Now I think it's standard. Can anyone confirm that? 

Then, after having realized that I had **_unneeded repetition_** with Network **devices** pointing to ("dead end") Network **interfaces**, I was then able to fix those inoperable VAP.

Final note:

Maybe this testament is irrelevant to the OP issue. 

Nevertheless, it may be relevant to anyone, facing the same issue of VAP that stopped working even after applying the suggested workaround. 

Since I don't quite know where should I have reported, I decided to post it here, in the hope that it could help anyone else. 

Thank you.

Benik3, 2025-04-04:
> Just one question: Does 5GHz WiFi now work with latest OpenWRT release 24.10.0 or still not?
>
> I had a hard time to update / downgrade my Archer C6 routers after I realized that 5G not working and do [...]

No, it doesn't work, you need to copy the mentioned board-2.bin file (which will be overwritten with OWRT update of course).
It's not mentioned on router page, true, but it's in release notes under known issues for 24.10.0 so you should mainly read this :)
https://openwrt.org/releases/24.10/notes-24.10.0