openwrt/openwrt #17178: AP+STA don't work on 5Ghz radio

saeid-lab, 2024-12-05:
### Describe the bug

Hello,

Trying to set up a repeater on Luci and unfortunately, I'm not able to use 5Ghz radio for this. Either Master OR client goes on (but not together)
Oddly, the same configuration works perfectly on 2.4Ghz.

Here is a piece of log:
```
daemon.err hostapd: Failed to set beacon parameters
daemon.notice hostapd: handle_probe_req: send failed
```
I noticed this problem was reported several times and yet I couldn't find any solution.
Appreciate the help
Regards

### OpenWrt version

23.05.5 r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

saeid-lab, 2024-12-05:
> Invalid Version reported. `23.05.5 r24106-10cc5fcd00` Is this from a clean repository?

A few hours ago I upgraded the OpenWrt using the [sysupgrade image](https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt7621&id=xiaomi_mi-router-4a-gigabit) bin file in luci.

raenye, 2024-12-05:
Did you try a WDS setup?