openwrt/openwrt #17180: apk: not support HTTP 30x link redirection

muink, 2024-12-06:
### Describe the bug

apk does not support HTTP 30x link redirection

### OpenWrt version

r0-b4cc574

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

1. Append `https://github.com/fantastic-packages/rebuild/raw/refs/heads/SNAPSHOT-x86-64/targets/x86/64/packages/packages.adb` to `/etc/apk/repositories.d/distfeeds.list` will got:
``` shell
fetch https://github.com/fantastic-packages/rebuild/raw/refs/heads/SNAPSHOT-x86-64/targets/x86/64/packages/packages.adb
WARNING: updating and opening https://ghp.ci/github.com/fantastic-packages/rebuild/raw/refs/heads/SNAPSHOT-x86-64/targets/x86/64/packages/packages.adb: remote server returned error (try 'apk update')
```

2. But `https://raw.githubusercontent.com/fantastic-packages/rebuild/refs/heads/SNAPSHOT-x86-64/targets/x86/64/packages/packages.adb` works fine:
``` shell
fetch https://raw.githubusercontent.com/fantastic-packages/rebuild/refs/heads/SNAPSHOT-x86-64/targets/x86/64/packages/packages.adb
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-06:
Can you try installing the full GNU wget?

I suspect this is a limitation in uclient-fetch

muink, 2024-12-06:
@robimarko After I installed the following packages, apk worked fine:
```
cgi-io libiwinfo libiwinfo-data liblucihttp liblucihttp-ucode luci luci-app-firewall luci-base 
luci-light luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 
luci-proto-ppp luci-ssl luci-theme-bootstrap px5g-mbedtls rpcd rpcd-mod-file rpcd-mod-iwinfo 
rpcd-mod-luci rpcd-mod-rrdns rpcd-mod-ucode ucode-mod-html ucode-mod-math uhttpd uhttpd-mod-ubus
```
I'm not sure which dependency is missing which is causing this. but `wget-nossl` or `wget-ssl` doesn't install.

Maybe we should add this dependency to config.buildinfo of SNAPSHOT by default?

robimarko, 2024-12-08:
Which dependency, since you installed a whole bunch of packages?

muink, 2024-12-09:
@robimarko I retested. Must installing `wget-ssl` get HTTP 30x link redirection support.
Installing the above packages does not solve the problem.

robimarko, 2024-12-09:
@muink No, we wont be including the full GNU wget by default, uclient-fetch is the minimized version