openwrt/openwrt #17181: Why you stop supporting huawei hg556a version A?

Foufa25, 2024-12-06:
### Describe the bug

Why you stop supporting huawei hg556a version A in snapshots feed and 24.10 rc ? And the hg556a ver B is supported by the two feed where the two versions have the same hardware identically !

### OpenWrt version

24.10 rc and snapshots

### OpenWrt release

24.10 rc and snapshots

### OpenWrt target/subtarget

24.10 rc and snapshots

### Device

HG556A ver A

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-06:
The support is in -main branch and snapshots , you could make it back into stable branch by cherry picking support from -main into 24.10 and testing that it works.
Come back with PR or issue report confirming it is fit for your hardware.

Foufa25, 2024-12-06:
> The support is in -main branch and snapshots , you could make it back into stable branch by cherry picking support from -main into 24.10 and testing that it works. Come back with PR or issue report confirming it is fit for your hardware.

No snapshot for hg556a ver A after i test the b firmware  work but no wifi detected  
https://downloads.openwrt.org/snapshots/targets/bmips/bcm6358/

brada4, 2024-12-06:
Helps to click on first viable link sometimes....
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=bcm63xx%2Fgeneric&id=huawei_echolife-hg556a-a

Foufa25, 2024-12-06:
> Helps to click on first viable link sometimes.... https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=bcm63xx%2Fgeneric&id=huawei_echolife-hg556a-a

This snapshots for hg556a stop 07/2024 so its snapshot for 23.05 build

Noltari, 2025-01-04:
This was also asked on https://github.com/openwrt/openwrt/pull/14559?w=1#issuecomment-2546333053

> @Foufa25 a lot of bcm63xx weren't migrated because I didn't have time to do it.
> However, in the case of HG556a A, it wasn't migrated because the switch isn't supported correctly, which affects all BCM6358 devices.
> 
> I had a Pull Request in which I made some efforts to get it working but unfortunately I never could:
> https://github.com/openwrt/openwrt/pull/12285

Foufa25, 2025-01-06:
Thanks for your big effort to support the old devices and keep it alive
with openwrt . But HG556 ver B is the same as ver A with little changes in
partitions size and another atheros wifi driver please add it if you can .
Thanks again