openwrt/openwrt #17183: Ruijie RG-X60 Pro WAN Port Not Working - (Realtek RTL8221B-VB-CG)

v4lkyr, 2024-12-06:
### Describe the bug

WAN port on my Ruijie RG-X60 Pro, which is controlled by a Realtek RTL8221B-VB-CG is not working and in a dead state. I have tried several builds and all of them have the same problem. I also built openwrt myself, from main branch (can be seen [here](https://forum.openwrt.org/t/ruijie-rg-x60-pro-wan-problem/217446/27)) still not working.

Image kind = both official downloaded image and self-built ones.

### OpenWrt version

24.10-rc2, snapshot r28252-1aa2695620

### OpenWrt release

1aa2695620

### OpenWrt target/subtarget

mediatek/filogic - mt7986a

### Device

Ruijie RG-X60 Pro

### Image kind

Official downloaded image

### Steps to reproduce

Flashing the "sysupgrade" version of image on TTL via tftp, checking the logs. Observing that socket is not working without any modifications. `logread -f` doesn't show anything when plugging the cable in and out on WAN socket. LAN sockets are fine.

### Actual behaviour

WAN port isn't working _at all._

### Expected behaviour

WAN port working when I plug in cable and the small activity led in the corner blinking.

### Additional info

I have tried 24.10-rc2 official build for my device and then built the main branch myself with minimal config for my device + added luci. WAN port not working at all. Also I don't see RTL8221 getting loaded on the boot log, too.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

v4lkyr, 2024-12-06:
> Invalid Version reported. `24.10-rc2, snapshot r28252-1aa2695620` Is this from a clean repository?

Dear bot, I wanted to emphasize I've tried multiple versions. Do you not have a support for that?

v4lkyr, 2024-12-06:
> Invalid Target/Subtarget reported. `mediatek/filogic - mt7986a` Is this from a supported device?

Well that's the target and I'm 100% sure. Maybe I messed up the syntax for it.

v4lkyr, 2024-12-06:
> Invalid Release reported. `1aa2695620` Is this from a clean repository?

Yes, it's from a clean repository. It's the `main `branch.

v4lkyr, 2024-12-12:
> I encountered a similar problem with the RTL8221B 2.5Gbps port on my [Netgear WAX206](https://openwrt.org/toh/netgear/wax206) after upgrading (via ASU) from 23.05.5 to 24.10.0-rc2.
> 
> In my case, this appeared to be because link speed autonegotiation was disabled by default (and the peer on this port only supports up to 1Gbps). Adding the `autoneg` option explicitly to the `wan` device in `/etc/config/network` fixed the issue for me:
> 
> ```
> config device
> 	option name 'wan'
> 	option mtu '1508'
> 	option autoneg '1'
> ```
> 
> YMMV, but this might indicate the cause and a workaround for other similar cases.
> 
> Notes that may or may not be relevant:
> 
> * In my case, I already had the rest of this block with the `mtu` in my 23.05.5 config, to support full-size frames without fragmentation through a PPPoE WAN link. That may or may not be relevant.)
> * I'd already installed the `ethtool` package and (successfully) used that to manually enable link autonegotiation before discovering the config file option to ensure it's re-enabled after a reboot. (I don't know whether ethtool is required by the code that handles the config file option or not.)

Interesting.

However, in my case, WAN port appears completely dead, nothing happens when plugging in and out the cable, `logread -f `doesn't show anything too.

Was it the same with you before adding `autoneg` too?

Unfortunately, I can't try 23.05.5 version because the support to my device added after that version is released.

zak-k, 2024-12-13:
> However, in my case, WAN port appears completely dead, nothing happens when plugging in and out the cable, logread -f doesn't show anything too.

I didn't check for detection/logging of plug/unplug events while it wasn't working, I'm afraid. The interface and port were already recognised and attached to the driver, and showing in the boot log, but the _link_ didn't come up without first enabling autonegotiation. (I'm not sure whether or not you'd expect plug/unplug to be detected if an Ethernet link can't be established.)

hauke, 2024-12-25:
@dangowrt Could you please have a look at this and the related bug reports about the RTL8221B PHY auto negotiation problems please.

zak-k, 2024-12-26:
To follow up on my comment above, I can confirm the issue is still present in rc4 (no link is established on the WAX206 `wan` interface unless I explicitly enable autonegotiation via `ethtool` or `/etc/config/network`).

dangowrt, 2024-12-26:
@hauke The issue here is that on all devices with RTL8221B I own (TL-XDR6086, GL.iNet MT-6000) the PHY works just fine incl. autonegotiation. I have no idea why the kernel default for autoneg differs from device to device with supposedly the exact same phy -- the only way to explain that to me is that the reset GPIO could be wrong, and validating that obviously requires physical access.

aiamadeus, 2024-12-27:
@dangowrt I can confirm that it's not a problem with the reset GPIO.
I managed to reproduce this issue on another router, the problem is
in the "Disable SGMII AN" section. 
```
--- a/drivers/net/phy/realtek.c
+++ b/drivers/net/phy/realtek.c
@@ -887,9 +887,13 @@ static int rtl822xb_config_init(struct phy_device *phydev)
 		return ret;
 
 	ret = phy_read_mmd_poll_timeout(phydev, MDIO_MMD_VEND1, 0x7587,
-					val, !(val & BIT(0)), 500, 100000, false);
-	if (ret < 0)
+					val, !(val & BIT(0)), 500, 5000000, false);
+	if (ret < 0) {
+		phydev_err(phydev, "Polling 0x7587.0, val=0x%x\n", val);
 		return ret;
+	} else {
+		phydev_err(phydev, "Polling 0x7587.0 done");
+	}
 
 	return 0;
 }
```
When plugging in the Internet cable to the Wan port (via rtl8221b),
I got this output through the above patch:
```
[   17.757367] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
[   17.765719] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: Polling 0x7587.0, val=0x3
[   17.886459] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
```
Since the timeout is set to 5s and this problem still occurs, I think it's not the phy driver bug.

ab77, 2024-12-27:
In my case ([TR3000 v1](https://github.com/openwrt/openwrt/issues/17232)), adding `autoneg '1'` to the `wan` interface didn't help, however swapping eth0 and eth1  using labels in the DTS did, e.g.:

```
&eth {
	pinctrl-names = "default";
	pinctrl-0 = <&mdio_pins>;

	status = "okay";

	gmac0: mac@0 {
		compatible = "mediatek,eth-mac";
		reg = <0>;
		label = "lan1";
		phy-mode = "2500base-x";
		phy-handle = <&phy1>;
		nvmem-cell-names = "mac-address";
		nvmem-cells = <&macaddr_bdinfo_de00 1>;
	};

	gmac1: mac@1 {
		compatible = "mediatek,eth-mac";
		reg = <1>;
		label = "wan";
		phy-mode = "gmii";
		phy-handle = <&int_gbe_phy>;
		nvmem-cell-names = "mac-address";
		nvmem-cells = <&macaddr_bdinfo_de00 0>;
	};
};
```

aiamadeus, 2024-12-28:
After a long time of comparison and attempts, I'm sure that this problem has
nothing to do with the kernel, but what application caused it is still unknown.

casta, 2025-01-03:
> I encountered a similar problem with the RTL8221B 2.5Gbps port on my [Netgear WAX206](https://openwrt.org/toh/netgear/wax206) after upgrading (via ASU) from 23.05.5 to 24.10.0-rc2.
>
> In my case, this [...]

Hello,

WAX206 here, same issue with 24.10_rc4 (wan was staying in LOWERLAYERDOWN state), while wan was perfectly working with 23.05.5.
Your workaround of forcing autoneg on en /etc/config/network is working for me too. My wan upstream peer is 1Gbps, like your seems to be too.

If that may help:
```
root@coruscant:~# dmesg  | grep mt7530-mdio
[    2.604002] mt7530-mdio mdio-bus:1f: configuring for fixed/2500base-x link mode
[    2.612902] mt7530-mdio mdio-bus:1f: Link is Up - 2.5Gbps/Full - flow control rx/tx
[    2.624095] mt7530-mdio mdio-bus:1f lan1 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7531 PHY] (irq=133)
[    2.646411] mt7530-mdio mdio-bus:1f lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7531 PHY] (irq=134)
[    2.668435] mt7530-mdio mdio-bus:1f lan3 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7531 PHY] (irq=135)
[    2.690444] mt7530-mdio mdio-bus:1f lan4 (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7531 PHY] (irq=136)
[    2.938320] mt7530-mdio mdio-bus:1f wan (uninitialized): PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)
[    4.152027] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[    5.019758] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[    5.907827] mt7530-mdio mdio-bus:1f lan1: Link is Down
[   14.594091] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[   14.623578] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
[   14.643197] mt7530-mdio mdio-bus:1f lan1: entered promiscuous mode
[   14.684402] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   14.713040] mt7530-mdio mdio-bus:1f lan2: entered allmulticast mode
[   14.722912] mt7530-mdio mdio-bus:1f lan2: entered promiscuous mode
[   14.746497] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
[   14.779486] mt7530-mdio mdio-bus:1f lan3: entered allmulticast mode
[   14.790721] mt7530-mdio mdio-bus:1f lan3: entered promiscuous mode
[   14.816486] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
[   14.838656] mt7530-mdio mdio-bus:1f lan4: entered allmulticast mode
[   14.848099] mt7530-mdio mdio-bus:1f lan4: entered promiscuous mode
[   14.863999] mt7530-mdio mdio-bus:1f wan: configuring for phy/2500base-x link mode
[   17.120696] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.040376] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.186806] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[   19.129776] mt7530-mdio mdio-bus:1f lan2: Link is Down
[   21.277755] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
```

By the way, I’m not using the default image but a build generated in https://firmware-selector.openwrt.org with `luci-ssl` `luci-proto-wireguard` `mdio-tools` in addition to the default package set.

ab77, 2025-01-09:
To add, in 23.05.current, on Cudy TR3000 V1 this issue manifests itself slightly differently. The WAN interface does have layer 2 link according to ethtool/ifconfig/etc., negotiates the right speed, but has pending status in `ifstatus` and no DHCP IP until `/etc/init.d/network restart` is issued:

```
[   21.320818] RTL8221B-VB-CG 2.5Gbps PHY mdio-bus:01: rtl8221b_config_init failed: -110
[   21.360155] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[   21.370256] mtk_soc_eth 15100000.ethernet wan: configuring for phy/2500base-x link mode
...
[   24.550714] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   24.550729] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
```

```
$ ifconfig wan
wan       Link encap:Ethernet  HWaddr 80:AF:CA:31:XX:XX
          inet6 addr: fe80::82af:caff:fe31:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:6044 (5.9 KiB)
          Interrupt:82 

$ ifstatus wan
{
        "up": false,
        "pending": true,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcp",
        "device": "wan",
        "data": {

        }
}

$ ip link
2: wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 80:af:ca:31:xx:xx brd ff:ff:ff:ff:ff:ff

$ ethtool wan
Settings for wan:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
                                             1000baseT/Full 
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes
```

.. then, after network restart:

```
$ /etc/init.d/network restart                                                                                                      
[  222.176460] br-lan: port 2(wlan1) entered disabled state                                                                                           
[  222.182872] device lan1 left promiscuous mode                                                                                                      
[  222.187326] br-lan: port 1(lan1) entered disabled state                                                                                            
[  222.242495] device wlan1 left promiscuous mode
[  222.247005] br-lan: port 2(wlan1) entered disabled state
[  222.719950] mtk_soc_eth 15100000.ethernet wan: Link is Down
[  224.342394] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[  224.356905] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[  224.368743] br-lan: port 1(lan1) entered blocking state
[  224.374011] br-lan: port 1(lan1) entered disabled state
[  224.379394] device lan1 entered promiscuous mode
[  224.770150] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[  224.780256] mtk_soc_eth 15100000.ethernet wan: configuring for phy/sgmii link mode
[  225.382531] br-lan: port 2(wlan1) entered blocking state
[  225.387851] br-lan: port 2(wlan1) entered disabled state
[  225.393426] device wlan1 entered promiscuous mode
[  226.362143] br-lan: port 2(wlan1) entered blocking state
[  226.367467] br-lan: port 2(wlan1) entered forwarding state
[  226.373613] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  227.910622] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  227.919099] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready

$ ifconfig wan
wan       Link encap:Ethernet  HWaddr 80:AF:CA:31:XX;XX
          inet addr:192.168.10.186  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::82af:caff:fe31:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:227 errors:0 dropped:4 overruns:0 frame:0
          TX packets:272 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:169525 (165.5 KiB)  TX bytes:51275 (50.0 KiB)
          Interrupt:82 
```

ab77, 2025-01-10:
I've also tried configuring `phy-mode = "gmii";` in the DTS, replacing `phy-mode = "2500base-x";`, got a different error `mtk_soc_eth 15100000.ethernet: mtk_mac_config: GMAC0 mode gmii err: -22!` and an unworkable WAN interface with link, but no DHCP even after running `/etc/init.d/network restart`:

```
$ dmesg | grep mtk_soc_eth
[    2.341186] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc009480000, irq 82
[    2.351198] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc009480000, irq 82
[    4.180458] mtk_soc_eth 15100000.ethernet wan: renamed from eth0
[    4.250212] mtk_soc_eth 15100000.ethernet lan1: renamed from eth1
[    4.331891] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[    4.346752] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[    8.488254] mtk_soc_eth 15100000.ethernet lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   13.127965] mtk_soc_eth 15100000.ethernet lan1: Link is Down
[   13.161735] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   13.176715] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[   13.478032] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[   13.488148] mtk_soc_eth 15100000.ethernet wan: configuring for phy/gmii link mode
[   13.495634] mtk_soc_eth 15100000.ethernet: mtk_mac_config: GMAC0 mode gmii err: -22!
[   16.648599] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   17.368231] mtk_soc_eth 15100000.ethernet lan1: Link is Up - 1Gbps/Full - flow control rx/tx

$ ip l
2: wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether 80:af:ca:31:xx:xx brd ff:ff:ff:ff:ff:ff

$ ifstatus wan
{
        "up": false,
        "pending": true,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcp",
        "device": "wan",
        "data": {

        }
}

$ ifconfig wan
wan       Link encap:Ethernet  HWaddr 80:AF:CA:31:XX:XX
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:54 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:15216 (14.8 KiB)
          Interrupt:82 

$ ethtool wan
Settings for wan:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
                                             1000baseT/Full 
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes
```

dangowrt, 2025-01-15:
I've ordered Cudy TR3000 from ebay today, sadly the Ruijie RG-X60 Pro cannot be purchased easily apparently, and the WAX206 is ridiculously expensive... I will try to reproduce and fix the issue on the TR3000 once I got it.

ab77, 2025-01-15:
@dangowrt let me know if you need anything from me as far as testing goes. In the meantime, I've kicked off a build with DTS changes as per [this PR](https://github.com/openwrt/openwrt/pull/17600) to see if that fixes the PHY issue, specifically [these](https://github.com/openwrt/openwrt/pull/17600/files#diff-e83c3b4055f5c90494f5d249c0fd786f8c0d88f144582dc1fab7111f6b9f4b59L108-R110).

aiamadeus, 2025-01-15:
Unfortunately, the above PR has nothing to do with this issue.
This bug only occurs when booting with negotiated 2.5G.
I'm busy lately and don't have much time to debug this.

dangowrt, 2025-01-17:
@zak-k @aimadeus If anyone can give me access to a test setup, ideally including access to the serial port, that would be truly amazing. Otherwise I will setup Cudy device myself in about a week from now, it was shipped yesterday and will supposedly take about a week to get to me.

ab77, 2025-01-18:
@dangowrt I have a spare stock tr3kv1 here I can give you access to via a terminal device if you like..

v4lkyr, 2025-01-21:
> [@zak-k](https://github.com/zak-k) [@aimadeus](https://github.com/aimadeus) If anyone can give me access to a test setup, ideally including access to the serial port, that would be truly amazing. Otherwise I will setup Cudy device myself in about a week from now, it was shipped yesterday and will supposedly take about a week to get to me.
> 
> Edit: I don't think it will fix the issue, but I'd be glad if people would anyway help and test [#17634](https://github.com/openwrt/openwrt/pull/17634)

Hello Dan! @dangowrt 

My Ruijie RG-X60 Pro is at your command, with TTL access and all. I am in GMT+8 zone.

dangowrt, 2025-01-21:
@v4lkyr Great. I'm ready when you are. Contact me by email regarding the credentials.

My Cudy TR3000 v1 has also arrived today and I see the problem with current snapshots:
```
[   12.907612] RTL8221B-VB-CG 2.5Gbps PHY (C45) mdio-bus:01: rtl822xb_config_init failed: -110
[   13.026607] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
```

The fix it easy: increase `reset-deassert-us` from `100000` to `300000`. I assume the difference with other devices for which it works with 100000us is that the reset circuits on the devices which need to prolonged deassert time probably miss a pull-up resistor, which is why it takes longer for PHY to come out of reset. The fact that it was working on older OpenWrt versions can be explained by phylink doing things slightly different, so the involuntary delay between deassert and bus access became smaller...

ab77, 2025-01-21:
> `[   14.128982] RTL8221B-VB-CG 2.5Gbps PHY mdio-bus:01: rtl8221b_config_init failed: -110` is still there on boot.

I tried setting a higher deassert, but it appears to have had no effect. Some hopefully relevant messages from the kernel log. I don't know if `can't parse 'reset-gpio' property of node ...` are distractions or related to this problem.

```
root@OpenWrt:/# dmesg | grep mtk_soc_eth
[    2.741127] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc009400000, irq 82
[    2.751146] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc009400000, irq 82
[    4.585760] mtk_soc_eth 15100000.ethernet wan: renamed from eth0
[    4.640144] mtk_soc_eth 15100000.ethernet lan1: renamed from eth1
[    4.701890] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[    4.716761] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[   13.539793] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   13.554791] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[   14.167961] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=62)
[   14.178306] mtk_soc_eth 15100000.ethernet wan: configuring for phy/2500base-x link mode
[   16.964410] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
```

```
root@OpenWrt:/# dmesg | grep eth
[    0.000000] psci: probing for conduit method from DT.
[    1.687992] mt7981-pinctrl 11d00000.pinctrl: request pin 36 (SMI_MDC) for 15100000.ethernet
[    1.696331] mt7981-pinctrl 11d00000.pinctrl: request pin 37 (SMI_MDIO) for 15100000.ethernet
[    1.704762] mt7981-pinctrl 11d00000.pinctrl: enable function eth group smi_mdc_mdio
[    1.852508] of_get_named_gpiod_flags: can't parse 'reset-gpios' property of node '/ethernet@15100000/mdio-bus[0]'
[    1.862780] of_get_named_gpiod_flags: can't parse 'reset-gpio' property of node '/ethernet@15100000/mdio-bus[0]'
[    1.897286] of_get_named_gpiod_flags: can't parse 'reset-gpios' property of node '/ethernet@15100000/mdio-bus/ethernet-phy@0[0]'
[    1.908867] of_get_named_gpiod_flags: can't parse 'reset-gpio' property of node '/ethernet@15100000/mdio-bus/ethernet-phy@0[0]'
[    1.959729] of_get_named_gpiod_flags: parsed 'reset-gpios' property of node '/ethernet@15100000/mdio-bus/phy@1[0]' - status (0)
[    2.741127] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc009400000, irq 82
[    2.751146] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc009400000, irq 82
[    4.585760] mtk_soc_eth 15100000.ethernet wan: renamed from eth0
[    4.640144] mtk_soc_eth 15100000.ethernet lan1: renamed from eth1
[    4.701890] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[    4.716761] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[    8.623180] mt7981-pinctrl 11d00000.pinctrl: enable function eth group wf0_mode1
[   13.539793] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   13.554791] mtk_soc_eth 15100000.ethernet lan1: configuring for phy/gmii link mode
[   14.167961] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=62)
[   14.178306] mtk_soc_eth 15100000.ethernet wan: configuring for phy/2500base-x link mode
[   16.964410] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
```

```
root@OpenWrt:/# dmesg | grep mdio-bus
[    1.840730] mdio_bus mdio-bus: GPIO lookup for consumer reset
[    1.846479] mdio_bus mdio-bus: using device tree for GPIO lookup
[    1.852508] of_get_named_gpiod_flags: can't parse 'reset-gpios' property of node '/ethernet@15100000/mdio-bus[0]'
[    1.862780] of_get_named_gpiod_flags: can't parse 'reset-gpio' property of node '/ethernet@15100000/mdio-bus[0]'
[    1.872954] mdio_bus mdio-bus: using lookup tables for GPIO lookup
[    1.879128] mdio_bus mdio-bus: No GPIO consumer reset found
[    1.885007] mdio_bus mdio-bus:00: GPIO lookup for consumer reset
[    1.891022] mdio_bus mdio-bus:00: using device tree for GPIO lookup
[    1.897286] of_get_named_gpiod_flags: can't parse 'reset-gpios' property of node '/ethernet@15100000/mdio-bus/ethernet-phy@0[0]'
[    1.908867] of_get_named_gpiod_flags: can't parse 'reset-gpio' property of node '/ethernet@15100000/mdio-bus/ethernet-phy@0[0]'
[    1.920345] mdio_bus mdio-bus:00: using lookup tables for GPIO lookup
[    1.926774] mdio_bus mdio-bus:00: No GPIO consumer reset found
[    1.947432] mdio_bus mdio-bus:01: GPIO lookup for consumer reset
[    1.953452] mdio_bus mdio-bus:01: using device tree for GPIO lookup
[    1.959729] of_get_named_gpiod_flags: parsed 'reset-gpios' property of node '/ethernet@15100000/mdio-bus/phy@1[0]' - status (0)
[    4.701890] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   13.539793] mtk_soc_eth 15100000.ethernet lan1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   14.128982] RTL8221B-VB-CG 2.5Gbps PHY mdio-bus:01: rtl8221b_config_init failed: -110
[   14.167961] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=62)
```

zak-k, 2025-01-22:
> Edit: I don't think it will fix the issue, but I'd be glad if people would anyway help and test [#17634](https://github.com/openwrt/openwrt/pull/17634)

I've tested on my WAX206 with 24.10-SNAPSHOT, r28367-96ff9c3353, which appears to include a cherry-pick of that commit - as suspected, no discernible impact on this bug - it seems the kernel still initialises the port with autonegotiation off, requiring it to be manually enabled before a link can be established. There are no dmesg differences that appear to relate to this port in any way between rc5 and the snapshot.

Re-flashing 23.05.5 temporarily, the port again worked as expected, establishing a link on boot. I don't see anything obviously relevant that differs in dmesg relating to this port, mostly just mdio-bus renumbering - the only differences which appear to relate to this port are:
```
< [    2.661845] mt7530-mdio mdio-bus:00 wan (uninitialized): PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
< [   10.452917] mt7530-mdio mdio-bus:00 wan: configuring for phy/2500base-x link mode
< [   14.654368] mt7530-mdio mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control off
---
> [    2.958595] mt7530-mdio mdio-bus:1f wan (uninitialized): PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)
> [   14.932151] mt7530-mdio mdio-bus:1f wan: configuring for phy/2500base-x link mode
```
(23.05.5 is above; 24.10.0-rc5 below. With the latter, an equivalent "link is up" line appears _only_ after manually enabling autonegotiation.)

Unfortunately, this device is mostly in operational use so it's not straightforward to set it up for remote testing, and I don't currently have the means to establish a serial connection, although I'm happy to run diagnostics or test further potential fixes.

dangowrt, 2025-01-23:
I did a `git bisect` and found the first bad commit where things went south is ca4469045f6a4845dbf8aeabd4b04b421209f66e. Going into the details now...

dimitris-personal, 2025-01-24:
FWIW I have a WAX206 running 24.10.0-rc5 with the WAN port connected to a 2.5G cable modem.  It seems to come up and autonegotiate without issues:

```
[    2.968586] mt7530-mdio mdio-bus:1f wan (uninitialized): PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)
...
[   15.938269] mt7530-mdio mdio-bus:1f wan: configuring for phy/2500base-x link mode
[   15.979594] mt7530-mdio mdio-bus:1f wan: Link is Up - 2.5Gbps/Full - flow control off
```

dangowrt, 2025-01-24:
@dimitris-personal Are you sure that the PHY performed auto-negotiation and not just ended up using fixed 2500M link? Because the latter always worked and has never been a problem. Please check with `ethtool wan` (install `ethtool` if needed) and tell us if you see `Auto-negotiation: on`. I'd be surprised.

All that being said, I found this interesting looking commit in MediaTek's SDK repository today and I wonder if it could be related:
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/unified/global/24.10/files/target/linux/mediatek/patches-6.6/999-2700-net-ethernet-mtk_eth_soc-add-mdio-reset-delay.patch

dimitris-personal, 2025-01-24:
@dangowrt indeed, it seems it just used 2.5G without autoneg:

```
# ethtool wan
Settings for wan:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: No
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: No
	Advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 7
	Transceiver: external
	Auto-negotiation: off
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes
```

It looks like the a fix has been [confirmed for the WAX206 at least](https://github.com/openwrt/openwrt/pull/17701#issuecomment-2610407171).  If it's not superfluous I'd offer to test on my (spare) WAX206 if I could get around [my newbie build system showstopper](https://forum.openwrt.org/t/first-time-attempt-to-build-a-package-cant-locate-metadata-pm/222230/3).  I assume that would be just a cherry pick of #17701 into the 24.10 branch?

closed 2025-01-26 by openwrt-bot

zak-k, 2025-01-26:
Confirmed now working on my WAX206 in latest 24.10-SNAPSHOT, r28404-d30773abe8, which includes the above fix/workaround. (Interface now has auto-negotiation enabled out of the box on boot, and correctly establishes a link with a 1Gbps peer without having to manually enable this.)

Thanks all!

v4lkyr, 2025-02-08:
I don't know why this issue has turned into problems of other devices such as WAX206 or Cudy TR3000, but I have a Ruijie RG-X60 Pro (image: [https://images.shoutwiki.com/techinfodepot/e/ec/RG-X60_PRO.jpg](url) ) and the problem I've mentioned in this issue **_still persists._** (as of SNAPSHOT r28780-4b6886d9fd) Let's please stick to Ruijie RG-X60 Pro. You can always create new issues for your own devices.

@dangowrt

dangowrt, 2025-02-08:
@v4lkyr Please share a complete bootlog of the device running latest snapshot and your `/etc/config/network` (removing credentials, obviously). If that doesn't help the best way for me to figure out what's happening could be to have remote access.

v4lkyr, 2025-02-09:
@dangowrt Dear Dan, I talked too soon, apologies for that. The WAN port is now working with SNAPSHOT r28780-4b6886d9fd. However, the mini green activity LED on the port is not working at all, so that misled me. I was looking at the LED only at that time haha

Here's the bootlog: https://pastebin.com/bPGUH9ay

Here's the config: https://pastebin.com/wEd5EL4A

dangowrt, 2025-02-09:
@v4lkyr No worries. And it's true that LEDs connected to RTL822x do need some love...

PR #17701 (2025-01-26): generic: fix probe issues with RealTek RTL8221B PHYs
Import patch "net: phy: realtek: mark existing MMDs as present"

    When using Clause-45 mode to access RealTek RTL8221B 2.5G PHYs some
    versions of the PHY fail to report the MMDs present on the PHY.
    Mark MMDs PMAPMD, PCS and AN which are always existing according to
    the datasheet as present to fix that.

Fixes: #17183, #17232

I'm not sure this fix is actually correct and fixes the root cause of the issue, it might rather just be a work-around for a deeper problem with the behavior of those PHYs right after reset...