openwrt/openwrt #17184: Unable to install packages. Checksum error

Eksol, 2024-12-06:
### Describe the bug

I'm trying to install curl but I get an error.

root@OpenWrt:~# opkg install curl

Installing curl (8.7.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a53/packages/curl_8.7.1-r1_aarch64_cortex-a53.ipk
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package curl. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package curl.

I tried updating the 'opkg update' but it didn't help.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

opkg update 
opkg install curl

### Actual behaviour

opkg_install_cmd: Cannot install package curl.

### Expected behaviour

opkg_install_cmd: Cannot install package curl.

### Additional info

successful installation of curl

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-12-06:
try curl-ssl

Eksol, 2024-12-06:
A simple reboot of the router helped.

closed 2024-12-06 by Eksol