openwrt/openwrt #17185: [24.10] -rc2 breaks WIFI - Google Wifi

GoetzGoerisch, 2024-12-06:
### Describe the bug

Wireless device fails with:

```
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  341.923146] ------------[ cut here ]------------
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  341.934012] WARNING: CPU: 2 PID: 1198 at backports-6.11.2/net/mac80211/main.c:272 0xbf230888 [mac80211@86cb772f+0x77000]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  341.938766] Modules linked in: nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) vpd_sysfs nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 memconsole_coreboot memconsole libcrc32c hwmon coreboot_table compat(O) sha512_arm ghash_arm_ce cmac leds_gpio ramoops reed_solomon pstore gpio_button_hotplug(O) f2fs ext4 mbcache jbd2 crc32c_generic crc32_generic
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  341.987058] CPU: 2 PID: 1198 Comm: hostapd Tainted: G        W  O       6.6.63 #0
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.009079] Hardware name: Generic DT based system
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.016451] Function entered at [<c030d1d8>] from [<c0308ec8>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.021133] Function entered at [<c0308ec8>] from [<c0a3e96c>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.026948] Function entered at [<c0a3e96c>] from [<c032165c>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.032765] Function entered at [<c032165c>] from [<c03217dc>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.038580] Function entered at [<c03217dc>] from [<bf230888>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.044398] Function entered at [<bf230888>] from [<bf2309dc>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.050212] Function entered at [<bf2309dc>] from [<c0858830>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.056029] Function entered at [<c0858830>] from [<c0858c94>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.061846] Function entered at [<c0858c94>] from [<c0858d2c>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.067661] Function entered at [<c0858d2c>] from [<c0928754>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.073476] Function entered at [<c0928754>] from [<c092afa4>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.079292] Function entered at [<c092afa4>] from [<c082ac1c>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.085110] Function entered at [<c082ac1c>] from [<c04c634c>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.090924] Function entered at [<c04c634c>] from [<c0300040>]
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.096739] Exception stack(0xc4ffffa8 to 0xc4fffff0)
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.102561] ffa0:                   00000010 bee94828 00000010 00008914 bee94828 00000000
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.107686] ffc0: 00000010 bee94828 00000010 00000036 00000001 00000001 00000000 00000020
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.115843] ffe0: 0065ee60 bee946e8 00574bb8 b6f9766c
Wed Dec  4 16:04:28 2024 kern.warn kernel: [  342.124079] ---[ end trace 0000000000000000 ]---

```

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

ipq40xx/chromium

### Device

Google WiFi (Gale)

### Image kind

Official downloaded image

### Steps to reproduce

- install 24.10.0-rc2 from r28155

### Actual behaviour

- devices can no longer connect

```
Wed Dec  4 16:07:24 2024 daemon.notice hostapd: phy0-ap2: STA xx:xx:xx:xx:xx:f1 IEEE 802.11: did not acknowledge authentication response
Wed Dec  4 16:07:24 2024 daemon.notice hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:f1 IEEE 802.11: did not acknowledge authentication response
Wed Dec  4 16:07:24 2024 daemon.notice hostapd: phy0-ap1: STA xx:xx:xx:xx:xx:f1 IEEE 802.11: did not acknowledge authentication response

```

### Expected behaviour

Wireless working as before

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-12-06:
Any chance you can bisect this?

GoetzGoerisch, 2024-12-06:
No, sorry, don't know how to accomplish this.

As said last good revision was r28155

robimarko, 2024-12-06:
Do you at least have the commit hash from r28155?

GoetzGoerisch, 2024-12-06:
r28155-17bdccb4a5

GoetzGoerisch, 2024-12-06:
While trying to isolate this, I found in the logs:

```
Fri Dec  6 12:48:18 2024 kern.warn kernel: [  966.367017] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Fri Dec  6 12:48:18 2024 kern.warn kernel: [  966.367102] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
Fri Dec  6 12:48:18 2024 kern.info kernel: [  966.401997] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Fri Dec  6 12:48:18 2024 kern.info kernel: [  966.402979] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
Fri Dec  6 12:48:18 2024 kern.info kernel: [  966.683873] ath10k_ahb a000000.wifi: rts threshold -1
Fri Dec  6 12:48:18 2024 kern.warn kernel: [  966.684369] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4

```

GoetzGoerisch, 2024-12-06:
Checking with r28155 I already see this in the dmesg output.

robimarko, 2024-12-13:
I remember seeing those forever.

I looked at the git history and after `17bdccb4a5` there are only 3 commits and none of them are not kernel, wireless, hostapd related at all

GoetzGoerisch, 2024-12-13:
Ok, thank you.
I have investigated also further.
The device which cannot connect if no device I found in my household
Its a Tuya device which tries to connect to all three different SSIDs setup.

Thanks you your effort.

poranje, 2025-01-20:
> Ok, thank you. I have investigated also further. The device which cannot connect if no device I found in my household Its a Tuya device which tries to connect to all three different SSIDs setup.

Here also with a Tuya device (MAC: 1c:90:ff:xx:xx:xx = Tuya Smart) on 3 different SSIDs.
This is on a Zyxel EX5600-T1 (MediaTek MT7986 802.11ax/b/g/n) on the 2g band running 24.10.0-rc5.

Black listing the MAC helped as a work around.