openwrt/openwrt #17187: GL-Inet MT6000  phy1-ap1 doesn't get bought up on boot if an L3 interface

dragon2611, 2024-12-06:
### Describe the bug

I have a secondary SSID on the 5Ghz radio (So phy1-ap1) since this was for a network segment that only served wireless clients via the inbuilt AP the phy1-ap1 was configured as WLAN2 and had the IP address.etc assigned directly to the interface

On device bootup radio1 does not work, If I restart networking via /etc/init.d/networking restart or restart radio1 via luci it starts working.

Creating a dummy bridge (br-wlan2) and assigning the L3 Interface to the bridge seems to solve this issue. 

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Create a wireless interface that is not connected to a bridge but rather a L3 interface with an IP itself.

### Actual behaviour

Wireless does not autostart properly on boot

### Expected behaviour

wireless to start automatically on boot.

### Additional info

Suspect it may be a timing related issue/race condition.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2024-12-08:
I am using an empty bridge for this use case.
```
config device 'lantv_dev'
	option type 'bridge'
	option name 'br-lantv'
	option bridge_empty 1
```
Make sure to set the `bridge_empty` option.

dragon2611, 2024-12-08:
Empty bridge is how I’m doing it now.

pretty sure this was working on 23.05, although I did have a couple instances where the 5ghz stopped working (once after a day or so and once after a couple weeks).

Will look to post config later, (not there at the moment)