openwrt/openwrt #17194: Initramfs unpacking failed

ptpt52, 2024-12-07:
### Describe the bug

with config:
```
CONFIG_USES_SEPARATE_INITRAMFS=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_XZ=y
```

```
[    0.033243] Unpacking initramfs...
[    0.033254] Initramfs unpacking failed: compression method xz not configured
```

or 
```

CONFIG_USES_SEPARATE_INITRAMFS=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_LZMA=y

[    0.033243] Unpacking initramfs...
[    0.033254] Initramfs unpacking failed: compression method lzma-openwrt not configured
```

### OpenWrt version

x

### OpenWrt release

x

### OpenWrt target/subtarget

x

### Device

x

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2024-12-07:
ping @dangowrt

ptpt52, 2024-12-07:
build on `mediatek/filogic`

dangowrt, 2024-12-07:
Support for initramfs decompression is only compiled into the kernel image intended to be used for initramfs. The kernel image intended tomload squashfs never contains support for initramfs decompression, no matter if initramfs is bundled or separate.

ptpt52, 2024-12-07:
why `CONFIG_RD_LZMA` option is disabled by default for openwrt
```diff
diff --git a/target/linux/generic/config-6.6 b/target/linux/generic/config-6.6
index 4715ef134d2..4d27de4f5c2 100644
--- a/target/linux/generic/config-6.6
+++ b/target/linux/generic/config-6.6
@@ -4888,7 +4888,7 @@ CONFIG_RCU_STALL_COMMON=y
 # CONFIG_RD_BZIP2 is not set
 # CONFIG_RD_GZIP is not set
 # CONFIG_RD_LZ4 is not set
-# CONFIG_RD_LZMA is not set
+CONFIG_RD_LZMA=y
 # CONFIG_RD_LZO is not set
 # CONFIG_RD_XZ is not set
 # CONFIG_RD_ZSTD is not set

```

ptpt52, 2024-12-07:
something issue here?
```diff
diff --git a/include/kernel-defaults.mk b/include/kernel-defaults.mk
index f94ed332305..d7a0b67b8bc 100644
--- a/include/kernel-defaults.mk
+++ b/include/kernel-defaults.mk
@@ -49,7 +49,7 @@ ifeq ($(CONFIG_TARGET_ROOTFS_INITRAMFS),y)
   ifeq ($(CONFIG_TARGET_ROOTFS_INITRAMFS_SEPARATE),y)
     define Kernel/SetInitramfs/PreConfigure
        { \
-               grep -v -e CONFIG_BLK_DEV_INITRD $(2)/.config.old > $(2)/.config; \
+               grep -v -e INITRAMFS -e CONFIG_RD_ -e CONFIG_BLK_DEV_INITRD $(2)/.config.old > $(2)/.config; \
                echo 'CONFIG_BLK_DEV_INITRD=y' >> $(2)/.config; \
                echo 'CONFIG_INITRAMFS_SOURCE=""' >> $(2)/.config; \
        }
```

ptpt52, 2024-12-08:
1. with `CONFIG_RD_LZMA` enabled, initramfs kernel boot ok.

robimarko, 2026-01-09:
And why is this an issue?

Like @dangowrt stated, if its an initramfs kernel it has support for unpacking that compression, otherwise not

ptpt52, 2026-01-09:
```
./scripts/diffconfig.sh 
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_tenda_be12l-pro=y
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_LUCI_LANG_zh_Hant=y
CONFIG_MOLD=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-i18n-attendedsysupgrade-zh-cn=y
CONFIG_PACKAGE_luci-i18n-attendedsysupgrade-zh-tw=y
CONFIG_PACKAGE_luci-i18n-base-zh-cn=y
CONFIG_PACKAGE_luci-i18n-base-zh-tw=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-tw=y
CONFIG_PACKAGE_luci-i18n-package-manager-zh-cn=y
CONFIG_PACKAGE_luci-i18n-package-manager-zh-tw=y
CONFIG_PACKAGE_luci-lib-uqr=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-log=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_MOLD=y
# CONFIG_USE_MOLD is not set
```

dangowrt, 2026-01-09:
This is done by
https://github.com/openwrt/openwrt/blob/main/include/kernel-defaults.mk#L96
and yes, tested, and works fine for me and all buildbot-generated images I could test, including those with `CONFIG_USES_SEPARATE_INITRAMFS=y`.