openwrt/openwrt #17195: 24.10 edgerouter-4 USB stack no longer present/working

grobian, 2024-12-07:
### Describe the bug

The USB-3 port on the Ubiquity EdgeRouter-4 (octeon-generic-ubnt_edgerouter-4) no longer works in the 24.10 release candidates.  I compared 23.05.5 and 24.10.0-rc2 firmwares from openwrt downloads.

Below is a dmesg | grep usb from both booted versions, and the tail from dmesg after plugging an USB-3 capable stick in.

23.05.5

[    4.564588] usbcore: registered new interface driver usbfs
[    4.569932] usbcore: registered new interface driver hub
[    4.575270] usbcore: registered new device driver usb
[    4.892956] usbcore: registered new interface driver usb-storage
[   16.260542] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   16.279520] usbcore: registered new interface driver uas

[   94.915862] usb 2-1: new SuperSpeed USB device number 2 using xhci-hcd
[   94.947476] usb-storage 2-1:1.0: USB Mass Storage device detected
[   94.953994] scsi host0: usb-storage 2-1:1.0
[   95.976692] scsi 0:0:0:0: Direct-Access     SanDisk  Ultra USB 3.0    1.00 PQ: 0 ANSI: 6
[   95.986069] sd 0:0:0:0: [sda] 30031250 512-byte logical blocks: (15.4 GB/14.3 GiB)
[   95.995012] sd 0:0:0:0: [sda] Write Protect is off
[   95.999832] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[   96.000214] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[   96.033442]  sda: sda1 sda2
[   96.038021] sd 0:0:0:0: [sda] Attached SCSI removable disk

24.10.0-rc2

[    4.621750] usbcore: registered new interface driver usbfs
[    4.627140] usbcore: registered new interface driver hub
[    4.632396] usbcore: registered new device driver usb
[    4.950032] usbcore: registered new interface driver usb-storage
[   16.306973] usbcore: registered new interface driver uas

On 24.10 nothing happens, and tools like lsusb report an error (-99), /dev/bus/usb does not exist, etc.

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

octeon/generic

### Device

Ubiquiti EdgeRouter 4

### Image kind

Official downloaded image

### Steps to reproduce

- boot the device
- install usbutils
- run lsusb

### Actual behaviour

lsusb: unable to initialize libusb: -99

### Expected behaviour

# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 5.15.167 xhci-hcd xHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.167 xhci-hcd xHCI Host Controller

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

grobian, 2024-12-08:
I tried the following by building an image myself on top of 24.10.0-rc2 with the following changes
- add CONFIG_OCTEON_USB=y which seems present in 23.05.5, but absent in the 6.6 kernel config
- revert 308630aea6da1b5618d169d6ebf208aaa249ef72 (disable PCIe)
both changes appear to make no change to USB being detected/available.

borouhin, 2024-12-21:
I'm facing exactly the same behavior with 23.05.5 as well...
Trying to install OpenWRT on my EdgeRouter 4 - no /dev/sda after booting kernel, no reaction on removing/inserting USB stick (the one I've just booted the kernel from), error -99 running lsusb.
The same with 24.10.0-rc2.

grobian, 2024-12-21:
Hmmm, I tried two different ER-4s and both work with 23.05.5 for me.  Very odd!

borouhin, 2024-12-21:
Maybe I'm doing something wrong, it's the first time I'm trying install OpenWRT on EdgeRouter... So far I've tested 24.10.0-rc2, 23.05.5, 23.05.4, 22.03.7 - no luck with USB :(
Tried installing every possible kmod-* package related to USB, running "block detect" - no result, as if there were no USB devices at all.
Anyway, as my problem seems to be version independent, it probably has nothing to do with this issue. I'll investigate it further myself...

grobian, 2024-12-25:
Tried with 24.10.0-rc4 and the same behaviour persists.  It looks like the usb port doesn't even deliver power to the device attached.

grobian, 2024-12-27:
FWIW I just tested snapshot, and the USB port doesn't work there either.

grobian, 2024-12-27:
I found the problem.  Since https://lore.kernel.org/all/ZJC3LLpUlatnLdnv@lenoch/ we have dwc3-octeon, but this is not enabled/built.  I've made some changes for kmod-usb-dwc3-octeon and now the USB port is working.  Trying to come up with a PR.

grobian, 2024-12-27:
For reference: https://github.com/openwrt/openwrt/pull/17393

robimarko, 2024-12-28:
Fixed via:
- https://github.com/openwrt/openwrt/commit/9adcf61abb7746919114db4030ea1e1a6fc416eb
- https://github.com/openwrt/openwrt/commit/0cba20f0820efa9ac7a728206a3bde43719ced6d

And backported to 24.10

closed 2024-12-28 by robimarko