openwrt/openwrt #17196: [24.10-rc2][ath79][TL-WR1043ND v1] WiFi not working

BlueMax, 2024-12-07:
### Describe the bug

WiFi driver fails during boot-up with:
```
[   29.811702] ath: phy0: Unable to initialize hardware; initialization status: -5
[   29.819119] ath9k 180c0000.wmac: failed to initialize device
[   29.824865] ath9k: probe of 180c0000.wmac failed with error -5
```

Build made with Imagebuilder (see [here](https://forum.openwrt.org/t/only-32mb-of-ram-still-works-for-simple-tasks/136050/23)).

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR1043ND v1

### Image kind

Official downloaded image

### Steps to reproduce

Build image with Imagebuilder:
https://forum.openwrt.org/t/only-32mb-of-ram-still-works-for-simple-tasks/136050/23

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pkgadd, 2024-12-07:
This is a 8/32 device, which is not enough RAM for modern OpenWrt - and the device has been disabled from 24.10.x builds for that reason (you're just circumventing that via the imagebuilder).

BlueMax, 2024-12-07:
Its running as dumb AP only with low memory tweaks (vm.page-cluster=0) and optionally 8MB zram0. There are no serious problems on 23.05.5.
```
# free -h
              total        used        free      shared  buff/cache   available
Mem:          23816       13664        3436         152        6716        6196
Swap:             0           0           0
```
Are there other ath9k devices confirmed working?

StifflersMom, 2024-12-08:
> Are there other ath9k devices confirmed working?

Yes. Ubiquiti UniFi AC Mesh with Qualcomm Atheros QCA9880 running without problems on 24.10.0-rc2 r28161.

BlueMax, 2024-12-09:
The 1043ND has an AR9103 wlan chip. Probably different scenarios here.

Might be related:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=f5a9181d6e94044657df00d3531cf97f2e8d1d16
https://forum.openwrt.org/t/tp-link-wdr4300v1-art-partition-overwrite/56410
https://www.reddit.com/r/openwrt/comments/fj8upa/ath9k_no_wireless_wdr4300v1/

heis2201, 2024-12-11:
> Are there other ath9k devices confirmed working?
My TP-Link TL-WDR4300 v1 (Atheros AR9344 rev 2, ath79/generic) is working fine with `OpenWrt 24.10.0-rc2 r28161-ea17e958b9 / LuCI openwrt-24.10 branch 24.337.27339~b1968d9`

heis2201, 2024-12-11:
> Might be related: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=f5a9181d6e94044657df00d3531cf97f2e8d1d16 https://forum.openwrt.org/t/tp-link-wdr4300v1-art-partition-overwrite/56410 https://www.reddit.com/r/openwrt/comments/fj8upa/ath9k_no_wireless_wdr4300v1/

I don't see anything like that in my logs (TP-Link TL-WDR4300 v1)

BlueMax, 2024-12-11:
I'm currently building a clean image (with CONFIG_PACKAGE_ATH_DEBUG=y) to see if it makes any difference compared to imagebuilder. I will try to raise kernel debug level to get more verbose logs.

BlueMax, 2024-12-12:
Same result on full image. The driver itself loads successfully but probing the hardware fails after 2 seconds. I've tried 'modprobe ath9k debug=0xffffffff' but no change either. I'm not sure if (stripped down?) modprobe even passes arguments here.
If there are other debugging approaches i could try let me know.

DragonBluep, 2024-12-12:
Please test https://github.com/DragonBluep/openwrt/commit/668fac3e6213c5a603d5ac57695c1fdc5ad1bcbf.

BlueMax, 2024-12-12:
Works. Thank you!

closed 2024-12-12 by BlueMax