openwrt/openwrt #17198: mtd-write "Segmentation fault" error when flashing openWRT factory image to ASUS RT-AX1800S (RT-AX54)

jdmonin, 2024-12-07:
### Describe the bug

I attempted to flash the openWRT factory firmware via mtd-write, but I get a "Segmentation fault" message in the terminal. This is on a brand new ASUS RT-AX1800S unit.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ramips/mt7621

### Device

asus rt-ax54

### Image kind

Official downloaded image

### Steps to reproduce

- following directions from device page https://openwrt.org/toh/asus/rt-ax54
- ssh into device as delivered (is running Asus firmware 3.0.0.4_388_33381-g0cf44bd)
- download firmware https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt7621/openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin with wget
- verify sha256sum
- mtd-write segfaults , both from root's dir in `/tmp` and from `/jffs`

### Actual behaviour

- mtd-write segfaults , both from root's dir in `/tmp` and from `/jffs`

```
linksys@RT-AX1800S-BC38:/tmp/home/root# uname -a
Linux RT-AX1800S-BC38 5.4.225 #1 SMP Thu Mar 28 14:46:29 CST 2024 aarch64 ASUSWRT

linksys@RT-AX1800S-BC38:/tmp/home/root# wget https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt7621/openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin
[...]
openwrt-23.05.4-ramips-mt7621- 100%[====================================================>]   9.00M  8.41MB/s    in 1.1s    
2024-12-07 09:41:29 (8.41 MB/s) - 'openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin' saved [9437184/9437184]

linksys@RT-AX1800S-BC38:/tmp/home/root# openssl sha256 < openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin
(stdin)= b96143923e5707fa0bc75308cd5c3b77718dc2316230596273404f14cbb9e036

linksys@RT-AX1800S-BC38:/tmp/home/root# mtd-write -i openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin -d Kernel ; date
Segmentation fault

# also fails from /jffs

/jffs/tmp# mtd-write -i openwrt-23.05.4-ramips-mt7621-asus_rt-ax54-squashfs-factory.bin -d Kernel
Segmentation fault

# dmesg

nvram_commit(): pid 1 comm [init]
_nvram_free: 1(init) nvram_idx(1 / 2)
found private data block and skip it
nvram_commit(): pid 2048 comm [aaews]
_nvram_free: 2048(aaews) nvram_idx(0 / 2)
found private data block and skip it
value=1, hook is enabled now!
7981@C23L3rax0,DfsCacEndUpdate() 3310: [RDM] CAC end. Enable MAC TX.
nvram_commit(): pid 2723 comm [firmware_check_]
_nvram_free: 2723(firmware_check_) nvram_idx(1 / 2)
found private data block and skip it
potentially unexpected fatal signal 11.
===DDD===
7f8e5b0000 - 7f8e5d1000, [libgcc_s.so.1]
7f8e5d3000 - 7f8e5f0000, [libcodb.so]
7f8e5f2000 - 7f8e6ea000, [libsqlite3.so.0.8.6]
7f8e6f1000 - 7f8e712000, [libev.so.4.0.0]
7f8e714000 - 7f8e726000, [libasuslog.so]
7f8e728000 - 7f8e747000, [libjansson.so]
7f8e749000 - 7f8e76c000, [liblldpctl.so.4.8.0]
7f8e770000 - 7f8e7b4000, [libamas-utils.so]
7f8e7b6000 - 7f8e7ce000, [libjson-c.so.2.0.2]
7f8e7d0000 - 7f8e831000, [libcurl.so.4.8.0]
7f8e835000 - 7f8e84b000, [libletsencrypt.so]
7f8e84d000 - 7f8e86e000, [libvpn.so]
7f8e871000 - 7f8e882000, [libcfgmnt.so]
7f8e884000 - 7f8e8b3000, [libbwdpi.so]
7f8e8b8000 - 7f8e936000, [libssl.so.1.1]
7f8e93f000 - 7f8eb76000, [libcrypto.so.1.1]
7f8eb99000 - 7f8ec18000, [libshared.so]
7f8ec29000 - 7f8ec3b000, [libnvram.so]
7f8ec3d000 - 7f8ec5e000, [libconn_diag.so]
7f8ec62000 - 7f8ece2000, [libc.so]
CPU: 1 PID: 3147 Comm: mtd-write Tainted: P                  5.4.225 #1
Hardware name: RT-AX57M (DT)
pstate: 00000000 (nzcv daif -PAN -UAO)
pc : 000000000045ef90
lr : 000000000045ef90
sp : 0000007fef027d50
x29: 0000007fef027d50 x28: 0000007f8ecf5000
x27: 0000000000000005 x26: 0000000000000000
x25: 0000007f8e76f440 x24: 000000000050fdf4
x23: 0000000000000000 x22: 0000000000000000
x21: 00000000ffffffff x20: 0000000000900000
x19: 0000007fef027f80 x18: 0000000000000000
x17: 0000007f8eca0f7c x16: 0000000000560090
x15: 0000000000800000 x14: 0000000000800000
x13: 000000000050fe2f x12: 0000007fef027a8c
x11: 00000000ffffffff x10: 0000007fef027fba
x9 : 000000000000000a x8 : 00000000000000d7
x7 : fefefefefefefeff x6 : 0000000000000000
x5 : 0000000000000000 x4 : 0000000000000002
x3 : 0000000000000000 x2 : 0000000000000000
x1 : 0000000000900000 x0 : 0000000000000000

### Expected behaviour

- mtd-write succeeds, allowing me to boot into openwrt

### Additional info

Because I'm not yet running openwrt, guessed required field "OpenWrt version" from https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt7621/version.buildinfo

I saw a recent very similar issue (but with less info provided, so it's tagged invalid): #17119

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jdmonin, 2024-12-07:
I'm excited to start using OpenWRT, so i'd like to get this sorted out. I have a bit of developer experience (but not with embedded or openwrt) and would like to try to help.

Thanks for taking a look at this issue.

I saw a recent very similar issue (but with less info provided, so it's tagged invalid): #17119

jdmonin, 2024-12-07:
Just noticed this, there is no mtd named `Kernel` but there is a `linux`:

```
linksys@RT-AX1800S-BC38:/tmp/home/root# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00400000 00020000 "Bootloader"
mtd1: 07c00000 00020000 "UBI_DEV"
mtd2: 0001f000 0001f000 "nvram"
mtd3: 000f8000 0001f000 "Factory"
mtd4: 000f8000 0001f000 "Factory2"
mtd5: 045fe000 0001f000 "linux"
mtd6: 042e3724 0001f000 "rootfs"
```

jdmonin, 2024-12-07:
fwiw the sticker on the retail box shows:
```
Model: RT-AX1800S V2
H/W Ver: A1
```

Could this be new hardware that needs a slightly different install?

dmesg includes: `payload: t-neptune-main-2111-7981-imp-MT7981_E1_ASIC_ROM_RAM_REBB_IMP-20230814171402`

jdmonin, 2024-12-07:
OK yes, the wiki page that says it supports this router shows `MIPS` in boot logs.  But my device's `uname -a` shows `aarch64` instead

musashino205, 2024-12-08:
RT-AX1800S (v1): MediaTek MT7621A (MIPS32, ramips/mt7621)
RT-AX1800S v2: MediaTek MT7981 (ARM64, mediatek/filogic)  ---> usupported model

remittor, 2024-12-08:
@jdmonin 
> Model: RT-AX1800S V2

https://forum.openwrt.org/t/openwrt-support-for-asus-rt-ax57m-ax1800s-v2/208534

@musashino205 
> RT-AX1800S v2: MediaTek MT7981 (ARM64, mediatek/filogic) ---> usupported model

Well, that's not quite true.

musashino205, 2024-12-08:
>> RT-AX1800S v2: MediaTek MT7981 (ARM64, mediatek/filogic) ---> usupported model
>
> Well, that's not quite true.

I see, but at least there is no official support.

jdmonin, 2024-12-08:
Thank you both for the info.

At this point my worry is that others may make the same mistake. My goal was to buy a router supported by OpenWRT to run OpenWRT, with specific speeds and features. That took me to https://openwrt.org/toh/views/toh_available_16128_ax-wifi , which took me to the [RT-AX54 device page](https://openwrt.org/toh/asus/rt-ax54), which gave the marketing names I used to search online retailers.

It makes sense that the table doesn't mention RT-AX1800S V2, because RT-AX1800S is a marketing name for RT-AX54, but the device page doesn't mention there's an incompatible RT-AX1800S V2 , which you can tell by checking `uname -m` (aarch64) or `cat /proc/device-tree/model` (RT-AX57M).

Would it make sense to add that warning to the device page? Is there a typical way to do so? Or maybe clarify the first paragraph something like "This device has few marketing name : AX1800HP , AX1800S , AX54HP. However there is an AX1800S V2 which is actually model RT-AX57M and doesn't use the instructions or firwmare on this page."

aavilaherrera, 2024-12-08:
I made the same mistake recently, even after reading warnings about ASUS's misleading naming schemes. There was no mention of "V2" on the product page I ordered from. The box says "AX1800 Dual Band" and "RT-AX1800S" in giant font. Only the tiny barcode sticker says "RT-AX1800S V2". Very disappointing.

Now, that I see @remittor has built a working image, perhaps I'll take this as an opportunity to learn the openwrt build process; but I'll probably just send it back for a tiny x86 or banana/raspberry pi if I still can.

warrenmr, 2025-12-12:
Happened to me just now.  Happy to find this thread.

theofficialgman, 2025-12-12:
@warrenmr I have made a PR (see the link right above your commit). Waiting on maintainer response to hopefully merge even though the WAN port doesn't work (issue unknown; you can map one of the LAN ports as WAN if you require a WAN port).

theofficialgman, 2025-12-12:
[openwrt-mediatek-filogic-asus_rt-ax57m-.zip](https://github.com/user-attachments/files/24119184/openwrt-mediatek-filogic-asus_rt-ax57m-.zip)

here is my build from that PR (which is off of main) as of 2 week ago. install instructions in the commit https://github.com/openwrt/openwrt/pull/20491/commits

nam-van-nguyen, 2025-12-16:
My understanding is that [remittor](https://github.com/openwrt-xiaomi/openwrt/commits?author=remittor) also create a branch that support this router.  Maybe I am mistaken, and it's only supporting the processor, not this particular model?

Or, does this branch (from theofficialgman) provides a more complete support somehow?  

Thanks.

theofficialgman, 2025-12-26:
All, support for this router family has been added to openwrt main (snapshot). PRs for 24.10 (https://github.com/openwrt/openwrt/pull/21236) and 25.12 (https://github.com/openwrt/openwrt/pull/21237) have been made but haven't merged yet.

You can install the snapshot build from the official firmware downloader https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Ffilogic&id=asus_rt-ax57m . I suggest using the "customize installed packages" option and "request build" to make an image that includes the UI pre-installed like the release builds.

theofficialgman, 2026-01-07:
The PR for 25.12 was merged and a new pre-release has been cut that includes support https://github.com/openwrt/openwrt/releases/tag/v25.12.0-rc2

https://firmware-selector.openwrt.org/?version=25.12.0-rc2&target=mediatek%2Ffilogic&id=asus_rt-ax57m