openwrt/openwrt #17199: Banana Pi BPI-R3 Mini fan issue

leonidbobovich, 2024-12-07:
### Describe the bug

On Banana Pi BPI-R3 Mini  Pam fan is stopping to work immidiadly after boot completion 

Writing 0 into /sys/devices/platform/pwm-fan/hwmon/hwmon1/pwm1_enable solves the problem

Hacked by adding 
echo 0 > /sys/devices/platform/pwm-fan/hwmon/hwmon1/pwm1_enable

into /etc/rc.local file

NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28246-ce663856fb"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28246-ce663856fb"
OPENWRT_BUILD_DATE="1733305726"

### OpenWrt version

r28246-ce663856fb

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPi-R3 Mini

### Image kind

Official downloaded image

### Steps to reproduce

Boot device
Observe that fan is not working
Wait for device overheating
Write 0 into sys/devices/platform/pwm-fan/hwmon/hwmon1/pwm1_enable
Fan is working and device is cooling down

### Actual behaviour

Device is overheating 

### Expected behaviour

Fan is working and device stay cool 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

alexmbird, 2025-01-25:
I have a bpi r3-mini and have confirmed the behaviour.  `echo 0 > /sys/devices/platform/pwm-fan/hwmon/hwmon1/pwm1_enable` and my expensive 5G modem stops cooking 🔥 .  Please fix :)

ghost-tara, 2025-07-27:
echo 0 > /sys/devices/platform/pwm-fan/hwmon/hwmon2/pwm1_enable resolves this issue when using an image built myself, while the other resolved it on distributed image.

semiscary, 2025-11-08:
> Execute the command, the fan works normally, and write to the /etc/rc.local file to be applied after reboot:
> 
> echo 100000 > /sys/class/thermal/thermal_zone0/trip_point_0_temp
> echo 75000 > /sys/class/thermal/thermal_zone0/trip_point_1_temp
> echo 60000 > /sys/class/thermal/thermal_zone0/trip_point_2_temp
> echo 50000 > /sys/class/thermal/thermal_zone0/trip_point_3_temp
> echo 45000 > /sys/class/thermal/thermal_zone0/trip_point_4_temp

Execute what command? I don't understand the point of your echos.